felis/USB_Host_Shield_2.0 #128: Wired Xbox Controller Not Working

SpencerLittle, 2015-01-02:
I'm relatively new to Arduino and am having difficulties. I'm using an Arduino Uno with a SainSmart USB host and am trying to get a wired Xbox 360 controller simply to work off of the example code given. What happens is when I upload the code, the controller blinks the LED's once and when I open the console it reads "XBOX USB Library Started", the controller will blink again, and then nothing will happen or display on the console window. I'm not sure what exactly is going on. I've tried external power via a wall adapter with 9V at 1A but nothing helps. Please help and thanks.

Lauszus, 2015-01-02:
Please enable serial debugging: https://github.com/felis/USB_Host_Shield_2.0#enable-debugging and post the output.

SpencerLittle, 2015-01-02:
Well the good news is it is doing something now. The bad news is it isn't in my favor.
Now it says:

XBOX USB Library Started
Unknown Device Connected - VID: 162E PID: BEEF
Xbox 360 Init Failed, error code: D1

Not sure exactly what that means nor what error code D1 is.

Lauszus, 2015-01-03:
Try to comment out the following line: https://github.com/felis/USB_Host_Shield_2.0/blob/master/XBOXUSB.cpp#L108-L109 and tell me if it works.

SpencerLittle, 2015-01-03:
That seemed to do the trick. Thank you so much and I'll let you know if anything else comes up.

Lauszus, 2015-01-03:
I just added support for it properly: https://github.com/felis/USB_Host_Shield_2.0/commit/96dfd7e37d395a70d3d8c1bf211eaa8323b2d857. Please redownload the library and confirm that it's working.

SpencerLittle, 2015-01-06:
I apologize for my delayed response. The new library works fine. It actually worked the first upload without any problems.

Lauszus, 2015-01-07:
@SpencerLittle thanks for your feedback ;)

closed 2015-01-07 by Lauszus