felis/USB_Host_Shield_2.0 #129: HID keyboard reading ASCII codes enter, backspace

virtapoika, 2015-01-07:
Hi! I'm using your example sketch for the usb kbd input. I'm trying to catch the event when enter or backspace is pressed. Other buttons like abcd..., 1,2,3... and some other buttons like [;] [:] etc. give the ASCII values and catch the event but enter and backspace won't give ASCII code or give any event. :/ What's wrong?

 (inside the OnKeyPressed function example modifications)
  else if((char)key == '='){ }
  else if((char)key == '/'){}

davidbkemp, 2015-01-17:
I had another look at this.  My keyboard combiner converts various non-printable keys like backspace to the constants expected by the Arduino Keyboard.press() function which may not actually be what you are after.  So, on the off chance that you or anyone else is interested, I wrote a gist that prints the ASCII codes of not just the normal printable ASCII keys, but also several of the non-printable keys like backspace, delete, and escape:
https://gist.github.com/davidbkemp/1602e1efd2ab7a6188c7

Lauszus, 2015-01-17:
@davidbkemp thanks for sharing :)

closed 2015-03-24 by Lauszus