felis/USB_Host_Shield_2.0 #130: Wired XBOX controller

fred42, 2015-01-15:
I downloaded the library for the USB Host Shield to use the XBOXUSB library to try and run an XBOX controller through an arduino and USB Host Shield. I've had problems getting the example sketch to work, though. Whenever I run the code and open up the Serial Monitor, I get the response that OSC did not start. I've tried testing both the controller and the arduino and shield, and they all work fine by themselves, but I can't seem to get them to work together for this sketch. How do I fix this problem?

Lauszus, 2015-01-15:
What shield are you using?

xxxajk, 2015-01-15:
I think we are starting to need a general FAQ about these other
incompatible boards...
That said; there are only a few boards that are indeed 100%
clones/compatible.

First, if your board lacks the ICSP connection to your arduino, then we
can't help you software-wise. This means you will have to do the required
modifications to your board yourself.
If you can't do hose modifications, we recommend that you send the board
back to the place where you bought it from, and buy a genuine USB Host
Shield from circuits@home or other authorized dealer. Buying the real
product helps us help you by funding software development.

felis, 2015-01-16:
We already have a page listing compatible BT dongles. This text can be added there and page can be made into a FAQ.

xxxajk, 2015-01-16:
We still should place a big fat warning about shields that are not from
circuits@home right here on github, in the readme...

fred42, 2015-01-21:
I don't know exactly what the shield is called, but it is the USB Host Shield for Arduino that I found on linksprite.com

Lauszus, 2015-01-23:
@fred42 sorry but that shield is not supported. Please support us by buying the official shield at either: http://www.circuitsathome.com/products-page/arduino-shields or http://shop.tkjelectronics.dk/product_info.php?products_id=43.

closed 2015-01-23 by Lauszus

mgeorgedeveloper, 2015-02-09:
Just wasted two days before randomly coming across this. I also have a USB host shield that says "www.linksprite.com" on it, that I bought from Maplin. I got past the "OSC did not start" issue by powering the Arduino/shield via external adapter or through vin, but then the next issue is that Xbox.Xbox360Connected line never goes true, neither does Xbox.XboxReceiverConnected for the wireless controller example code.

So the question is, will this solve my issue:

http://store.arduino.cc/index.php?main_page=product_info&cPath=37_5&products_id=541

I'm hoping to buy locally in the UK.

Thanks in advance for any advice.

Lauszus, 2015-02-09:
No please buy the official shield from either: http://shop.tkjelectronics.dk/product_info.php?products_id=43 or http://www.circuitsathome.com/products-page/arduino-shields.

xxxajk, 2015-02-09:
http://store.linksprite.com/usb-host-shield/
Is this your board? 
As far as I can tell, this should work on an __UNO ONLY__ and *possibly* with the ancient 1.0 library
Worse, it is a VERY old design :-) and it does not help us at all to buy such broken clones.

xxxajk, 2015-02-09:
Nice cross post :-) Someone needs to tell linksprite that their product is not compatible.
@mgeorgedeveloper Get your money back if you can for that clone board.

mgeorgedeveloper, 2015-02-10:
@Lauszus - "No please buy the official shield from either...."

This is really confusing, because this page:

http://arduino.cc/en/Main/ArduinoUSBHostShield

Has clear references to the library:

"The Arduino USB Host Shield can be used with the "USB Host Library for Arduino" hosted by Oleg Mazurov and Alexei Glushchenko from circuits@home, Kristian Lauszus and Andrew Kroll on GitHub (download)."

And also specifically states all the controllers as supported:

"-HID devices: keyboards, mice, joysticks, etc. 
-Game controllers: Sony PS3, Nintendo Wii, Xbox360. 
......
"

Are they not telling the truth?

#consumerminefield

Lauszus, 2015-02-10:
Yes it is supported, but if you want to support us, the people who have actually written the code, then please buy the shield from one of us.

mgeorgedeveloper, 2015-02-10:
Ah thanks for clarifying. It's basically down to me being impatient, and wanting this thing delivered next day - so I'm buying locally (UK). I'm assuming from the $ pricing that the board you're selling would be coming from the USA? Apologies if I made the wrong assumptions here.

Can you point me to the correct donations page for supporting this project directly?

Lauszus, 2015-02-10:
Okay no problem. You can donate directly to us here: https://github.com/felis/USB_Host_Shield_2.0#donate.

jwhitmarsh, 2016-03-12:
@mgeorgedeveloper did you ever get your board to work? I've just gone through the same thing as you.  I'm not trying to avoid supporting the developers - I just didn't realise this shield wasn't supported.

mgeorgedeveloper, 2017-04-26:
@jwhitmarsh sorry didn't see this until now. It's been a while, so I can't actually remember how it worked out in the end. Sorry I couldn't be of more help.