felis/USB_Host_Shield_2.0 #131: Display Panel with Prolific USB to Serial Connection

stanley1211, 2015-01-23:
Hi, I am building a scoring system with a LED display panel. The panel I acquired has a USB interface to recieve data there. This panel uses Prolific chip for converting USB to serial internally, its serial baud rate is working at pretty low rate of only 1200bps. Anyway, my scoring system is run on Arduino Mega 2560 and I have the newest Circuit@Home USB host shield 2.0v. I followed the instruction to install the USB Host Shield library and compiled the PL2303 terminal Sketch successfully, when I run the code, suppose a turn on/ turn off code will be sent to the display panel, however, it didn't work that way. I dump the debug screen (I set the ENABLE_UHS_DEBUGGING to 1) as shown

Start
PL Init
Addr:01
NC:01
0000: 09 02 27 00 01 01 00 80 32 09 04 00 00 03 FF 00 
0010: 00 00 07 05 81 03 0A 00 01 07 05 02 02 40 00 00 
0020: 07 05 83 02 40 00 00 Conf.Val: 01
Iface Num: 00
Alt.Set: 00
Endpoint descriptor:
Length:		07
Type:		05
Address:	81
Attributes:	03
MaxPktSize:	000A
Poll Intrv:	01
Conf.Val: 01
Iface Num: 00
Alt.Set: 00
Endpoint descriptor:
Length:		07
Type:		05
Address:	02
Attributes:	02
MaxPktSize:	0040
Poll Intrv:	00
Conf.Val: 01
Iface Num: 00
Alt.Set: 00
Endpoint descriptor:
Length:		07
Type:		05
Address:	83
Attributes:	02
MaxPktSize:	0040
Poll Intrv:	00
Conf:01
PL configured

Looks like the device was initiallized successfully, however, it seems that no data was sent out by the SndData(), the RcvData(0 seems to have a long wait before returning. Can anybody give me some hints?

Lauszus, 2015-03-24:
This might be related to: https://github.com/felis/USB_Host_Shield_2.0/pull/110.

Lauszus, 2017-10-12:
Closing due to lack of feedback. #110 should fix this.

closed 2017-10-12 by Lauszus