felis/USB_Host_Shield_2.0 #132: Intel Edison breakout board compatibility.

r2DoesInc, 2015-01-27:
I am trying to use this library with the AndroidAccessory Library to communicate with my Intel Edison. Currently your library does not seem to support the edison. Do you have plans to bring Edison support to this project?

xxxajk, 2015-01-28:
Donate one so we have one to work with and we can possibly do just that, :-)

r2DoesInc, 2015-02-02:
The company I am employed with - Cloudspace, https://github.com/cloudspace - would love to send you one if you think we could get a relatively decent turn around. We are currently using this library with our in progress ArdroBot project, and we are looking to switch to the Intel Edison chip from our ATmega2560 based current board. 

Do you have a way we can setup a more direct communication?

xxxajk, 2015-02-05:
email/hangouts?

r2DoesInc, 2015-02-06:
Sure, send me an email @ ken@cloudspace.com

Lauszus, 2015-03-19:
@r2DoesInc I just added support for the Intel Edison at the following branch: https://github.com/felis/USB_Host_Shield_2.0/tree/Galileo. Please test and report back, as I can not test it, as I don't have the hardware.

Lauszus, 2015-03-19:
Also please play around with the following value: https://github.com/felis/USB_Host_Shield_2.0/blob/Galileo/usbhost.h#L55 and report back what the maximum speed you can reliable use is.

Lauszus, 2015-03-25:
Also see the following PR: https://github.com/felis/USB_Host_Shield_2.0/pull/142.

r2DoesInc, 2015-04-22:
Apologies, I have only recently switched back to this project to test your changes.
Unfortunately, the board is unable to connect to the device.

Using the basic code 
```
void setup() {
  Serial.begin(115200);
  while (!Serial); // Wait for serial port to connect - used on Leonardo, Teensy and other boards with built-in USB CDC serial connection
  if (Usb.Init() == -1) {
    Serial.print("\r\nOSCOKIRQ failed to assert");
    while (1); // halt
  }
  Serial.print("\r\nArduino Blink LED Started");
}
```

I only get the OSCOKIRQ failed to assert message.

Lauszus, 2015-04-22:
Okay. Could you try to set this to 0: https://github.com/felis/USB_Host_Shield_2.0/blob/master/avrpins.h#L1026.

Also this example might give some more information: https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/board_qc/board_qc.ino.

r2DoesInc, 2015-04-23:
Changing the value to 0 did not seem to have an effect.

```
Circuits At Home 2011
USB Host Shield Quality Control Routine
Reading REVISION register... Die revision invalid. Value returned: 00
Unrecoverable error - test halted!!
0x55 pattern is transmitted via SPI
Press RESET to restart test
```

Lauszus, 2015-04-24:
Okay. Try lowering this value: https://github.com/felis/USB_Host_Shield_2.0/blob/master/usbhost.h#L55.

r2DoesInc, 2015-06-03:
Stepping that value down by a power of 10 4 times did not seem to have any effect.

Lauszus, 2015-06-03:
Okay. It's pretty hard for me to debug since I don't have an Edison.

Lauszus, 2015-07-13:
@r2DoesInc if you are still happy to donate one I will look into this issue.

commit 42a26346 (2015-03-19): Added support for Intel Edison

Closes #132