felis/USB_Host_Shield_2.0 #133: problem with USB bluetooth dongle to connect ps3 controller ?

nmn11, 2015-02-08:


Lauszus, 2015-02-08:
Can you give some more information? I will close this for now. Reopen if needed.

closed 2015-02-08 by Lauszus

nmn11, 2015-02-09:
My bluetooth usb dongle could not be detected in the library  usb host shield 2.0  - >> bluetooth--- >> PS3BT ?
Can you give me some information to bluetooth dongle suitable for usb-host master library 2.0..??

Lauszus, 2015-02-09:
Please enable serial debugging: https://github.com/felis/USB_Host_Shield_2.0#enable-debugging and post the output, so I can see what is going on.

nmn11, 2015-02-10:
OK, thank's very much for your information.. i have try it..
now,my bluetooth dongle can be used but why my ps3 controller dualshock can
not pair with it... example like that on image 111.PNG and my ps3
controller always blink...

Lauszus, 2015-02-10:
Have you followed the paring procedure as described here: https://github.com/felis/USB_Host_Shield_2.0#ps3-library?

Lauszus, 2015-02-10:
Btw I can't see the image.

nmn11, 2015-02-11:
like that,..
[image]

please help me !!

nmn11, 2015-02-11:
I've done it according to the procedure as above. but always still like that...

Lauszus, 2015-02-11:
This is a problem with your dongle, as it doesn't support 2.0+EDR, so the last data sent from the controller is not received.

There is nothing else to do than to try another dongle. A list is available here: https://github.com/felis/USB_Host_Shield_2.0/wiki/Bluetooth-dongles.