felis/USB_Host_Shield_2.0 #134: xbox one controller support

guruthree, 2015-02-12:
I'm trying to write an xbox one controller class. I'm not a USB expert or anything, but based on the xpad kernel driver, it looked like it should be fairly trivial to copy the XBOXUSB code: take out rumble and LED support, change the PID, swap the input/output endpoints, add an init packet, and then interpret the packets sent back according to the new protocol. Unfortunately, I've hit a stumbling block in that I can't get my code to initialise the controller. Sending the initialisation packet of 0x05, 0x20 with XboxCommand returns an hrSTALL and that's the end of that. Any idea what I may be doing wrong?

Lauszus, 2015-02-12:
It shouldn't be necessary to send any special command, at least I don't believe so. Have you found any other drivers for instance for Linux? That is what I normally use as inspiration.

guruthree, 2015-02-12:
The xpad driver in the linux kernel has Xbox one controller support, which is what I've based my work off of (https://github.com/torvalds/linux/blob/master/drivers/input/joystick/xpad.c#l954). I've also found a driver written in go (https://github.com/kylelemons/xbox/blob/master/xbox.go#l431) and a protocol description (https://github.com/quantus/xbox-one-controller-protocol#0x05-start-controller-with-input). All three sources agree that you need to send a packet to the controller to initialize it so that it starts sending data back. I just can't figure out why when I try and send the initialization packet ctrlReq returns 0x05 (hrSTALL).

guruthree, 2015-02-14:
By changing from ctrlReq to outTransfer to send the initialisation, it's worked.

closed 2015-02-14 by guruthree

Lauszus, 2015-02-15:
Good to hear you got it working!