felis/USB_Host_Shield_2.0 #141: Arduino 1.6.1 Compatibility?

k3ng, 2015-03-22:
I'm having issues attempting to compile under Arduino 1.6.1 (using Sublime / Stino).  Below is the verbose compile output, compiling example code.  I can compile OK under Arduino 1.0.5.  It appears 1.6.1 includes some USB support which may be conflicting?  Thanks for your help!

Compiling USBHIDBootKbd...
Creating /Users/anthonygood/Documents/Arduino/build/USBHIDBootKbd/USBHIDBootKbd.ino.cpp.o...
"/Applications/Arduino 1.6.1.app/Contents/Resources/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=161 -DARDUINO_AVR_ADK -DARDUINO_ARCH_   "-I/Users/anthonygood/Documents/Arduino/build/USBHIDBootKbd" "-I/Users/anthonygood/Documents/Arduino/sketchbook/libraries/USB_Host_Shield/examples/HID/USBHIDBootKbd" "-I/Applications/Arduino 1.6.1.app/Contents/Resources/Java/hardware/tools/avr/avr/include" "-I/Applications/Arduino 1.6.1.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino 1.6.1.app/Contents/Resources/Java/hardware/arduino/avr/variants/mega" "-I/Applications/Arduino 1.6.1.app/Contents/Resources/Java/libraries/USBHost/src" "-I/Users/anthonygood/Documents/Arduino/sketchbook/libraries/USB_Host_Shield" "-I/Applications/Arduino 1.6.1.app/Contents/Resources/Java/hardware/arduino/avr/libraries/SPI"  "/Users/anthonygood/Documents/Arduino/build/USBHIDBootKbd/USBHIDBootKbd.ino.cpp" -o "/Users/anthonygood/Documents/Arduino/build/USBHIDBootKbd/USBHIDBootKbd.ino.cpp.o"
In file included from /Users/anthonygood/Documents/Arduino/build/USBHIDBootKbd/USBHIDBootKbd.ino.cpp:3:0:
/Applications/Arduino 1.6.1.app/Contents/Resources/Java/libraries/USBHost/src/hidboot.h: In member function 'virtual void HIDBoot<BOOT_PROTOCOL>::EndpointXtract(uint32_t, uint32_t, uint32_t, uint32_t, const USB_ENDPOINT_DESCRIPTOR*)':
/Applications/Arduino 1.6.1.app/Contents/Resources/Java/libraries/USBHost/src/hidboot.h:474:62: error: 'UOTGHS_HSTPIPCFG_PTYPE_INTRPT' was not declared in this scope
   pipe = UHD_Pipe_Alloc(bAddress, epInfo[index].deviceEpNum, UOTGHS_HSTPIPCFG_PTYPE_INTRPT, UOTGHS_HSTPIPCFG_PTOKEN_IN, epInfo[index].maxPktSize, 10, UOTGHS_HSTPIPCFG_PBK_1_BANK);
                                                              ^
/Applications/Arduino 1.6.1.app/Contents/Resources/Java/libraries/USBHost/src/hidboot.h:474:93: error: 'UOTGHS_HSTPIPCFG_PTOKEN_IN' was not declared in this scope
   pipe = UHD_Pipe_Alloc(bAddress, epInfo[index].deviceEpNum, UOTGHS_HSTPIPCFG_PTYPE_INTRPT, UOTGHS_HSTPIPCFG_PTOKEN_IN, epInfo[index].maxPktSize, 10, UOTGHS_HSTPIPCFG_PBK_1_BANK);
                                                                                             ^
/Applications/Arduino 1.6.1.app/Contents/Resources/Java/libraries/USBHost/src/hidboot.h:474:151: error: 'UOTGHS_HSTPIPCFG_PBK_1_BANK' was not declared in this scope
   pipe = UHD_Pipe_Alloc(bAddress, epInfo[index].deviceEpNum, UOTGHS_HSTPIPCFG_PTYPE_INTRPT, UOTGHS_HSTPIPCFG_PTOKEN_IN, epInfo[index].maxPktSize, 10, UOTGHS_HSTPIPCFG_PBK_1_BANK);
                                                                                                                                                       ^
/Applications/Arduino 1.6.1.app/Contents/Resources/Java/libraries/USBHost/src/hidboot.h:474:178: error: there are no arguments to 'UHD_Pipe_Alloc' that depend on a template parameter, so a declaration of 'UHD_Pipe_Alloc' must be available [-fpermissive]
   pipe = UHD_Pipe_Alloc(bAddress, epInfo[index].deviceEpNum, UOTGHS_HSTPIPCFG_PTYPE_INTRPT, UOTGHS_HSTPIPCFG_PTOKEN_IN, epInfo[index].maxPktSize, 10, UOTGHS_HSTPIPCFG_PBK_1_BANK);
                                                                                                                                                                                  ^
/Applications/Arduino 1.6.1.app/Contents/Resources/Java/libraries/USBHost/src/hidboot.h:474:178: note: (if you use '-fpermissive', G++ will accept your code, but allowing the use of an undeclared name is deprecated)
/Applications/Arduino 1.6.1.app/Contents/Resources/Java/libraries/USBHost/src/hidboot.h: In member function 'virtual uint32_t HIDBoot<BOOT_PROTOCOL>::Release()':
/Applications/Arduino 1.6.1.app/Contents/Resources/Java/libraries/USBHost/src/hidboot.h:500:54: error: there are no arguments to 'UHD_Pipe_Free' that depend on a template parameter, so a declaration of 'UHD_Pipe_Free' must be available [-fpermissive]
  UHD_Pipe_Free(epInfo[epInterruptInIndex].hostPipeNum);
                                                      ^
In file included from /Users/anthonygood/Documents/Arduino/sketchbook/libraries/USB_Host_Shield/Usb.h:33:0,
                 from /Users/anthonygood/Documents/Arduino/sketchbook/libraries/USB_Host_Shield/usbhub.h:20,
                 from /Users/anthonygood/Documents/Arduino/build/USBHIDBootKbd/USBHIDBootKbd.ino.cpp:4:
/Users/anthonygood/Documents/Arduino/sketchbook/libraries/USB_Host_Shield/address.h: At global scope:
/Users/anthonygood/Documents/Arduino/sketchbook/libraries/USB_Host_Shield/address.h:32:8: error: redefinition of 'struct EpInfo'
 struct EpInfo {
        ^
In file included from /Applications/Arduino 1.6.1.app/Contents/Resources/Java/libraries/USBHost/src/Usb.h:27:0,
                 from /Users/anthonygood/Documents/Arduino/build/USBHIDBootKbd/USBHIDBootKbd.ino.cpp:2:
/Applications/Arduino 1.6.1.app/Contents/Resources/Java/libraries/USBHost/src/address.h:42:8: error: previous definition of 'struct EpInfo'
 struct EpInfo
        ^
In file included from /Users/anthonygood/Documents/Arduino/sketchbook/libraries/USB_Host_Shield/Usb.h:33:0,
                 from /Users/anthonygood/Documents/Arduino/sketchbook/libraries/USB_Host_Shield/usbhub.h:20,
                 from /Users/anthonygood/Documents/Arduino/build/USBHIDBootKbd/USBHIDBootKbd.ino.cpp:4:
/Users/anthonygood/Documents/Arduino/sketchbook/libraries/USB_Host_Shield/address.h:57:8: error: redefinition of 'struct UsbDeviceAddress'
 struct UsbDeviceAddress {
        ^
In file included from /Applications/Arduino 1.6.1.app/Contents/Resources/Java/libraries/USBHost/src/Usb.h:27:0,
                 from /Users/anthonygood/Documents/Arduino/build/USBHIDBootKbd/USBHIDBootKbd.ino.cpp:2:
/Applications/Arduino 1.6.1.app/Contents/Resources/Java/libraries/USBHost/src/address.h:75:8: error: previous definition of 'struct UsbDeviceAddress'
 struct UsbDeviceAddress
        ^
In file included from /Users/anthonygood/Documents/Arduino/sketchbook/libraries/USB_Host_Shield/Usb.h:33:0,
                 from /Users/anthonygood/Documents/Arduino/sketchbook/libraries/USB_Host_Shield/usbhub.h:20,
                 from /Users/anthonygood/Documents/Arduino/build/USBHIDBootKbd/USBHIDBootKbd.ino.cpp:4:
/Users/anthonygood/Documents/Arduino/sketchbook/libraries/USB_Host_Shield/address.h:75:8: error: redefinition of 'struct UsbDevice'
 struct UsbDevice {
        ^
In file included from /Applications/Arduino 1.6.1.app/Contents/Resources/Java/libraries/USBHost/src/Usb.h:27:0,
                 from /Users/anthonygood/Documents/Arduino/build/USBHIDBootKbd/USBHIDBootKbd.ino.cpp:2:
/Applications/Arduino 1.6.1.app/Contents/Resources/Java/libraries/USBHost/src/address.h:102:8: error: previous definition of 'struct UsbDevice'
 struct UsbDevice
        ^
In file included from /Users/anthonygood/Documents/Arduino/sketchbook/libraries/USB_Host_Shield/Usb.h:33:0,
                 from /Users/anthonygood/Documents/Arduino/sketchbook/libraries/USB_Host_Shield/usbhub.h:20,
                 from /Users/anthonygood/Documents/Arduino/build/USBHIDBootKbd/USBHIDBootKbd.ino.cpp:4:
/Users/anthonygood/Documents/Arduino/sketchbook/libraries/USB_Host_Shield/address.h:83:7: error: redefinition of 'class AddressPool'
 class AddressPool {
       ^
In file included from /Applications/Arduino 1.6.1.app/Contents/Resources/Java/libraries/USBHost/src/Usb.h:27:0,
                 from /Users/anthonygood/Documents/Arduino/build/USBHIDBootKbd/USBHIDBootKbd.ino.cpp:2:
/Applications/Arduino 1.6.1.app/Contents/Resources/Java/libraries/USBHost/src/address.h:113:7: error: previous definition of 'class AddressPool'
 class AddressPool
       ^
/Applications/Arduino 1.6.1.app/Contents/Resources/Java/libraries/USBHost/src/address.h:131:26: error: template parameter 'long unsigned int MAX_DEVICES_ALLOWED'
[... 251 lines cut ...]
[... cut ...]
/Applications/Arduino 1.6.1.app/Contents/Resources/Java/libraries/USBHost/src/hidboot.h:213:1: note:   no known conversion for argument 1 from 'USB*' to 'USBHost*'
/Applications/Arduino 1.6.1.app/Contents/Resources/Java/libraries/USBHost/src/hidboot.h:177:7: note: HIDBoot<1u>::HIDBoot(const HIDBoot<1u>&)
 class HIDBoot : public HID
       ^
/Applications/Arduino 1.6.1.app/Contents/Resources/Java/libraries/USBHost/src/hidboot.h:177:7: note:   no known conversion for argument 1 from 'USB*' to 'const HIDBoot<1u>&'
[Stino - Error 1]

Lauszus, 2015-03-23:
Is this a problem only when using Stino or also if you are using Arduino 1.6.1 directly? Because I can't replicate this on my local machine when using Arduino 1.6.1.

It looks like a problem with Stino, as it should not even try to compile the USBHost library for the Due: https://github.com/arduino/Arduino/tree/master/libraries/USBHost when you are using the Arduino ADK: https://github.com/arduino/Arduino/blob/master/libraries/USBHost/library.properties#L9.
It looks like it does not support the new library structure: https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification.

k3ng, 2015-03-23:
My apologies.  This does appear to be a problem with Stino.  I can get it to compile with Arduino 1.6.1.

Thanks for your help!

closed 2015-03-23 by k3ng

Lauszus, 2015-03-23:
You're welcome :)