felis/USB_Host_Shield_2.0 #143: testusbhostFAT.ino will not compile

barcesat, 2015-04-04:
I tried to compile the example both for Teensy 3.1 and Arduino Mega 2560 and did not succeed.
For Arduino Mega 2560 i get these errors:
testusbhostFAT.ino: In function 'void loop()':
testusbhostFAT:391: error: 'UHS_USB_Storage' was not declared in this scope
testusbhostFAT:391: error: 'B' was not declared in this scope
testusbhostFAT:394: error: expected `;' before 'if'
testusbhostFAT.ino:390: warning: unused variable 'My_File_Object_x'
testusbhostFAT:720: error: expected `}' at end of input
testusbhostFAT.ino: At global scope:
testusbhostFAT.ino:122: warning: 'int tty_stderr_flush(__file*)' defined but not used
testusbhostFAT.ino:137: warning: 'int tty_std_flush(__file*)' defined but not used
testusbhostFAT.ino:100: warning: 'parts' defined but not used
testusbhostFAT.ino:105: warning: 'My_Buff_x' defined but not used

For Teensy i get these errors:
testusbhostFAT.ino: In function 'void loop()':
testusbhostFAT:403: error: 'UHS_USB_Storage' was not declared in this scope
testusbhostFAT:403: error: 'B' was not declared in this scope
testusbhostFAT:416: error: expected ';' before 'serialEvent'

Using Arduino 1.0.6

Lauszus, 2015-04-04:
Please try to download the newest version of the Arduino IDE, then please replace ```UHS_USB_Storage ``` with ```UHS_USB_BulkOnly```.

Lauszus, 2015-04-04:
I just tried myself and this is the errors that I am getting after replacing ```UHS_USB_Storage``` with ```UHS_USB_BulkOnly```:

```
Arduino: 1.6.3 (Mac OS X), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

testusbhostFAT.cpp.o: In function `setup':
/Users/Lauszus/testusbhostFAT.ino:275: undefined reference to `freeHeap'
testusbhostFAT.cpp.o: In function `loop':
/Users/Lauszus/testusbhostFAT.ino:396: undefined reference to `freeHeap'
/Users/Lauszus/testusbhostFAT.ino:612: undefined reference to `freeHeap'
generic_storage/generic_storage.cpp.o: In function `RTClock()':
/Users/Lauszus/Dropbox/Arduino/libraries/generic_storage/FAT/FAT.cpp:31: undefined reference to `RTCnow()'
/Users/Lauszus/Dropbox/Arduino/libraries/generic_storage/FAT/FAT.cpp:31: undefined reference to `DateTime::FatPacked()'
collect2: error: ld returned 1 exit status
Error compiling.
```

@xxxajk can you take a look?

Lauszus, 2015-04-04:
Nevermind. I fixed it by updating to the newest version from here: https://github.com/xxxajk/testusbhostFAT. I will push it to master now.

Lauszus, 2015-04-04:
It is now in master. Please redownload the code and report back if it is still not working.

closed 2015-04-04 by Lauszus

barcesat, 2015-04-06:
compilied and uploaded! thanks!!

xxxajk, 2015-04-06:
Cool! :-D

commit 67a7a66d (2015-04-04): Updated to newest version of testusbhostFAT

Fixes #143