felis/USB_Host_Shield_2.0 #144: PS3 Navigation Controller issues

xxinfinityxx, 2015-04-12:
Hello there,

I'm using the USB Host Shield for Arduino Pro Mini with a 3.3v 8mhz Arduino Pro clone board.

It seems like the library freezes and somehow stays in a confused state after few minutes when running the PS3BT sketch compiled with Arduino 1.6.0.
Also, for some reason, the Navigation Controller won't stay on and keeps flashing... It always shuts down shortly after the "freezing" event. 

```
PS3 Bluetooth Library Started
Bluetooth Dongle Initialized
HCI Reset complete
Write class of device
Local Bluetooth Address: 00:1A:7D:DA:71:0A
Wait For Incoming Connection Request
*** swapping dongle & controller
Navigation Controller Connected
Bluetooth Address was set to: 00:1A:7D:DA:71:0A
*** swapping dongle & controller
Bluetooth Dongle Initialized
HCI Reset complete
Write class of device
Local Bluetooth Address: 00:1A:7D:DA:71:0A
Wait For Incoming Connection Request
Mouse is connecting *** ?! Wtf
Incoming Connection Request
Remote Name: Navigation Controller
Connected to Device: 00:18:00:21:18:1D
L2CAP Connection Request - PSM: 00 11 SCID: 00 40 Identifier: 01
HID Control Incoming Connection Request
HID Control Successfully Configured
L2CAP Connection Request - PSM: 00 13 SCID: 00 41 Identifier: 03
HID Interrupt Incoming Connection Request
HID Interrupt Successfully Configured
Wait For Incoming Connection Request
Navigation Controller Enabled

Right
Left
Right
Left
Right
Down
Up
Left
Right
Down
Left
Up
Right
Down
Left
****** Freezes here, no more input from controller even if I press buttons... 
****** (enabling report debugging shows that the arduino does not receive anything anymore)
HCI Disconnected from Device *** Controller shuts down
Wait For Incoming Connection Request
****** Powering up the controller by pressing PS button
Mouse is connecting  *** ?! Wtf - still
Incoming Connection Request
Remote Name: Navigation Controller
Connected to Device: 00:18:00:21:18:1D
Wait For Incoming Connection Request
HCI Disconnected from Device
Wait For Incoming Connection Request
Mouse is connecting
Incoming Connection Request
Remote Name: Navigation Controller
Connected to Device: 00:18:00:21:18:1D
L2CAP Connection Request - PSM: 00 11 SCID: 00 40 Identifier: 01
Wait For Incoming Connection Request
HCI Disconnected from Device
Wait For Incoming Connection Request
Mouse is connecting
Incoming Connection Request
Remote Name: Navigation Controller
Connected to Device: 00:18:00:21:18:1D
Wait For Incoming Connection Request
HCI Disconnected from Device
Wait For Incoming Connection Request
Mouse is connecting
Incoming Connection Request
Remote Name: Navigation Controller
Connected to Device: 00:18:00:21:18:1D
Wait For Incoming Connection Request
HCI Disconnected from Device
Wait For Incoming Connection Request
****** And so-on...
```

Can you help ?

Lauszus, 2015-04-13:
Okay first of all make sure that the Navigation controller is fully charged. Also how long are the wires or is is properly soldered onto the Pro Mini?

Finally are you applying 5V to VBUS?

xxinfinityxx, 2015-04-14:
Yep, fully charged. Actually I mentioned that everything works fine with a
regular PS3 controller so I guess my wiring is fine ;)

5v is connected to vbus through the pad for this purpose (cut the jumper
too).

The shield is about 4mm under the pro mini soldered with header pins.

Any clues?

Lauszus, 2015-04-14:
That sounds very strange as the Navigation controller just acts like the left side of the normal PS3 controller, so I am wondering if it is a problem with your Navigation controller.

Is it an official controller?

xxinfinityxx, 2015-04-14:
It's funny you suggest that - I bought another one yesterday from a US supplier. The only difference between the two is the actual model number CECH-ZCS1U vs CECH-ZCS1H. I just had the chance to test it and it seems like the new one (ZCS1U) works just fine!

It seems like there is some sort of difference between the two put my PS3 does not seem to care...

Any idea why?

Lauszus, 2015-04-17:
I have no idea. Mine is a CECH-ZCS1E and works fine as well.

xxinfinityxx, 2015-04-21:
Do you have any suggestion as to where to start to fix this? I could have one shipped to you so you can analyse it with your USB tool if you'd like.

Lauszus, 2015-04-21:
Hmm yes that might be the only way. Please email me at: kristianl@tkjelectronics.dk and I will send you my shipping address.

jsenecal, 2015-04-27:
Let me know when you get it :)

Lauszus, 2015-04-27:
@jsenecal I will :)

Lauszus, 2015-05-16:
@jsenecal I just received the controller!

The first thing I noticed was that the L1, CROSS and CIRCLE buttons all made a click while pressing it and the D-pad felt a bit mushy. Anyway I decided to take the two controllers apart to compare them and the first thing that struck me was that the one you sent me (CECH-ZCS1H) was missing two screws!

Anyway inside the difference were must bigger. First of all the battery wasn't held down as the original one and the same goes with the PCB for the joystick. Also the components on the PCB's looks way different. Please see attached photos.

My conclusion is that the one you sent me (CECH-ZCS1H) must be fake! A really good one nevertheless. My guess is that someone got hold of the original plastic enclosure and then decided to design their own PCB's.

Anyway I will close this issue for now. Let me know if you want me to take any more pictures.

CECH-ZCS1H controller:
[image]

CECH-ZCS1E controller - notice how both the joystick PCB are held down using some custom arms parts and the battery is inside a held in place my a plastic holder. 
[image]

CECH-ZCS1H battery:
[image]

CECH-ZCS1H PCB:
[image]

CECH-ZCS1E PCB:
[image]

closed 2015-05-16 by Lauszus

Lauszus, 2015-05-16:
I also just noticed that the plastic guide for the reset button must have broken off during production of the CECH-ZCS1H controller (it wasn't inside the controller).

CECH-ZCS1H controller:
[image]

CECH-ZCS1E controller:
[image]

Here is also a picture showing how the CECH-ZCS1H PCB for the joystick is not secured like on the CECH-ZCS1E controller:
[image]

Lauszus, 2015-05-16:
I can't even put it together again, as their is a gab between the two halves. The USB connector also kind of sucks as well, as it is much harder than usual to plug the USB cable in and out.