felis/USB_Host_Shield_2.0 #147: Are any USB Host Shield examples compatible with Teensy 3.1?

frankstripod, 2015-04-19:
I am having trouble using the USB Host Shield 2.0 examples (or any UHS code) with a Teensy 3.1. Every time I change the board to Teensy in the Arduino IDE, all the USB Host Shield 2.0 Examples (and only those) return “Error compiling” and never get to upload.

The PJRC forum was helpful for getting previous versions of Teensyduino.

I did get spi4teensy3 speedtest to upload and work with the serial monitor in all versions of Arduino with Teensyduino! :)

I think I have read all of the instructions, manuals and readme files at circuitsathome.com, pjrc.com, and GitHub. It feels like I'm missing something stupid. Can you please help me, or point me in the right direction?
.
.
.

Example below “Error compiling” message for USBHIDBootKbdAndMouse:

Arduino: 1.6.3 (Windows 7), TD: 1.22, Board: "Teensy 3.1, Serial, 96 MHz optimized (overclock), US English"

Build options changed, rebuilding all
Using library USB Host Shield Library 2.0 in folder: C:\Program Files (x86)\Arduino\Arduino\libraries\USB_Host_Shield_20
Using library spi4teensy3 in folder: C:\Program Files (x86)\Arduino\Arduino\libraries\spi4teensy3 (1.0.x format)
Using library SPI in folder: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\SPI (1.0.x format)

C:\Program Files (x86)\Arduino/hardware/tools/arm/bin/arm-none-eabi-g++ -c -O -g -Wall -ffunction-sections -fdata-sections -MMD -nostdlib -fno-exceptions -felide-constructors -std=gnu++0x -fno-rtti -mthumb -mcpu=cortex-m4 -D__MK20DX256__ -DTEENSYDUINO=122 -DARDUINO=10603 -DF_CPU=96000000 -DARDUINO_ARCH_AVR -DUSB_SERIAL -DLAYOUT_US_ENGLISH -IC:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3 -IC:\Program Files (x86)\Arduino\Arduino\libraries\USB_Host_Shield_20 -IC:\Program Files (x86)\Arduino\Arduino\libraries\spi4teensy3 -IC:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\SPI C:\Users\Frank\AppData\Local\Temp\build8167078760123349979.tmp\USBHIDBootKbdAndMouse.cpp -o C:\Users\Frank\AppData\Local\Temp\build8167078760123349979.tmp\USBHIDBootKbdAndMouse.cpp.o
In file included from C:\Program Files (x86)\Arduino\Arduino\libraries\USB_Host_Shield_20/Usb.h:36:0,
from C:\Program Files (x86)\Arduino\Arduino\libraries\USB_Host_Shield_20/hid.h:20,
from C:\Program Files (x86)\Arduino\Arduino\libraries\USB_Host_Shield_20/hidboot.h:20,
from USBHIDBootKbdAndMouse.ino:1:
C:\Program Files (x86)\Arduino\Arduino\libraries\USB_Host_Shield_20/usbhost.h: In member function 'uint8_t MAX3421e<SPI_SS, INTR>::regRd(uint8_t)':
C:\Program Files (x86)\Arduino\Arduino\libraries\USB_Host_Shield_20/usbhost.h:257:9: error: 'SPI' was not declared in this scope
SPI.transfer(reg);
^
In file included from C:\Program Files (x86)\Arduino\Arduino\libraries\USB_Host_Shield_20/hidboot.h:20:0,
from USBHIDBootKbdAndMouse.ino:1:
C:\Program Files (x86)\Arduino\Arduino\libraries\USB_Host_Shield_20/hid.h: In member function 'virtual HIDReportParser* HID::GetReportParser(uint8_t)':
C:\Program Files (x86)\Arduino\Arduino\libraries\USB_Host_Shield_20/hid.h:159:63: warning: no return statement in function returning non-void [-Wreturn-type]
virtual HIDReportParser* GetReportParser(uint8_t id) {};
^
C:\Program Files (x86)\Arduino\Arduino\libraries\USB_Host_Shield_20/hid.h: In member function 'virtual bool HID::SetReportParser(uint8_t, HIDReportParser*)':
C:\Program Files (x86)\Arduino\Arduino\libraries\USB_Host_Shield_20/hid.h:170:73: warning: no return statement in function returning non-void [-Wreturn-type]
virtual bool SetReportParser(uint8_t id, HIDReportParser *prs) {};
^
Error compiling.

Lauszus, 2015-04-19:
Due to a bug in the Arduino IDE you need to include the SPI library in your sketch as well. Please see: https://github.com/felis/USB_Host_Shield_2.0#boards.

frankstripod, 2015-04-20:
Thank you for replying Lauszus. I tried following the instructions again on that page with no results.
I did notice someone you helped with a similar problem, at 
http://www.circuitsathome.com/mcu/teensy-3-0-now-supported-by-the-usb-host-library/comment-page-1#comment-25432
Even though his include statements got deleted, its clear he was only missing the spi4teensy3 library. Your answer seemed to be clear at
https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/Bluetooth/PS3BT/PS3BT.ino#L11-L14
This is how the sketch includes look on mine:
```
#include <hidboot.h>
#include <usbhub.h>
// Satisfy IDE, which only needs to see the include statment in the ino.
#ifdef dobogusinclude
#include <spi4teensy3.h>
#include <SPI.h>
#endif
```
spi4teensy3 and "USB_Host_Shield_20" are in the right folder. spi4teensy3 speedtest works every time.
I followed your link to adafruit's libraries for beginners and decided to re-installed 1.6.3 into a folder without spaces, which like before didn't help, but a different “error compiling” message appears (below). Any help would be appreciated, Thank you! :)
...
Arduino: 1.6.3 (Windows 7), TD: 1.22, Board: "Teensy 3.1, Serial, 96 MHz optimized (overclock), US English"
Using library USB Host Shield Library 2.0 in folder: C:\Users\Frank\Arduino\Arduino\libraries\USB_Host_Shield_20  Using library spi4teensy3 in folder: C:\Users\Frank\Arduino\Arduino\libraries\spi4teensy3 (1.0.x format) Using library SPI in folder: C:\Users\Frank\Arduino\hardware\teensy\avr\libraries\SPI (1.0.x format)  C:\Users\Frank\Arduino/hardware/tools/arm/bin/arm-none-eabi-g++ -c -O -g -Wall -ffunction-sections -fdata-sections -MMD -nostdlib -fno-exceptions -felide-constructors -std=gnu++0x -fno-rtti -mthumb -mcpu=cortex-m4 -D__MK20DX256__ -DTEENSYDUINO=122 -DARDUINO=10603 -DF_CPU=96000000 -DARDUINO_ARCH_AVR -DUSB_SERIAL -DLAYOUT_US_ENGLISH -IC:\Users\Frank\Arduino\hardware\teensy\avr\cores\teensy3 -IC:\Users\Frank\Arduino\Arduino\libraries\USB_Host_Shield_20 -IC:\Users\Frank\Arduino\Arduino\libraries\spi4teensy3 -IC:\Users\Frank\Arduino\hardware\teensy\avr\libraries\SPI C:\Users\Frank\AppData\Local\Temp\build3323459009749724220.tmp\USBHIDBootKbdAndMouse.cpp -o C:\Users\Frank\AppData\Local\Temp\build3323459009749724220.tmp\USBHIDBootKbdAndMouse.cpp.o  USBHIDBootKbdAndMouse.ino:1:21: fatal error: hidboot.h: No such file or directory
compilation terminated.
Error compiling.

xxxajk, 2015-04-20:
I still have to try this myself on 1.6.3. I know for a fact that up to 1.6.2 is fine, but then again, I don't use windows either, I use Linux. I'll give it a shot later this afternoon.

frankstripod, 2015-04-20:
Thank you so much for helping!!!
I tested all of these on a Win7 laptop, some tested on an Ubuntu14.04 desktop. None of them worked :(

Teensyduino1.22 for Arduino 1.0.6, 1.6.1, 1.6.2, 1.6.3
Teensyduino1.21 for Arduino 1.0.5, 1.0.6, 1.6.0, 1.6.1
Teensyduino1.20 for Arduino 1.0.5, 1.0.6

One thing they have in common is everything was recently downloaded; Arduino.cc,Teensyduino, USB_Host_Shield_20, spi4teensy3, 49-teensy.rules.
@xxxajk spi4teensy3 worked on every one! Nice work.
I will try again later today with 1.6.2 and 1.22 on my Ubuntu PC to make sure I didn't miss that combination.

Lauszus, 2015-04-20:
Thanks for reporting. The commit: https://github.com/felis/USB_Host_Shield_2.0/commit/0ef5fa5357e9e152b2fd74644475281ac9515738 should solve this issue.

closed 2015-04-20 by Lauszus

frankstripod, 2015-04-21:
My Circuits@Home board is working :):):) Thank you!!!

I'm not requesting help with anything below, I just wanted to know if this is normal. If this is typical for a Teensy 3.1 setup, then leaving this as solved / closed gives me something to play with.

Results almost all identical with Win7 and Ubuntu 14.04: Arduino 1.6.3 and 1.6.2 with Teensyduino 1.22; 1.6.1 with 1.22 & 1.21.
board_qc: “All tests passed”
USBHIDBootMouse: works perfect (didn't expect this to work the best.)
USBHIDBootKbdAndMouse: Keyboard works great, But no mouse.
USBFTDILoopback: serial monitor prints ….....
USBHIDBootKbd: Serial monitor reads “Start”, then nothing.
USBHID_desc: Serial monitor reads “Start”, then nothing.
USB_desc: has a compiling error in the code (dissapointing.)
HUB_demo: has a compiling error in the code.

May be just my pc's, but with 1.6.3:
Ubuntu; Teensyduino has a hanging double popup window.
Win7; COM port seemed more stubborn, but not impossible.

FYI : I connected the Teensy 3.1 to UHSmini (Vbus jumper cut, 100uF cap, all 3V) to 4port Hub w/ 5V supply. This works perfect with USBHIDBootMouse.ino, and I was able to hot swap three different mice :)

Thank you again @Lauszus

Lauszus, 2015-04-22:
I have now fixed the issue in USB_desc.ino and hub_demo.ino that caused the issue with the Teensy 3.1. Please redownload the library.

In the future please post the relevant error output here on Github.

commit 0ef5fa53 (2015-04-20): Always prioritise USING_SPI4TEENSY3 if set to 1

Fixes: #147
commit 5053e8df (2015-04-22): Don't use deprecated prog_ macro

Fixes: #147