felis/USB_Host_Shield_2.0 #148: Wireless Xbox360 controller button click problem!!!

AqmalGit, 2015-05-02:
I've managed to connect and control my robot car using the analog button (RT and LT) and the analog stick. I'm working on to make the robot move as I hold a button. Eg : Button A, button LB. The problem is that the arduino only read once even though I was holding the button (I use the example provided). Unlike the analog button, the arduino will continuously read even if the value is the same.

Any fix or tips?

[image]

AqmalGit, 2015-05-02:
As you can see, I've managed to get the RT and RB working. I'm trying to make the robot move as I command to if I click or hold the LB button. It just wont work. Searched everywhere but all I found was push button thing. Need help. URGENT!!!

Lauszus, 2015-05-02:
Simply use:
```
if (Xbox.getButtonPress(L1, i)) {
```

closed 2015-05-02 by Lauszus

AqmalGit, 2015-05-02:
Thanks a lot!! I slept late for 2 days just to figure this out! I taught the press is only for analog button :)

Lauszus, 2015-05-03:
You are welcome. By the way documentation for the different libraries is located here: http://felis.github.io/USB_Host_Shield_2.0/.

For instance here is the documentation for the XBOXUSB library: http://felis.github.io/USB_Host_Shield_2.0/class_x_b_o_x_u_s_b.html.