felis/USB_Host_Shield_2.0 #149: Problem trying to control LED using Xbox 360 wireless controller!!!! HELP!!!

AqmalGit, 2015-05-04:
I've trying to make an LED to light up if I click button B. But it won't work. The LED kept light up. I don't know whats the problem..
[image]
[image]

These are samples of  my codes. Both didn't work, both gave the same result, kept light up!!
NEED HELP!!!

Lauszus, 2015-05-04:
You can not use pin 50-53, as they are used for SPI communication. Please use a different pin.

closed 2015-05-04 by Lauszus

AqmalGit, 2015-05-04:
but what about the codes? Is it okay or need some improvement?

Lauszus, 2015-05-04:
Why don't you just try it and see if it works?

You can just do it as simple as this:
```C++
/*
 Example sketch for the Xbox Wireless Reciver library - developed by Kristian Lauszus
 It supports up to four controllers wirelessly
 For more information see the blog post: http://blog.tkjelectronics.dk/2012/12/xbox-360-receiver-added-to-the-usb-host-library/ or
 send me an e-mail:  kristianl@tkjelectronics.com
 */

#include <XBOXRECV.h>

// Satisfy the IDE, which needs to see the include statment in the ino too.
#ifdef dobogusinclude
#include <spi4teensy3.h>
#include <SPI.h>
#endif

USB Usb;
XBOXRECV Xbox(&Usb);

#define LED LED_BUILTIN

void setup() {
  pinMode(LED, OUTPUT);

  Serial.begin(115200);
#if !defined(__MIPSEL__)
  while (!Serial); // Wait for serial port to connect - used on Leonardo, Teensy and other boards with built-in USB CDC serial connection
#endif
  if (Usb.Init() == -1) {
    Serial.print(F("\r\nOSC did not start"));
    while (1); //halt
  }
  Serial.print(F("\r\nXbox Wireless Receiver Library Started"));
}
void loop() {
  Usb.Task();
  if (Xbox.XboxReceiverConnected) {
    for (uint8_t i = 0; i < 4; i++) {
      if (Xbox.Xbox360Connected[i])
        digitalWrite(LED, Xbox.getButtonPress(B, i));
    }
  } else
    digitalWrite(LED, LOW);
}
```

PS: I saw your comment at my blog: http://blog.tkjelectronics.dk/2012/12/xbox-360-receiver-added-to-the-usb-host-library/comment-page-2/#comment-545324, please don't double post the same question.

AqmalGit, 2015-05-05:
Yeah, sorry about that. Just want to find the answers quickly.
The code does work, but I need to hold the button to make it light up. I've tried to change from 'getButtonPress' to 'getButtonClick'. It'll also work, but only lights up as I click the button, then lights off.

Still searching for an answers..

Lauszus, 2015-05-05:
Okay. This is not really related to the library though. I recommend asking at something like the Arduino forum for general programming questions.

Lauszus, 2015-05-05:
Anyway here is the code:
```C++
bool ledState = false;
void loop() {
  Usb.Task();
  if (Xbox.XboxReceiverConnected) {
    for (uint8_t i = 0; i < 4; i++) {
      if (Xbox.Xbox360Connected[i]) {
        if (Xbox.getButtonClick(B, i))
          ledState = !ledState;
      }
    }
  } else
    ledState = false;
  digitalWrite(LED, ledState);
}
```