felis/USB_Host_Shield_2.0 #150: OSCOKIRQ failed to assert on 1.6.2/3/4

r2DoesInc, 2015-05-15:
Using the USB Host 2.0 library BlinkLED example - and anything that uses this library in general - fails on 1.6.2, 1.6.3, 1.6.4. After reverting to 1.6.1, all works as expected.

Using the Arduino Mega ADK board with no shield. I have tried versions of the library as far back as 
42a26346fa7b552f7a271ce2d276b4fe669b70e4 with no luck.

As soon as I reverted the ide to 1.6.1, all versions of the library began working.

Lauszus, 2015-05-15:
Are you talking about this example: https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/adk/ArduinoBlinkLED/ArduinoBlinkLED.ino?

r2DoesInc, 2015-05-15:
Correct. I just mention that I used the example because it eliminates the possibility of an issue in my code. Ive also used the sketch here https://github.com/cloudspace/ArdrobotAndroid/blob/master/Resources/Arduino/motor_controller/motor_controller.ino

Lauszus, 2015-05-16:
@r2DoesInc I just tried the newest version of the library: https://github.com/felis/USB_Host_Shield_2.0/releases/tag/1.1.0 and it works fine on my Mega ADK. Can you try to re-download the newest code and try again.

For some reason the Arduino library manager is still using 1.0.0: http://downloads.arduino.cc/libraries/library_index.json and I suspect that this issue might be related to: https://github.com/felis/USB_Host_Shield_2.0/commit/5efed1938a57c4c60579b952d30cb28e68561ad1.

r2DoesInc, 2015-06-03:
Ya, that seems to have been the problem. 

After manually pulling in the library it seems to be working.

closed 2015-06-03 by r2DoesInc