felis/USB_Host_Shield_2.0 #158: Send data issue

ahmed-belal, 2015-06-20:
My application is set up to continuously send a constant string to the Android phone when it is connected in peripheral mode. However for some reason the first time the during teaches the phone correctly but all subsequent transmissions, although they report correct number of bytes read, have byte values 0x00 for each byte. Any ideas what might be going wrong?

felis, 2015-06-20:
Perhaps, you have an error in your code?

ahmed-belal, 2015-06-20:
Tried paying the same SndData line twice. The first call gets data through
and all subsequent calls deliver bytes 0x00

Lauszus, 2015-06-20:
Have you tried the following example: https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/adk/ArduinoBlinkLED/ArduinoBlinkLED.ino and the Android app that is meant for it: https://github.com/Lauszus/ArduinoBlinkLED?

closed 2015-06-27 by ahmed-belal