felis/USB_Host_Shield_2.0 #159: Phone reconnection issue

ahmed-belal, 2015-06-20:
I have a host shield which I'm pretty sure it's not the official shield. I don't have access to a sparkfun shield either. Anyway whenever I connect the Android phone to my Uno via the shield, it connects properly. If I disconnect and reconnect it does not communicate and adk reports device isn't ready. The phone doesn't connect again until the board is reset.

Lauszus, 2015-06-20:
Have you tried to connect an external power supply to the Arduino? Also it might be a problem with the shield, as the Chinese clones seems to have all kind of weird issues.

ahmed-belal, 2015-06-27:
@Lauszus there isn't power issues, I've connected 9v and 12v adapters at times, the arduino side does not brown out or reset on connection or reconnection. It always defaults to error E0: Additional_Reset_Required for any connection after the first time I disconnect. Any ideas?

ahmed-belal, 2015-06-27:
@Lauszus also the board passed all tests, runs all hr examples properly but has the same reconnection issue on blink led example.

Lauszus, 2015-07-01:
You should try with the official shield, as it works fine on my setup.

Lauszus, 2015-07-01:
Also have you tried another phone?

closed 2015-07-01 by Lauszus

ahmed-belal, 2015-07-02:
@Lauszus I have tried with the official shield as well and I still have this issue.

Lauszus, 2015-07-03:
@ahmed-belal have you tried another Android phone?