felis/USB_Host_Shield_2.0 #160: Problem with keyboard and lock keys

custodium, 2015-06-30:
Hello,

I've a problem with a Keyboard and USB Host Shield libraries, when I press any lock key (CapsLock, NumLock or ScrollLock) the led came on, but if I press again it does not came off, it only came off when I press any other lock key.

Best regards.

Lauszus, 2015-07-01:
Please describe your setup. What development board are you using, keyboard etc.

gorakk, 2015-08-27:
Hi there,

I happened to notice this issue yesterday - pressing a lock key will turn on the appropriate LED on the keyboard - but another press will not turn it off.  Also - multiple LEDs cannot be turned on.  If NumLock is pressed the NumLock LED will turn on, if CapsLock is then pressed the NumLock LED will turn off and the CapsLock LED will turn on.

In hidboot.h starting at line 166 - it appears that SetReport is clearing kbdLockingKeys.bLeds - here is the original:

```
  if(old_keys != kbdLockingKeys.bLeds && hid)
    return (hid->SetReport(0, 0/*hid->GetIface()*/, 2, 0, 1, &kbdLockingKeys.bLeds));
```

My quick and dirty fix was to add a temp variable.  I haven't dug further into the code so I'm not sure if there is some reason for kbdLockingKeys.bLeds to be cleared.  But with this change the LEDs will toggle properly and multiple LEDs will light.

```
  if(old_keys != kbdLockingKeys.bLeds && hid)
  {
    uint8_t lockLeds = kbdLockingKeys.bLeds;  // added

    return (hid->SetReport(0, 0/*hid->GetIface()*/, 2, 0, 1, &lockLeds));  // changed to use the temp variable
  }
```

Lauszus, 2015-08-30:
@gorakk thanks. Do you mind sending a pull request?

Lauszus, 2015-09-03:
https://github.com/felis/USB_Host_Shield_2.0/pull/172 should fix this issue.

closed 2015-09-03 by Lauszus

lnxsrt, 2020-10-21:
I am still having an issue with keyboard lock LEDs inconsistently lighting. The `kbdLockingKeys.bLeds` variable is consistent and updating appropriately. However, I think the `SetReport` call may not be working correctly. It looks like it occasionally sends an incorrect value causing all of the LEDs to turn off. I am just using the default USBHIDBootKbd.ino code with a standard wired keyboard. Everything else works well so I don't suspect a physical setup issue.

DanielGibson, 2021-01-01:
@lnxsrt: try using the latest code from git instead of version 1.3.2 from the Arduino library manager - seems like this issue has been fixed since then (see also https://github.com/felis/USB_Host_Shield_2.0/issues/577)