felis/USB_Host_Shield_2.0 #161: USB Host Shield and Teensy 2++

custodium, 2015-07-01:
Hello,

I have a Teensy 2++ and USB Host Shield for Arduino Pro Mini, like in this tutorial: https://www.pjrc.com/teensy/td_libs_USBHostShield.html I wired :
INT pin9
SS pin10
MOSI pin22
MISO pin23
SCK pin21
RESET vcc (3.3v)
3.3v vcc (3.3v)
GND GND

Also I'm using the latest USBHostShield2 libraries downloaded from github and Arduino IDE 1.6.1, when I load the board_qc example i only obtain this:
Circuits At Home 2011
USB Host Shield Quality Control Routine
Reading REVISION register... Die revision invalid. Value returned: 00
Unrecoverable error - test halted!!
0x55 pattern is transmitted via SPI
Press RESET to restart test

can somebody help me?

Best regards

custodium, 2015-07-01:
Solved, SS must be wired to pin9 and INT must be wired to pin8.

closed 2015-07-01 by custodium

Lauszus, 2015-07-01:
Good to hear you got it working :)