felis/USB_Host_Shield_2.0 #162: Xbox One and Xbox 360 controllers turn off

gigawatts, 2015-07-07:
Using a sparkfun RedBoard (UNO clone) and a sparkfun USB Host shield, the offcial Xbox One controller will turn off after a few seconds of the sketch restarting if no input is given to the controller. This is using the un-modified XBOXONE example, with an external 12V power source powering the arduino. If I try to power the arduino and controller from just USB, the controller turns on long enough to rumble, then turns off before input can be given.

On the un-modified XBOXUSB example, an official wired Xbox 360 controller behaves very similarly.

Also, on both examples, the F() function needs to be removed from every Serial.println() in order to compile the sketch without "reinterpret_cast from type 'const char*' to type '__FlashStringHelper*' casts away constness" errors. This is using Arduino IDE 1.0.5 on Windows.  Trying to compile the XBOXONE example with version 1.6.4 on Windows gives the following error "libraries\USB_Host_Shield_2_0/settings.h:133:176: fatal error: SPI.h: No such file or directory"

Lauszus, 2015-07-07:
You can not just use a hub instance, please see: https://github.com/felis/USB_Host_Shield_2.0#faq.

It does indeed sound like your controllers are pulling too much current for your power supply to handle. Please try the hub again and report back.

The problem with the missing SPI library has now been resolved: https://github.com/felis/USB_Host_Shield_2.0/commit/607932b437161bb20e284d4d05df0740d3ce591d. Please redownload the library and try again.

Lauszus, 2016-02-25:
Closing due to lack of feedback.

closed 2016-02-25 by Lauszus