felis/USB_Host_Shield_2.0 #163: USB Host Shield 2.0 Library fails to compile

elisidir, 2015-07-07:
On 7/5/2015 5:07 PM, Sam Denard wrote:
> I downloaded and manually installed the USB Host Library 2.0 on 
> 7/5/15.  I downloaded from 
> https://github.com/felis/USB_Host_Shield_2.0.   It appears to have 
> installed successfully: the Sketch > Include Library > Manage 
> Libraries dialog shows the library as "Installed".  I have attempted 
> to compile several of the library's examples.  So far, every attempt 
> has failed; i.e., each compilation begins and then hangs without error 
> messages.
>
> I uninstalled and re-installed the USB Host Library 2.0; the examples 
> still fail to compile.  Examples from other libraries compile 
> successfully, as does my own code.
>
> I am using Arduino IDE 1.6.4.  The compilation target is an Arduino 
> Mega 2560 on COM10.
>
> Would you please help me understand what is going wrong and how to fix 
> it?

Thank you, Oleg, for your quick response.  As you requested, I've attached a screenshot of the Arduino IDE during an attempt to compile le3dp.ino.  As you can see, the process hangs; other examples also hang.  In my own code, this consistently happens when there are spurious characters in a file or when I mis-match data types and an implicit conversion fails.  Please let me know what else I can do to help figure this out.

[image]

Lauszus, 2015-07-07:
This sounds more line an issue with the Arduino IDE and not the library. Can you try it on another computer and see if it works there?

elisidir, 2015-07-08:
Other libraries and my own code run fine on the current computer; but I will try a different one.  However, I will be concerned if the USB Host Library's compilation is machine-dependent.

Lauszus, 2015-07-08:
The library is not machine-dependent. Again it looks like a bug in the IDE and not in the library.

xxxajk, 2015-07-08:
Does the compile fail with 1.6.5? 1.6.4 had a lot of nasty bugs and did not last long before it was replaced.

elisidir, 2015-07-10:
Thank you, xxajk.  With version 1.6.5, the Library compiled successfully with nothing more than warnings about unused parameters.  I will keep a closer eye on the Arduino site in the future.

And, I appreciate your help also, Lauszus.

closed 2015-07-10 by elisidir

Lauszus, 2015-07-11:
@elisidir good to hear you got it working :+1: