felis/USB_Host_Shield_2.0 #164: USB device error

benbiles, 2015-07-09:
Hi, when I connect a novation launchcontrol USB midi device I get the error code below in the console with USB_dec in example code. I have enabled debugging with #define ENABLE_UHS_DEBUGGING 1 in settings.h

Keyboard and Mouse example both work.

Any advise as to what to try next ? I don't think its lack of power as I am connected wth 12v power supply on arduino uno R3. 

Start

01
--

Request error. Error code:	0D

Addr:1(0.0.1)

benbiles, 2015-07-09:
OK, sorry, I added USB hub with 
#include <usbhub.h>

USB  Usb;
USBHub  Hub1(&Usb);

And now it appears its working. sorry, it must have been that the arduino could not provide enough power for the device even though it was plugged into the 12v adapter !!!

xxxajk, 2015-07-09:
Yeh, powered hub is needed for many devices since the arduino regulator is limited to something like 800ma -- and remember that has to power everything else too.

Lauszus, 2015-07-11:
Good to hear you got it working @benbiles :)

closed 2015-07-11 by Lauszus