felis/USB_Host_Shield_2.0 #166: XBox 360 Stops Responding for Short Periods

benahrens, 2015-07-10:
### Intro 
I'm using an [XBox 360 (Microsoft brand wired USB)](http://www.amazon.com/Microsoft-Xbox-360-Wired-Controller/dp/B003ZSN600) with an [Arduino Mega 2560](https://www.arduino.cc/en/Main/ArduinoBoardMega2560) and [Arduino USB Host Shield](https://www.arduino.cc/en/Main/ArduinoUSBHostShield). The XBox input will be translated by the Arduino into PWM control of some hydraulic valves. Full code is listed below.

I currently have LEDs attached to the outputs as loads to easily show what the PWM is doing. My code basically works as expected with one very important catch.

### The Problem
After anywhere from a couple seconds to 10-20 seconds, the LEDs stop responding to control from the XBox 360 controller. Based on debug print statements I added (not in the code below) I know the loop keeps going and `Xbox.Xbox360Connected = true`, but that `LeftHatXCurr = Xbox.getAnalogHat(LeftHatX);` is stuck on the same value instead of reflecting the way I manipulate the left hat. This continues for 3-10 seconds, sometimes longer, and then without my resetting or doing anything, everything works as expected again. It's possible I'm doing (or failing to do) something to cause this, so please take a look at my code and let me know whether I'm at fault or maybe there's something else at play.

### The Code
```c
#include <XBOXUSB.h>

// Satisfy the IDE, which needs to see the include statment in the ino too.
#ifdef dobogusinclude
#include <spi4teensy3.h>
#include <SPI.h>
#endif

USB Usb;
XBOXUSB Xbox(&Usb);
int16_t LeftHatXPrev, LeftHatYPrev, RightHatXPrev, RightHatYPrev;
int16_t LeftHatXCurr, LeftHatYCurr, RightHatXCurr, RightHatYCurr;
int16_t AnalogHatZeroThresh;
int16_t AnalogHatMoveThresh;

void setup() {
  AnalogHatMoveThresh = 500;
  AnalogHatZeroThresh = 8000;
  LeftHatXPrev = 0;
  LeftHatXCurr = 0;
  
  Serial.begin(115200);
  
#if !defined(__MIPSEL__)
  while (!Serial); // Wait for serial port to connect - used on Leonardo, Teensy and other boards with built-in USB CDC serial connection
#endif
  
  if (Usb.Init() == -1)
  {
    Serial.print(F("\r\nOSC did not start"));
    while (1); //halt
  }
  Serial.print(F("\r\nXBOX USB Library Started"));
}

void loop() {
  Usb.Task();
  if (Xbox.Xbox360Connected)
  {
    // **Left Hat X**
    checkLeftHatX();
      
    // **Left Hat Y**
    checkLeftHatY();
    
    // **Right Hat X**
    checkRightHatX();
      
    // **Right Hat Y**
    checkRightHatY();
  }
  else
  {
    Serial.print("\nNot connected");
  }
  delay(100);
}

void checkLeftHatX()
{
  // Make sure stick has moved AnalogHatMoveThresh before changing output
  LeftHatXCurr = Xbox.getAnalogHat(LeftHatX);  
  if(((LeftHatXCurr > LeftHatXPrev) && (LeftHatXCurr - LeftHatXPrev > AnalogHatMoveThresh)) || ((LeftHatXCurr < LeftHatXPrev) && (LeftHatXPrev - LeftHatXCurr > AnalogHatMoveThresh)))
  {
    if(LeftHatXCurr > AnalogHatZeroThresh)
    {
      // Extend 
      analogWrite(5, (float)LeftHatXCurr/32768*255-1);
      analogWrite(4, 0);
    }
    else if (LeftHatXCurr < -AnalogHatZeroThresh)
    {
      // Retract
      analogWrite(5, 0);
      analogWrite(4, (-((float)LeftHatXCurr)/32768)*255-1);
    }
    else
    {
      analogWrite(5,0);
      analogWrite(4,0);
    }
    
    LeftHatXPrev = LeftHatXCurr;
  }
}

void checkLeftHatY()
{
  // Make sure stick has moved AnalogHatMoveThresh before changing output
  LeftHatYCurr = Xbox.getAnalogHat(LeftHatY);
  if(((LeftHatYCurr > LeftHatYPrev) && (LeftHatYCurr - LeftHatYPrev > AnalogHatMoveThresh)) || ((LeftHatYCurr < LeftHatYPrev) && (LeftHatYPrev - LeftHatYCurr > AnalogHatMoveThresh)))
  {
    if(LeftHatYCurr > AnalogHatZeroThresh)
    {
      // Extend 
      analogWrite(6, floor((float)LeftHatYCurr/32768*255));
      analogWrite(7, 0);
    }
    else if (LeftHatYCurr < -AnalogHatZeroThresh)
    {
      // Retract
      analogWrite(6, 0);
      analogWrite(7, (-((float)LeftHatYCurr)/32768)*255);
    }
    else
    {
      analogWrite(6,0);
      analogWrite(7,0);
    }
    
    LeftHatYPrev = LeftHatYCurr;     
  }
}

void checkRightHatX()
{
  // Make sure stick has moved AnalogHatMoveThresh before changing output
  RightHatXCurr = Xbox.getAnalogHat(RightHatX);  
  if(((RightHatXCurr > RightHatXPrev) && (RightHatXCurr - RightHatXPrev > AnalogHatMoveThresh)) || ((RightHatXCurr < RightHatXPrev) && (RightHatXPrev - RightHatXCurr > AnalogHatMoveThresh)))
  {
    if(RightHatXCurr > AnalogHatZeroThresh)
    {
      // Extend 
      analogWrite(5, (float)RightHatXCurr/32768*255-1);
      analogWrite(4, 0);
    }
    else if (RightHatXCurr < -AnalogHatZeroThresh)
    {
      // Retract
      analogWrite(5, 0);
      analogWrite(4, (-((float)RightHatXCurr)/32768)*255-1);
    }
    else
    {
      analogWrite(5,0);
      analogWrite(4,0);
    }
    
    RightHatXPrev = RightHatXCurr;
  }
}

void checkRightHatY()
{
  // Make sure stick has moved AnalogHatMoveThresh before changing output
  RightHatYCurr = Xbox.getAnalogHat(RightHatY);
  if(((RightHatYCurr > RightHatYPrev) && (RightHatYCurr - RightHatYPrev > AnalogHatMoveThresh)) || ((RightHatYCurr < RightHatYPrev) && (RightHatYPrev - RightHatYCurr > AnalogHatMoveThresh)))
  {
    if(RightHatYCurr > AnalogHatZeroThresh)
    {
      // Extend 
      analogWrite(6, floor((float)RightHatYCurr/32768*255));
      analogWrite(7, 0);
    }
    else if (RightHatYCurr < -AnalogHatZeroThresh)
    {
      // Retract
      analogWrite(6, 0);
      analogWrite(7, (-((float)RightHatYCurr)/32768)*255);
    }
    else
    {
      analogWrite(6,0);
      analogWrite(7,0);
    }
    
    RightHatYPrev = RightHatYCurr;     
  }
}
```

Lauszus, 2015-07-11:
Try to remove the 100 ms delay. You have to call the Usb.Task() function approximately every 1 ms.

If you still need the 100 ms delay you can use millis/micros instead: https://www.arduino.cc/en/Tutorial/BlinkWithoutDelay.

benahrens, 2015-07-13:
Changing to 1ms delay works just fine. If you have time I'd be interested in why the 1ms is important but my problem is fixed so I will not be offended if I don't hear from you. Thanks for your time/help.

Ben

closed 2015-07-13 by benahrens

Lauszus, 2015-07-13:
The thing is that you can't read from the Xbox controller to quickly, that is why there is a delay :)

Lauszus, 2015-07-13:
But it can't be too long either as you have to read from it continuously.

xxxajk, 2015-07-13:
... and that's why the toggle was wrong? Sounds to me like a broken USB
device.

Well, at least in the 3.0 lib, you can just do this sort of thing in the
background without anything extra, since you would just schedule it to hit
every 1ms. :-)

benahrens, 2015-07-13:
@xxxajk I'm not sure what you mean by "the toggle was wrong"?

Lauszus, 2015-07-13:
@xxxajk the toggle was not wrong. It was simply reading from the input endpoint top fast, but yes 3.0 should be able to take care of it automatically.