felis/USB_Host_Shield_2.0 #167: Bug found

Lauszus, 2015-07-12:
It seems like the pull request: https://github.com/felis/USB_Host_Shield_2.0/pull/145 is causing issues for some users, as they are not able to pair with a PS3 controller.

@felis and @xxxajk can any of you comment on this? Can you guys think of a fix.

Lauszus, 2015-07-12:
I have just sent the user an email and asked him to provide some more information of the problem.

sanelss, 2015-07-12:
This is what happens with that commit(which fails to sync with controller):

PS3 Bluetooth Library Started
Bluetooth Dongle Initialized
HCI Reset complete
Write class of device
Local Bluetooth Address: 00:1B:10:00:2A:EC
Wait For Incoming Connection Request
Gamepad is connecting
Incoming Connection Request
Remote Name: PLAYSTA

which gets stuck there and after flashing the sync lights for a little bit they all turn off.

this is what happens when you undo that commit and it successfully pairs:

Gamepad is connecting
Incoming Connection Request
Remote Name: PLAYSTA
Connected to Device: 00:23:06:11:96:5F
HID Control Incoming Connection Request
HID Control Successfully Configured
HID Interrupt Incoming Connection Request
HID Interrupt Successfully Configured
Wait For Incoming Connection Request
Dualshock 3 Controller Enabled

xxxajk, 2015-07-12:
Yeah, so fix the toggle issue in your driver.

Lauszus, 2015-07-12:
Sorry Andrew, but he is using my driver and it is not setting the toggle register, but is just calling the inTransfer function, so I believe it is a bug in the Usb core.

Lauszus, 2015-07-12:
Btw @sanelss could you try to enable extra debugging as well by uncommenting this line: https://github.com/felis/USB_Host_Shield_2.0/blob/master/BTD.cpp#L20?

sanelss, 2015-07-12:
this is advanced debug when it fails:

PS3 Bluetooth Library Started
BTD ConfigureDevice
BTD Init
Addr: 01
Endpoint descriptor:
Length:		07
Type:		05
Address:	81
Attributes:	03
MaxPktSize:	0010
Poll Intrv:	01
Endpoint descriptor:
Length:		07
Type:		05
Address:	82
Attributes:	02
MaxPktSize:	0040
Poll Intrv:	01
Endpoint descriptor:
Length:		07
Type:		05
Address:	02
Attributes:	02
MaxPktSize:	0040
Poll Intrv:	01
Bluetooth Dongle Initialized
Connection established
HCI Reset complete
Write class of device
Local Bluetooth Address: 00:1B:10:00:2A:EC
Wait For Incoming Connection Request
Gamepad is connecting
Class of device: 00 05 08
Incoming Connection Request
Remote Name: PLAYSTA

sanelss, 2015-07-12:
this is advance debug when it works undoing that commit:

PS3 Bluetooth Library Started
BTD ConfigureDevice
BTD Init
Addr: 01
Endpoint descriptor:
Length:		07
Type:		05
Address:	81
Attributes:	03
MaxPktSize:	0010
Poll Intrv:	01
Endpoint descriptor:
Length:		07
Type:		05
Address:	82
Attributes:	02
MaxPktSize:	0040
Poll Intrv:	01
Endpoint descriptor:
Length:		07
Type:		05
Address:	02
Attributes:	02
MaxPktSize:	0040
Poll Intrv:	01
Bluetooth Dongle Initialized
Connection established
HCI Reset complete
Write class of device
Local Bluetooth Address: 00:1B:10:00:2A:EC
Wait For Incoming Connection Request
Gamepad is connecting
Class of device: 00 05 08
Incoming Connection Request
Remote Name: PLAYSTA
Unmanaged HCI Event: 6C
Unmanaged HCI Event: 6C
Connection established
Connected to Device: 00:23:06:11:96:5F
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
HID Control Incoming Connection Request
Connection established
Connection established
Connection established
HID Control Successfully Configured
HID Interrupt Incoming Connection Request
HID Interrupt Successfully Configured
Wait For Incoming Connection Request
Dualshock 3 Controller Enabled

Cross
Circle

Lauszus, 2015-07-12:
When it is connected does the joysticks work?

sanelss, 2015-07-12:
yes everything works when it connects

xxxajk, 2015-07-12:
sanelss: can you put in a line that tells us if there is a toggle error? if there is a toggle error then the toggle was not set up correctly. (yes really). 
The commit actually is a bugfix for code bugs that used to happen a long time ago, but shouldn't happen anymore. If they are still happening then the actual driver is really at fault.

sanelss, 2015-07-12:
"can you put in a line that tells us if there is a toggle error?"

what line where? I'm not familiar at all with the library.

Lauszus, 2015-07-12:
Hmm for some reason the full device name is cut short. This usually means that the dongle does not support 2.0+EDR, but then the bytes that got the joystick and accelerometer data will also be cut short. Have you tried another dongle?

And yes as @xxxajk suggested please add some debug info at the line, so we can see if it and when it is run.

Lauszus, 2015-07-12:
@sanelss after this line: https://github.com/felis/USB_Host_Shield_2.0/pull/145/files#diff-f382db0d87969c32dd22c1c4116009e1R237

xxxajk, 2015-07-12:
Before the line that was replaced in commit #145

xxxajk, 2015-07-12:
or after...  LOL

sanelss, 2015-07-12:
I added this to the file:

Serial.print("Toggle bit pre command: ");
Serial.println(pep->bmRcvToggle);
//pep->bmRcvToggle = (regRd(rHRSL) & bmSNDTOGRD) ? 0 : 1;	//works
pep->bmRcvToggle = (regRd(rHRSL) & bmRCVTOGRD) ? 0 : 1;		//fails
Serial.print("Toggle bit post command: ");
Serial.println(pep->bmRcvToggle);

sanelss, 2015-07-12:
output when it fails:

PS3 Bluetooth Library Started
BTD ConfigureDevice
BTD Init
Addr: 01
Endpoint descriptor:
Length:		07
Type:		05
Address:	81
Attributes:	03
MaxPktSize:	0010
Poll Intrv:	01
Endpoint descriptor:
Length:		07
Type:		05
Address:	82
Attributes:	02
MaxPktSize:	0040
Poll Intrv:	01
Endpoint descriptor:
Length:		07
Type:		05
Address:	02
Attributes:	02
MaxPktSize:	0040
Poll Intrv:	01
Bluetooth Dongle Initialized
Connection established
HCI Reset complete
Write class of device
Local Bluetooth Address: 00:1B:10:00:2A:EC
Wait For Incoming Connection Request
Gamepad is connecting
Class of device: 00 05 08
Incoming Connection Request
Remote Name: PLAYSTAToggle bit pre command: 1
Toggle bit post command: 0
Toggle bit pre command: 0
Toggle bit post command: 1
Toggle bit pre command: 1
Toggle bit post command: 0
Toggle bit pre command: 0
Toggle bit post command: 1
Toggle bit pre command: 1
Toggle bit post command: 0
Toggle bit pre command: 0
Toggle bit post command: 1
Toggle bit pre command: 1
Toggle bit post command: 0
Toggle bit pre command: 0
Toggle bit post command: 1
Toggle bit pre command: 1
Toggle bit post command: 0
Toggle bit pre command: 0
Toggle bit post command: 1
Toggle bit pre command: 1
Toggle bit post command: 0
Toggle bit pre command: 0
Toggle bit post command: 1
Toggle bit pre command: 1
Toggle bit post command: 0
Toggle bit pre command: 0
Toggle bit post command: 1
Toggle bit pre command: 1
Toggle bit post command: 0
Toggle bit pre command: 0
Toggle bit post command: 1
Toggle bit pre command: 1
Toggle bit post command: 0
Toggle bit pre command: 0
Toggle bit post command: 1
Toggle bit pre command: 1
Toggle bit post command: 0
Toggle bit pre command: 0
Toggle bit post command: 1
Toggle bit pre command: 1
Toggle bit post command: 0
Toggle bit pre command: 0
Toggle bit post command: 1
Toggle bit pre command: 1
Toggle bit post command: 0
Toggle bit pre command: 0
Toggle bit post command: 1
Toggle bit pre command: 1
Toggle bit post command: 0
Toggle bit pre command: 0
Toggle bit post command: 1
Toggle bit pre command: 1
Toggle bit post command: 0
Toggle bit pre command: 0
Toggle bit post command: 1
Toggle bit pre command: 1
Toggle bit post command: 0
Toggle bit pre command: 0
Toggle bit post command: 1

sanelss, 2015-07-12:
when it's trying to connect, every few seconds it would run that bit of code and the serial monitor would show an update until it eventually stops trying and the controller lights turn off

Lauszus, 2015-07-12:
What happens if you comment out both?
```
//pep->bmRcvToggle = (regRd(rHRSL) & bmSNDTOGRD) ? 0 : 1; //works
pep->bmRcvToggle = (regRd(rHRSL) & bmRCVTOGRD) ? 0 : 1; //fails
```

sanelss, 2015-07-12:
This is what happens when the working line is enabled:

PS3 Bluetooth Library Started
BTD ConfigureDevice
BTD Init
Addr: 01
Endpoint descriptor:
Length:		07
Type:		05
Address:	81
Attributes:	03
MaxPktSize:	0010
Poll Intrv:	01
Endpoint descriptor:
Length:		07
Type:		05
Address:	82
Attributes:	02
MaxPktSize:	0040
Poll Intrv:	01
Endpoint descriptor:
Length:		07
Type:		05
Address:	02
Attributes:	02
MaxPktSize:	0040
Poll Intrv:	01
Bluetooth Dongle Initialized
Connection established
HCI Reset complete
Write class of device
Local Bluetooth Address: 00:1B:10:00:2A:EC
Wait For Incoming Connection Request
Gamepad is connecting
Class of device: 00 05 08
Incoming Connection Request
Remote Name: PLAYSTAToggle bit pre command: 1
Toggle bit post command: 1

Unmanaged HCI Event: 6CToggle bit pre command: 1
Toggle bit post command: 1

Unmanaged HCI Event: 6CToggle bit pre command: 1
Toggle bit post command: 1
Toggle bit pre command: 1
Toggle bit post command: 1
Toggle bit pre command: 1
Toggle bit post command: 1
Toggle bit pre command: 1
Toggle bit post command: 1
Toggle bit pre command: 1
Toggle bit post command: 1
Toggle bit pre command: 1
Toggle bit post command: 1

Connection established
Connected to Device: 00:23:06:11:96:5F
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
HID Control Incoming Connection Request
Connection established
Connection established
Connection established
HID Control Successfully Configured
HID Interrupt Incoming Connection Request
HID Interrupt Successfully Configured
Wait For Incoming Connection Request
Dualshock 3 Controller Enabled

LeftHatX: 141	LeftHatY: 122	RightHatX: 132	RightHatY: 126
LeftHatX: 141	LeftHatY: 122	RightHatX: 132	RightHatY: 126

sanelss, 2015-07-12:
It also works when both lines are commented out:

PS3 Bluetooth Library Started
BTD ConfigureDevice
BTD Init
Addr: 01
Endpoint descriptor:
Length:		07
Type:		05
Address:	81
Attributes:	03
MaxPktSize:	0010
Poll Intrv:	01
Endpoint descriptor:
Length:		07
Type:		05
Address:	82
Attributes:	02
MaxPktSize:	0040
Poll Intrv:	01
Endpoint descriptor:
Length:		07
Type:		05
Address:	02
Attributes:	02
MaxPktSize:	0040
Poll Intrv:	01
Bluetooth Dongle Initialized
Connection established
HCI Reset complete
Write class of device
Local Bluetooth Address: 00:1B:10:00:2A:EC
Wait For Incoming Connection Request
Gamepad is connecting
Class of device: 00 05 08
Incoming Connection Request
Remote Name: PLAYSTAToggle bit pre command: 1
Toggle bit post command: 1

Unmanaged HCI Event: 6CToggle bit pre command: 1
Toggle bit post command: 1

Unmanaged HCI Event: 6CToggle bit pre command: 1
Toggle bit post command: 1
Toggle bit pre command: 1
Toggle bit post command: 1
Toggle bit pre command: 1
Toggle bit post command: 1
Toggle bit pre command: 1
Toggle bit post command: 1
Toggle bit pre command: 1
Toggle bit post command: 1
Toggle bit pre command: 1
Toggle bit post command: 1

Connection established
Connected to Device: 00:23:06:11:96:5F
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
HID Control Incoming Connection Request
Connection established
Connection established
Connection established
HID Control Successfully Configured
HID Interrupt Incoming Connection Request
HID Interrupt Successfully Configured
Wait For Incoming Connection Request
Dualshock 3 Controller Enabled

LeftHatX: 138	LeftHatY: 126	RightHatX: 132	RightHatY: 126
LeftHatX: 138	LeftHatY: 126	RightHatX: 132	RightHatY: 126
LeftHatX: 138	LeftHatY: 126	RightHatX: 132	RightHatY: 126

xxxajk, 2015-07-12:
Toggle errors should not happen if toggle is correctly handled, and if the device isn't horked.
Obviously the toggle is wrong, and needs to be set. 
Where, exactly, I do not know. I don't own that controller.
That's all the advice I can offer.

Lauszus, 2015-07-12:
Hmm can you try to add a 1 ms delay before this line: https://github.com/felis/USB_Host_Shield_2.0/blob/master/Usb.cpp#L234?

sanelss, 2015-07-12:
adding that delay fixed the problem:

Serial.print("Toggle bit pre command: ");
Serial.println(pep->bmRcvToggle);
//pep->bmRcvToggle = (regRd(rHRSL) & bmSNDTOGRD) ? 0 : 1;	//works
pep->bmRcvToggle = (regRd(rHRSL) & bmRCVTOGRD) ? 0 : 1;		//fails
Serial.print("Toggle bit post command: ");
Serial.println(pep->bmRcvToggle);

PS3 Bluetooth Library Started
BTD ConfigureDevice
BTD Init
Addr: 01
Endpoint descriptor:
Length:		07
Type:		05
Address:	81
Attributes:	03
MaxPktSize:	0010
Poll Intrv:	01
Endpoint descriptor:
Length:		07
Type:		05
Address:	82
Attributes:	02
MaxPktSize:	0040
Poll Intrv:	01
Endpoint descriptor:
Length:		07
Type:		05
Address:	02
Attributes:	02
MaxPktSize:	0040
Poll Intrv:	01
Bluetooth Dongle Initialized
HCI Reset complete
Write class of device
Local Bluetooth Address: 00:1B:10:00:2A:EC
Wait For Incoming Connection Request
Gamepad is connecting
Class of device: 00 05 08
Incoming Connection Request
Remote Name: PLAYSTATION(R)3 Controller
Connection established
Connected to Device: 00:23:06:11:96:5F
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
HID Control Incoming Connection Request
Connection established
HID Control Successfully Configured
HID Interrupt Incoming Connection Request
HID Interrupt Successfully Configured
Wait For Incoming Connection Request
Dualshock 3 Controller Enabled

LeftHatX: 139	LeftHatY: 126	RightHatX: 132	RightHatY: 125
LeftHatX: 139	LeftHatY: 126	RightHatX: 132	RightHatY: 125
LeftHatX: 139	LeftHatY: 126	RightHatX: 132	RightHatY: 125

Lauszus, 2015-07-12:
Okay. The reason is that it is trying to read from the device to fast if we are trying to read more than the maximum bytes of the endpoint. I will work on a fix. Thanks a lot for your feedback!

sanelss, 2015-07-12:
No problem. As noted I could run an old version but hope this will helps others who may have issues. I'm sure you'll hear from me again at a later time xD This is the third time my setup didn't get along with the current library version and needed your attention =p

Lauszus, 2015-07-12:
Can you add the following code below this line: https://github.com/felis/USB_Host_Shield_2.0/blob/master/Usb.cpp#L283.
```C++
 else
    delayMicroseconds(1000);
```
Then lower the value until it stops working.

Lauszus, 2015-07-12:
@xxxajk it might be better to have the inTransfer take the bInterval as an argument, so it can delay the appropriate time between transfers.

sanelss, 2015-07-12:
300us and higher it works.
200us and lower it fails

the timing issue may be due to the fact i'm running a teensy 3.0 at a much higher clock rate than most typical arduinos.

Lauszus, 2015-07-12:
Yes that could differently be the issue. Anyway this needs to be fixed properly. I will come up with a fix ;)

Lauszus, 2015-07-12:
@sanelss please try: https://github.com/felis/USB_Host_Shield_2.0/pull/168.

sanelss, 2015-07-12:
#168 fixes the issue

Lauszus, 2015-07-12:
@sanelss thank you for your feedback. I will merge it now then!

Lauszus, 2015-07-12:
Hmm I will wait for the Travis build to finish first ;)

PR #168 (2015-07-12): Add delay between multiple calls to dispatchpkt inside InTransfer
Fixes: #167
commit e7bb5faf (2015-07-12): Add delay between multiple calls to dispatchpkt inside InTransfer

Fixes: #167
closed 2015-07-12 by Lauszus