felis/USB_Host_Shield_2.0 #169: HCI Command Failed: 0B  - wiimote

moccault, 2015-07-13:
Hi,

I'm trying to connect my wiimote to arduino. I get this message:

[image]

Can't find the answer on the internet.

Thanks!

Michael

Lauszus, 2015-07-13:
Try to connect an external power supply, but to be honest it looks like a problem with your dongle.

moccault, 2015-07-14:
Hi,

Thanks for the fast reply!

I have 2 different bt dongles. They both work for the ps3 controller.
I powered my arduino with an external power supply.

Still I get the same message.

Regards

Michael

:

Lauszus, 2015-07-14:
Hmm it looks like the name is not read correctly. I believe https://github.com/felis/USB_Host_Shield_2.0/pull/168 fixes this, so can you download the newest version of the library and try again?

moccault, 2015-07-14:
Hi,

I only tested it with the latest version. To be sure I downloaded the
latest version again, still the same message.

Thanks,

Michael

:

Lauszus, 2015-07-14:
Okay. Can you try another dongle then?

HavenLau626, 2015-07-20:
Hi all,
I'm having the same issue as OP, 
I've tried powering the board with an external power supply and 3 different bluetooth dongles, I'm pretty sure the dongles work with my wiimote as I've had success with them pairing to my raspberry pi and reading values from it. 

Here is the serial monitor output:
Wiimote Bluetooth Library Started
BTD ConfigureDevice
BTD Init
Addr: 01
Endpoint descriptor:
Length:		07
Type:		05
Address:	81
Attributes:	03
MaxPktSize:	0010
Poll Intrv:	01
Endpoint descriptor:
Length:		07
Type:		05
Address:	82
Attributes:	02
MaxPktSize:	0040
Poll Intrv:	00
Endpoint descriptor:
Length:		07
Type:		05
Address:	02
Attributes:	02
MaxPktSize:	0040
Poll Intrv:	00
Bluetooth Dongle Initialized
No response to HCI Reset
HCI Reset complete
Write class of device
Local Bluetooth Address: 00:15:83:0C:BF:EB
Starting inquiry
Press 1 & 2 on the Wiimote
Or press the SYNC button if you are using a Wii U Pro Controller or a Wii Balance Board
Number of responses: 1
Class of device: 00 25 04
Wiimote found
Now just create the instance like so:
WII Wii(&Btd);
And then press any button on the Wiimote
Remote Name: Nintendo RVL-CNT-01
Wiimote is connecting
Connecting to Wiimote
HCI Command Failed: 0B
HCI Command Failed: 0B
HCI Command Failed: 0B
...

Thanks a lot.

Lauszus, 2015-07-28:
Unfortunately there is not much I can do about this. Please try one of the dongles on the Wiki list: https://github.com/felis/USB_Host_Shield_2.0/wiki/Bluetooth-dongles.

darkain, 2017-09-23:
Can we get this issue re-opened? It still exists within the current version of the library. I've currently isolated the issue and will be working on a patch over the next week or so. 

TLDR: when error code 0x0B is received, simply retransmitting the last HCI command will solve this issue.

Lauszus, 2017-09-23:
@darkain nice. Looking forward to the PR :)

Lauszus, 2017-11-10:
@moccault I have opened a PR that should hopefully fix this #327. I know this issue is old, but please test the code if you have time.