felis/USB_Host_Shield_2.0 #171: testusbhostFAT.ino example appears to be broken

marcah, 2015-08-20:
I have been testing a number of USB Host Shields with the intention of connecting two USB thumb drives through a hub for a data logging application.  Initially, all testing was performed with a set of SainSmart clones. Recently, I have received three genuine Circuits@Home USB Host Shields that seem to be failing in a similar manner.

In all cases, a version of the USB_Host_Shield_2.0 library installed a number of months ago seems to work fine.  Working fine in this case is defined as identifying a USB flash drive connected through a hub and printing a directory listing using the 'testusbhostFAT.ino example.  Recent updates of the library result in a report of the "USB state = 90" followed by a periodic report of the "Available Heap."

In an attempt to further understand the issue, I have tested a number of versions of the library with the following results:

Note: "current" = testusbhostFAT.ino from a recent download & installation
          "stock" = testusbhostFAT.ino from the tested SHA

2/19/2015
USB_Host_Shield_2.0-bd8c8e740419095f3c1d8875d199e16858058899
works w/ current testusbhostFAT
doesn't work w/ stock testusbhostFAT (UHS_USB_storage build error)

3/14/2015
USB_Host_Shield_2.0-46dc4e6feccea088091f8458239fbbe26a93fd72
works w/ current testusbhostFAT
doesn't work w/ stock testusbhostFAT (UHS_USB_storage build error)

3/24/2015
USB_Host_Shield_2.0-fcab83dcb398ec904db22009d21fa2c1fa133ba1
works w/ current testusbhostFAT
doesn't work w/ stock testusbhostFAT (freeHeap build error)

***  Appears to be broken from this point forward ***

4/3/2015
USB_Host_Shield_2.0-6a956962da048a9367015770744eaf286c0cba14
doesn't work w/ current testusbhostFAT
doesn't work w/ stock testusbhostFAT  (UHS_USB_storage)

4/4/2015
USB_Host_Shield_2.0-9e32aef86c12b4c58a721511b7f79e1b394061e0
doesn't work w/ current testusbhostFAT
doesn't work w/ stock testusbhostFAT

The failure appears to be related to line 480 of the sketch.  LUNIsGood() seems to always return 0 in the failed cases.  The same statement using the same hardware consistently returns 1.

480     if(UHS_USB_BulkOnly[B]->LUNIsGood(i)) {

The three additional libraries required by 'testusbhostFAT.ino' were all last changed prior to 2/22/2015 so it would appear that the issue is either within the USB Host Shield library or more likely an issue with the 'testusbhostFAT.ino' example code.

My objective is to learn how to use the C@H shield for data logging through the use of the example code.  Any additional pointers to example sketches demonstrating reading & writing files to multiple USB devices would be welcome.

I am available to assist anyone that is willing to investigate this issue further.  

Best Regards,
-marc

xxxajk, 2015-08-21:
Aware of the problems. I believe that something may be out of sync.

Since you want to do data logging, and since I assume you are not using an UNO, have you taken a look at UHS3 which is a lot more friendly, and hides the ugly details? It is a LOT easier in UHS 3.

Try it. https://github.com/felis/UHS30

marcah, 2015-08-21:
Hey Andrew,
Thanks for the response and the suggestion to consider UHS 3.  I am aware of UHS 3 but had sort of dismissed it as potentially not yet “mature.”  However, with your encouragement, I will seriously consider UHS 3.  Assuming no HW conflicts, do you foresee any fundamental problems with simultaneous support of both SD and UHS 3 for a logging application that will need to copy files from SD to USB devices?

Best Regards,
-marc

xxxajk, 2015-08-21:
If the SD library supports SPI transactions properly there should be no conflict at all.
What you will find really nice is that everything runs in the background to manage USB.
The API is very close to what you would find on a regular computer.
One note: expect to use unix paths, and mountpoints. After you get your head wrapped around those simple concepts, the rest of it will fall into place.
One question: why double storage? Why not just use two thumb drives and a hub, or use two host shields? UHS3 supports > 1 shield at a time.

marcah, 2015-08-21:
Hey Andrew,
The unix flavor shouldn’t pose an issue… my roots date back 3 decades w/ unix.

The application requires three storage devices, two of which must be USB.  I have selected SD as the primary storage medium for a couple of reasons: 1) lower power during the actual data logging states, 2) one configuration will potentially not need the USB devices at all.  So, the common denominator was a single, low power SD device with the option of adding 2 USB devices.  Both the USB and SD shields will be periodically enabled via solid state relays.  I have added a small 5 to 3.3v supply to the USB shield so that the shield can be powered by a single 5v supply.  The whole application must operate underwater for six months on a pair of Li Ion Polymer batteries.  The optional USB drives will be swappable underwater as they are over molded into separate “wet” connectors.

Best Regards,
-marc

xxxajk, 2015-08-21:
Are you aware that the C@H shield supports USB VBUS control with the addition of a capacitor and say-- a mic2004 on the shield? And yes, it is supported in both versions of the library too.

xxxajk, 2015-08-21:
see article here: https://www.circuitsathome.com/mcu/vbus-power-control-on-usb-host-shield

marcah, 2015-08-21:
I am now aware of the support for VBUS control via the current C@H shield.  I must admit that I started with a couple of clone shields before I really looked closer at the C@H offerings.  I now have three of the genuine C@H shields.  I will probably look at the use of the integrated VBUS control in the future.

Power has been a big concern.  I am using Mega 2560 boards which I have modified.  I have replaced the stock 5V regulator with one that draws almost nothing when the Mega is in sleep mode.  Also removed are the USB interface and associated chip.  With all of the changes, the Mega draws only 0.72 mA in sleep mode.  So in terms of the other shields, my objective was to turn them completely off.  I have not characterized the current usage of the C@H board, but will likely investigate your suggestion further in the future.   At this point, I have HW that I need to ship soon.  As best as possible, I need to “guarantee” that the HW is good as I will not have final FW until after the systems have been installed and sealed into a water tight enclosure.

I very much appreciate your insights and willingness to “engage” in conversation.

Regards,
-marc

xxxajk, 2015-08-21:
Helps to know the objectives, so I can guide you better.

marcah, 2015-08-23:
Hello Andrew,
Thanks again for your advice.

I have started looking into the use of UHS30 for my project.  I am having an issue getting the demo UHS_FS_NEW_DEMO to work.  Everything compiles and reports the following:
Start.

USB HOST READY.
USB HOST state 1d
No media. Waiting to mount /
USB HOST state 02
USB HOST state 0a
USB HOST state 03
USB HOST state 04
USB HOST state 05
USB HOST state 0c
USB HOST state 0d
USB HOST state 60
My configuration includes the following:
Arduino 1.6.5
Sainsmart Mega 2560
C@H USB Host Shield rev.2.0.1
Jumper connected between pins 9 & 3
8GB USB 2.0 Thumb Drive w/ a single 1GB FAT32 partition

I have tried a couple of other FAT32 drives w/ and w/o a hub with the same results.  Any chance I have overlooked something obvious?

As you are aware, UHS30 is currently described as follows: “This set of libraries is EXTERMELY ALPHA!”  Do you consider the libraries to be reasonably robust now 5 months from the initial release?

Best Regards,
-marc

xxxajk, 2015-08-23:
Remove the fat label from the volume to mount on /

marcah, 2015-08-23:
Thanks for the quick response.

Worked!  What you pointed out appears to be consistent with your observation yesterday… “One note: expect to use unix paths, and mountpoints. After you get your head wrapped around those simple concepts, the rest of it will fall into place.”

Do you have a feel for when UHS30 could be declared ready for release?

Again, thanks for your assistance!
-marc

xxxajk, 2015-08-23:
The core is what I consider stable, as well as the other bits. No real
problems reported. It lacks the extra drivers still.
Let me know if its stable for you.

marcah, 2015-08-23:
I will certainly keep you informed regarding any stability issues.

Based on the brief description of my intended use (logging to two thumb drives), would you have any reservation in deploying UHS30?  My sense is that with your assistance, I am already further along that I was with USB_Host_Shield_2.0 after several days.

Best Regards,
-marc

xxxajk, 2015-08-23:
No reservations at all. Glad you find the newer library easier to use.
For mount points just label the volume.
For example, if you label the volume 'foo', it mounts at /foo, and files
would be /foo/bar.txt.
This way no matter where it is plugged, you access the same volume.

One important point with multiple volumes with the same name: last one
mounted wins. So be sure to use a different name for each volume.
Also, note that a label-less '/' root volume is not required.
All volumes are mounted from '/' except label-less, which is a special
exception.
You also do not need the mount point directories.
The whole concept is a blended mix from several OS, but the naming, except
root concept, I borrowed from Apple's ProDOS which came out in the 80's.

marcah, 2015-08-24:
Hey Andrew,
I am continuing to test UHS30.  I have successfully mounted several thumb drives w/ and w/o a hub using the “label-less” root volume.  Per your instructions below, I have named a volume ‘foo’ and have successfully mounted it.

Is there a means prior to mounting to identify the volumes that are available for mounting?  My application will mount 2 of 10 available drives at any one time.  The drives will be setup with unique volume names so that any two can be utilized at the same time.  I will not know which of the 10 are available.  I could attempt to mount all 10 volume names and then stop when two are mounted.  Any advice?

Regards,
-marc

<mailto:notifications@github.com%3cmailto:%0b>> <mailto:notifications@github.com%3cmailto:%0b>> notifications@github.com<mailto:notifications@github.com>

xxxajk, 2015-09-06:
Has the new library met your expectations? 
How has your project worked out?

marcah, 2015-09-07:
Hello Andrew,
Thanks for checking in.  Thus far, I have had good success with UHS30.  Based on my experience with the library and your advice, I am committed to the use of UHS30.  I am not very far along with the actual firmware development as the recent focus has been ensuring that the hardware design is solid.  I expect to fully transition to firmware development by the end of this week.  You may recall that I had described a desire to utilize both SD and USB storage devices.  I am now on a path to utilize three USB devices and am not likely to support the SD device.

I appreciate your great work in support of USB and your willingness to share some advice.

Regards,
-marc

xxxajk, 2015-09-07:
Thanks for the positive feedback.
I knew that the ease-of-use and hiding messy details would be a huge help for everybody. 
Thank you very much for verifying this as a fact.

That said, if you have any improvements, suggestions, run into any other problems, or, just want to check in with more positive feedback, please post on on the 3.0 branch, since father discussions do indeed belong there.

Yes, I really am interested in all feedback. Please start a new thread there.

closed 2015-09-07 by xxxajk

xxxajk, 2015-09-07:
Oh and to answer the previous question I meissed about testing if a volume is mounted...

https://github.com/felis/UHS30/blob/master/libraries/UHS_FS/UHS_FS_INLINE.h#L288
Documentation is right there...

marcah, 2015-09-23:
Hello Andrew,

Per your request, I have posted “on the 3.0 branch” but it seems that there may not be anyone monitoring that branch.

Regards,
-marc

Lauszus, 2015-09-23:
@marcah we are monitoring the branch, but I have just been extremely busy. I have answered at the issue now.

xxxajk, 2015-09-24:
Ditto.