felis/USB_Host_Shield_2.0 #173: XBOXONE issue

Xiaoxin715, 2015-09-09:
Hi! We use your USB Host shield for Arduino to get the data from ThrustMaster TX Racing Wheel for XBOX ONE platform. The ThrustMaster TX Racing Wheel is a simulated driving controller. However, it didn't work, i modify the XBOX_VID and XBOX_ONE_PID to match the controller. The debug window shows the arduino has connected the controller already and the data array from the controller is the same except the third data which increases by 1 almost every second. However, the LED on the controller is still on which means it did not connect to the master as the LED is off when connected to the PC successfully. Do you have some suggestions about this, i am looking for your response. Thanks a lot!

Lauszus, 2015-09-09:
@Xiaoxin715 so you have uncomment the following lines: https://github.com/felis/USB_Host_Shield_2.0/blob/master/XBOXONE.cpp#L24-L25. Does the output printed in the serial monitor change when you move the wheel?

Please also post the output here in this issue. Also have you looked if there is any Linux driver available. This might help you develop a driver for the USB Host shield library.

Xiaoxin715, 2015-09-10:
@Lauszus Yeah, I have uncomment the EXTRADEBUG and PRINTREPORT in order to print the data. The output is the following data
Xbox Poll: 0202 20 84 1C 72 E2 31 F1 E8 02 00 00 4F 04 64 B6 01 00 00 00 30 00 01 00 01 01 01 00 01 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Xbox Poll: 0202 20 85 1C 72 E2 31 F1 E8 02 00 00 4F 04 64 B6 01 00 00 00 30 00 01 00 01 01 01 00 01 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Xbox Poll: 0202 20 86 1C 72 E2 31 F1 E8 02 00 00 4F 04 64 B6 01 00 00 00 30 00 01 00 01 01 01 00 01 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
     As I modified the EP_MAXPKTSIZE and XBOX_REPORT_BUFFER_SIZE both to 64 in XBOXONE.h, so the output length is 64. However, when we move the wheel, the output did not change except the third data, it increase by ONE almost every second. However, when the wheel keep static, the third data also increase one every time.
     As the Linux driver, i did not find the Linux driver except the driver of windows in the campany webpage (http://ts.thrustmaster.com/eng/index.php?pg=view_files&gid=12&fid=83&pid=366&cid=15).
     As for the output, do you have some idea about it? Thanks a lot!

Lauszus, 2015-09-10:
@Xiaoxin715 how are you even sure that it got the same descriptor as the Xbox One controller? Please run the following example: https://github.com/felis/USB_Host_Shield_2.0/tree/master/examples/USB_desc.

Also it might be easier to modify the following example instead: https://github.com/felis/USB_Host_Shield_2.0/tree/master/examples/HID/USBHIDJoystick.

Xiaoxin715, 2015-09-10:
@Lauszus Yeah! The offical document says it is for XBOX ONE platform or PC, so it is suggested to be a Xbox One controller. As the two examples you mentioned, i have tried to test it already but it didn't seem work. 
    
    In the page (https://github.com/quantus/xbox-one-controller-protocol), you introduced the XBOXONE  protocol, could you give us more information about it. I think that may the XBOXCommand() to the controller is not  appropriate. https://github.com/felis/USB_Host_Shield_2.0/blob/master/XBOXONE.cpp#L357-L370. Do you have suggestion about this protocol we use between the controller and the XBOXONE  master? Thanks a very much!

Xiaoxin715, 2015-09-10:
@Lauszus Yeah! The offical document says it is for XBOX ONE platform or PC, so it is suggested to be a Xbox One controller. As the two examples you mentioned, i have tried to test it already but it didn't seem work.

In the page (https://github.com/quantus/xbox-one-controller-protocol), you introduced the XBOXONE  protocol, could you give us more information about it. I think that may the XBOXCommand() to the controller is not  appropriate. https://github.com/felis/USB_Host_Shield_2.0/blob/master/XBOXONE.cpp. Do you have suggestion about this protocol we use between the controller and the XBOXONE  master? Thanks a very much!

Lauszus, 2015-09-10:
@Xiaoxin715 just because it says that it works with the Xbox One platform doesn't mean that it has the same descriptor as a Xbox One controller! Please run the first example: https://github.com/felis/USB_Host_Shield_2.0/tree/master/examples/USB_desc and post the output when you plug in the racing wheel.

Xiaoxin715, 2015-09-11:
@Lauszus Thanks for your suggestion. I have run the example you suggested and the output is:

Start

01
--

Device descriptor: 
Descriptor Length:	12
Descriptor type:	01
USB version:		0200
Device class:		FF
Device Subclass:	FF
Device Protocol:	FF
Max.packet size:	40
Vendor  ID:		044F
Product ID:		B664
Revision ID:		0101
Mfg.string index:	01
Prod.string index:	02
Serial number index:	03
Number of conf.:	01

Configuration descriptor:
Total length:		0020
Num.intf:		01
Conf.value:		01
Conf.string:		00
Attr.:			E0
Max.pwr:		FA

Interface descriptor:
Intf.number:		00
Alt.:			00
Endpoints:		02
Intf. Class:		FF
Intf. Subclass:		47
Intf. Protocol:		D0
Intf.string:		00

Endpoint descriptor:
Endpoint address:	01
Attr.:			03
Max.pkt size:		0040
Polling interval:	04

Endpoint descriptor:
Endpoint address:	81
Attr.:			03
Max.pkt size:		0040
Polling interval:	04

Addr:1(0.0.1)

This is the output data when I plug my wheel into the USB_HOST_SHIELD, is there something wrong? Do you have suggestion about that? Thanks very much!

Lauszus, 2015-09-11:
Thanks. Could you run the following example as well: https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/HID/USBHID_desc/USBHID_desc.ino?

Xiaoxin715, 2015-09-11:
@Lauszus Thanks! I have just run it. The output is:

Start
HU Init
Addr:01
NC:01

And there isn't any other output data in the debug window when I move the wheel. Do you have some  suggestion about what I should do next? Thanks a lot!

Xiaoxin715, 2015-09-11:
@Lauszus As I modified the XBOX_PID and XBOX_ONE_PID to match the controller. The XBOX_PID and XBOX_ONE_PID of the Microsoft XBOX_ONE controller are 0x045E and 0x02D1 respectively as you mentioned in the XBOXONE_h. However, the VID and PID of my controller are 0x044F(the thrustmaster company's ID) and 0xB664(The product's ID)respectively which aren't equal to the Microsoft‘s. Is there something wrong for me to use the XBOX_ONE library? Thanks!

Lauszus, 2015-09-23:
You can NOT just use the Xbox One controller code with your thrustmaster racing wheel.

I would recommend taking a look at the HID code and the provided examples: https://github.com/felis/USB_Host_Shield_2.0/tree/master/examples/HID.

I will close this issue, as it is really not a problem with the Xbox One code, but more a feature request.

closed 2015-09-23 by Lauszus