felis/USB_Host_Shield_2.0 #175: adk communication problem with android

clkasd, 2015-09-22:
I am sending some int value in every 5 seconds but some of them is not being sent to android.I am checking rcode variable but that doesn't make any sense.
My code is like this :
            timer = millis();
            rcode = adk.SndData(sizeof(timer), (uint8_t*)&timer);

felis, 2015-09-22:
This code sends much faster than every 5 seconds. You need to add a delay between these two lines.

clkasd, 2015-09-30:
Sure , there was a delay but I didn't post it. To be more specific I'd like to post an example code from your library :

//ArduinoBlinkLED example: https://github.com/felis/USB_Host_Shield_2.0/tree/master/examples/adk/ArduinoBlinkLED(Example URL)

 if (millis() - timer >= 1000) { // Send data every 1s
            timer = millis();
            rcode = adk.SndData(sizeof(timer), (uint8_t*)&timer);
            Serial.print(timer);
            Serial.print(" - ");
            Serial.print(rcode);
            Serial.println();
        }
}

This piece of code sends timer variable every 1 second.But I can't see all of these data from android.For example arduino sends 1140 ,1240,1340,1440,1540,1640 and so on, what I see on android is 1140,1340,1640.I mean some of them are missing.You should try it yourself.I have used referanced apk in .ino file(http://www.tkjelectronics.dk/uploads/ArduinoBlinkLED.apk).

clkasd, 2015-09-30:
Now I have tried with 3 android phones.The problem occurs only on samsung note4.

Lauszus, 2015-09-30:
@clkasd I have had the same problem previously. It's properly a hardware issue with the phone and nothing we can do anything about, so I will close this issue.

You could properly deal with this if you implemented a protocol, so you will get a acknowledge from the phone that the data has actually been received. If not, then send the data again.

closed 2015-09-30 by Lauszus