felis/USB_Host_Shield_2.0 #177: Bluetooth Dongle Addition / Clean-Up

spencercrissman, 2015-10-05:
I've purchased and used the Azio BTD-V201 USB Micro Bluetooth Adapter, Class 1, V2.1 + EDR, and can confirm it works with the host shield:  
[Azio BTD-V201 USB Micro Bluetooth Adapter, Class 1, V2.1 + EDR](http://www.amazon.com/gp/product/B003765X38)

Further, the following links from the existing dongle list are broken due to products no longer being available or the auctions having ended:
* http://www.ebay.com/itm/220704691620?ssPageName=STRK:MEWNX:IT&_trksid=p3984.m1439.l2649#ht_1477wt_1189
* http://www.ebay.com/itm/320869717301?ssPageName=STRK:MEWNX:IT&_trksid=p3984.m1439.l2649#ht_2973wt_1189
* http://www.ebay.com/itm/250537500254?ssPageName=STRK:MEWNX:IT&_trksid=p3984.m1439.l2649#ht_4377wt_956
* http://www.ebay.com/itm/110847675498?ssPageName=STRK:MEWNX:IT&_trksid=p3984.m1439.l2649#ht_1966wt_1732
* http://www.microcenter.com/single_product_results.phtml?product_id=0279381

It doesn't appear to be possible to submit a pull request for the wiki, or to attach a patch file to this issue, so hopefully this is still helpful.

spencercrissman, 2015-10-05:
Also, I don't know that this is called out anywhere, but I can confirm specifically that the Azio adapter works with a PS3 controller.

Lauszus, 2015-10-06:
@spencercrissman thanks. The Wiki has now been updated accordingly.

closed 2015-10-06 by Lauszus