felis/USB_Host_Shield_2.0 #180: Mass Storage on Teensy 2.0

custodium, 2015-10-29:
Hello, It's possible use Mas Storage functions on Teensy 2.0?

Best regards.

xxxajk, 2015-10-29:
If it reports as bulk storage and contains the minimum scsi command set,
yes. I still need to find this out myself. If you are only using mass
storage, you may like uhs 3 better, since it takes care of the fine details
for you.

custodium, 2015-11-03:
I meant if the testusbhostFAT example will work on Teensy 2.0 with Atmega32u4

xxxajk, 2015-11-04:
Short answer: yes, just use a usb-mini or full-sized shield.

xxxajk, 2015-11-04:
Additional note: Memory will be an extremely scarce resource.
You may not be able to do some operations, so don't expect too much.

custodium, 2015-11-04:
A lot of thanks Andrew, we are going to test it, we only need some basic tasks like read byte each byte some files and something else (not mapping files, directories or something else)

Lauszus, 2017-10-12:
It seems like this issue was solved, but never closed, so closing it now.

closed 2017-10-12 by Lauszus