felis/USB_Host_Shield_2.0 #181: testusbhostFAT + Teensy duino 2.0/++2.0

custodium, 2015-11-05:
Hi, I've tried compiling the testusbhostFAT example, with Teensy 3.0. The code works and compiles well, but with others versions of Teensy, somethings don't work. It's possible use testusbhostFAT with Teensy2.0 or Teensy++2.0? 

Thanks,

Compilation errors are the next one:

**FOR TEENSY 2.0:**

```In file included from /home/msubias/bin/arduino-1.6.3/libraries/xmem2/xmem.h:324:0,
from testusbhostFAT.ino:64:
/home/msubias/bin/arduino-1.6.3/libraries/xmem2/xmem.cpp: In function 'void xmem::begin(bool, bool)':
/home/msubias/bin/arduino-1.6.3/libraries/xmem2/xmem.cpp:278:17: error: 'XMCRB' was not declared in this scope
XMCRB = 0; // need all 64K. no pins released
^
/home/msubias/bin/arduino-1.6.3/libraries/xmem2/xmem.cpp:283:17: error: 'XMCRA' was not declared in this scope
XMCRA = 1 << SRE; // enable xmem, no wait states
^
/home/msubias/bin/arduino-1.6.3/libraries/xmem2/xmem.cpp:283:30: error: 'SRE' was not declared in this scope
XMCRA = 1 << SRE; // enable xmem, no wait states
^
In file included from /home/msubias/bin/arduino-1.6.3/hardware/tools/avr/avr/include/avr/io.h:99:0,
from /home/msubias/bin/arduino-1.6.3/hardware/tools/avr/avr/include/avr/interrupt.h:38,
from /home/msubias/bin/arduino-1.6.3/hardware/teensy/avr/cores/teensy/WProgram.h:18,
from /home/msubias/bin/arduino-1.6.3/hardware/teensy/avr/cores/teensy/Arduino.h:1,
from /home/msubias/bin/arduino-1.6.3/libraries/xmem2/xmem.h:25,
from testusbhostFAT.ino:64:
/home/msubias/bin/arduino-1.6.3/libraries/xmem2/xmem.cpp:291:31: error: 'PE7' was not declared in this scope
PORTE &= ~_BV(PE7);
^```

**FOR TEENSY++2.0:**

```In file included from /home/msubias/bin/arduino-1.6.3/libraries/xmem2/xmem.h:324:0,
from testusbhostFAT.ino:64:
/home/msubias/bin/arduino-1.6.3/libraries/xmem2/xmem.cpp: In function 'void xmem::begin(bool, bool)':
/home/msubias/bin/arduino-1.6.3/libraries/xmem2/xmem.cpp:278:17: error: 'XMCRB' was not declared in this scope
XMCRB = 0; // need all 64K. no pins released
^
/home/msubias/bin/arduino-1.6.3/libraries/xmem2/xmem.cpp:283:17: error: 'XMCRA' was not declared in this scope
XMCRA = 1 << SRE; // enable xmem, no wait states
^
/home/msubias/bin/arduino-1.6.3/libraries/xmem2/xmem.cpp:283:30: error: 'SRE' was not declared in this scope
XMCRA = 1 << SRE; // enable xmem, no wait states
^
In file included from /home/msubias/bin/arduino-1.6.3/hardware/tools/avr/avr/include/avr/io.h:99:0,
from /home/msubias/bin/arduino-1.6.3/hardware/tools/avr/avr/include/avr/interrupt.h:38,
from /home/msubias/bin/arduino-1.6.3/hardware/teensy/avr/cores/teensy/WProgram.h:18,
from /home/msubias/bin/arduino-1.6.3/hardware/teensy/avr/cores/teensy/Arduino.h:1,
from /home/msubias/bin/arduino-1.6.3/libraries/xmem2/xmem.h:25,
from testusbhostFAT.ino:64:
/home/msubias/bin/arduino-1.6.3/libraries/xmem2/xmem.cpp:291:31: error: 'PE7' was not declared in this scope
PORTE &= ~_BV(PE7);
^```

Lauszus, 2015-11-05:
Please update the Arduino IDE to 1.6.6: https://www.arduino.cc/en/Main/Software and try again.

Lauszus, 2015-11-05:
Or just don't include the Xmem library, as it is not needed.

xxxajk, 2015-11-05:
Yeah, xmem is only if you use eXternal MEMory

xxxajk, 2015-11-05:
This assumes you are using eXternal MEMory. You sgould install xmem2 and
grab an external memory for this to work. Teensy 2++/mega 1280/mega 2560
required along with external ram.

Lauszus, 2017-10-12:
Closing due to lack of feedback.

closed 2017-10-12 by Lauszus