felis/USB_Host_Shield_2.0 #182: BluClik HID remote hangs at Send HID Control Connection Request

superdavex, 2015-11-08:
Below is the connection log
HID Bluetooth Library Started
Bluetooth Dongle Initialized
HCI Reset complete
Write class of device
Local Bluetooth Address: 5C:F3:70:6D:AF:FC
Please enable discovery of your device
Keyboard found
HID device found
Now just create the instance like so:
BTHID bthid(&Btd);
And then press any button on the device
Connecting to HID device
Connected to HID device
HCI Command: 0x11 04 
Received Key Request
Bluetooth pin is set too: 0000
Pairing successful with HID device
Send HID Control Connection Request

These are the only events that follow
HCI Event Task: 05 04 00 0B 00 13 B5 29 24 26 2C DC 
HCI Event Task: 13 05 01 0B 00 01 00 29 24 26 2C DC

Below is a log of a successful connection to it from android
https://www.dropbox.com/s/60f2k9hmzcc9xt3/btsnoop_hci.log?dl=0

Lauszus, 2015-12-10:
@superdavex could you enable EXTRADEBUG both in BTD.cpp and SPP.cpp: https://github.com/felis/USB_Host_Shield_2.0/blob/8bd872832141055868a3381b59aa2e5959f0af36/BTD.cpp#L20 and https://github.com/felis/USB_Host_Shield_2.0/blob/523e66e827d345faa9611d9de8e24a4b128ca86b/BTHID.cpp and then post the output?

superdavex, 2015-12-10:
Here is the output with extra debug enabled.  I also added some additional debugging.
````
HID Bluetooth Library Started
BTD ConfigureDevice
BTD Init
Addr: 01
Endpoint descriptor:
Length:		07
Type:		05
Address:	81
Attributes:	03
MaxPktSize:	0010
Poll Intrv:	01
Endpoint descriptor:
Length:		07
Type:		05
Address:	82
Attributes:	02
MaxPktSize:	0040
Poll Intrv:	01
Endpoint descriptor:
Length:		07
Type:		05
Address:	02
Attributes:	02
MaxPktSize:	0040
Poll Intrv:	01
Bluetooth Dongle Initialized
HCI Command: 0x03 0C 00 
HCI Event Task: 0E 04 01 03 0C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Command Complete for: 030C
HCI Reset complete
HCI Command: 0x24 0C 03 04 08 00 
Write class of device
HCI Command: 0x09 10 00 
Local Bluetooth Address: 5C:F3:70:6D:AF:FC
HCI Command: 0x01 10 00 
Please enable discovery of your device
HCI Command: 0x01 04 05 33 8B 9E 30 0A 
HCI Event Task: 0F 04 00 01 01 04 00 00 10 06 0F 00 0E 22 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
HCI Status for: 01
Number of responses: 1
Class of device: 00 25 40
Keyboard found
HCI Command: 0x02 04 00 
HID device found
Now just create the instance like so:
BTHID bthid(&Btd);
And then press any button on the device
HCI Event Task: 0E 04 01 02 04 00 26 2C DC 01 00 00 40 25 00 C7 1A 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Command Complete for: 0204
Connecting to HID device
HCI Command: 0x05 04 0D B5 29 24 26 2C DC 18 CC 01 00 00 00 00 
HCI Event Task: 0F 04 00 01 05 04 00 00 00 00 00 00 00 00 00 00 1A 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
HCI Status for: 05
HCI Event Task: 03 0B 00 0B 00 B5 29 24 26 2C DC 01 00 00 00 00 1A 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Connection established
Connected to HID device
HCI Command: 0x11 04 02 0B 00 
HCI Event Task: 1B 03 0B 00 05 B5 29 24 26 2C DC 01 00 00 00 00 1A 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
HCI Event Task: 0F 04 00 01 11 04 29 24 26 2C DC 01 00 00 00 00 1A 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
HCI Status for: 11
HCI Event Task: 17 06 B5 29 24 26 2C DC 26 2C DC 01 00 00 00 00 1A 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Received Key Request
HCI Command: 0x0C 04 06 B5 29 24 26 2C DC 
HCI Event Task: 0E 0A 01 0C 04 00 B5 29 24 26 2C DC 00 00 00 00 1A 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Command Complete for: 0C04
HCI Event Task: 16 06 B5 29 24 26 2C DC 24 26 2C DC 00 00 00 00 1A 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Bluetooth pin is set too: 0000
HCI Command: 0x0D 04 17 B5 29 24 26 2C DC 04 30 30 30 30 00 00 00 00 00 00 00 00 00 00 00 00 
HCI Event Task: 0E 0A 01 0D 04 00 B5 29 24 26 2C DC 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Command Complete for: 0D04
HCI Event Task: 06 03 13 0B 00 00 B5 29 24 26 2C DC 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Pairing successful with HID device
Send HID Control Connection Request
HCI Event Task: 05 04 00 0B 00 13 B5 29 24 26 2C DC 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
HCI Event Task: 13 05 01 0B 00 01 00 29 24 26 2C DC 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
````

Lauszus, 2015-12-10:
@superdavex I guess the "HCI Event Task" is yours? Where exactly is it added and what lines is it printing?

Lauszus, 2015-12-10:
I mean does it start from index 0?

Lauszus, 2015-12-10:
@superdavex it seems like your device disconnect, as indicated by this line:
```
HCI Event Task: 05 04 00 0B 00 13 B5 29 24 26 2C DC 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```

Btw what software do you use to see the Android log?

superdavex, 2015-12-10:
````
void BTD::HCI_event_task() {
        uint16_t length = BULK_MAXPKTSIZE; // Request more than 16 bytes anyway, the inTransfer routine will take care of this
        uint8_t rcode = pUsb->inTransfer(bAddress, epInfo[ BTD_EVENT_PIPE ].epAddr, &length, hcibuf, pollInterval); // Input on endpoint 1

        if(!rcode || rcode == hrNAK) { // Check for errors
		
				if (last_hci_event != hcibuf[0] && hcibuf[0] != EV_INQUIRY_RESULT )
				{
					Notify(PSTR("\r\nHCI Event Task: "), 0x80);
					for(uint16_t ii = 0; ii < sizeof(hcibuf); ii++) // L2CAP C-frame
					{
						D_PrintHex<uint8_t > (hcibuf[ii], 0x80);
						Notify(PSTR(" "), 0x80);
					}
				}
````

superdavex, 2015-12-10:
I use Wireshark to open the log.  Android does a simple pairing with it instead of using the pin.  I have a  Satechi Bluetooth Button that has the same issue

Lauszus, 2015-12-10:
The 0x13 byte indicates the reason for the disconnection:
```
2.19 REMOTE USER TERMINATED CONNECTION (0x13)
The Remote User Terminated Connection error code indicates that the user on the remote device terminated the connection.
```

Lauszus, 2015-12-10:
Hmm are you sure 0000 is the right pin?

Lauszus, 2015-12-10:
What happens if you press a button on the device? Does it try to pair with the Arduino?

Lauszus, 2015-12-10:
Please also add some debugging at the "ACL_event_task", so I can see if it sends anything there.

Lauszus, 2015-12-10:
Also I find it a bit strange that it never sends a "User Confirmation Request Event".

superdavex, 2015-12-10:
This is the log with ACL Event added.  If you do not press the button it will pause after 'Command Complete for: 0D04'.    The manufacturer does not list a pin for the device.  I have tried all common pins.  It does return a pin successful event though.  On android it does not use the pin and connects via simple pairing.  I've started working on implementing simple pairing, but, still have the same issue so far.

````

HID Bluetooth Library Started
BTD ConfigureDevice
BTD Init
Addr: 01
Endpoint descriptor:
Length:		07
Type:		05
Address:	81
Attributes:	03
MaxPktSize:	0010
Poll Intrv:	01
Endpoint descriptor:
Length:		07
Type:		05
Address:	82
Attributes:	02
MaxPktSize:	0040
Poll Intrv:	01
Endpoint descriptor:
Length:		07
Type:		05
Address:	02
Attributes:	02
MaxPktSize:	0040
Poll Intrv:	01
Bluetooth Dongle Initialized
HCI Command: 0x03 0C 00 
HCI Event Task: 0E 04 01 03 0C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Command Complete for: 030C
HCI Reset complete
HCI Command: 0x24 0C 03 04 08 00 
Write class of device
HCI Command: 0x09 10 00 
Local Bluetooth Address: 5C:F3:70:6D:AF:FC
HCI Command: 0x01 10 00 
Please enable discovery of your device
HCI Command: 0x01 04 05 33 8B 9E 30 0A 
HCI Event Task: 0F 04 00 01 01 04 00 00 10 06 0F 00 0E 22 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
HCI Status for: 01
Number of responses: 1
Class of device: 06 01 0C
Gamepad found
HCI Command: 0x02 04 00 
HID device found
Now just create the instance like so:
BTHID bthid(&Btd);
And then press any button on the device
HCI Event Task: 0E 04 01 02 04 00 19 D8 60 01 00 00 0C 01 06 77 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Command Complete for: 0204
Connecting to HID device
HCI Command: 0x05 04 0D 30 41 FA 19 D8 60 18 CC 01 00 00 00 00 
HCI Event Task: 0F 04 00 01 05 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
HCI Status for: 05
HCI Event Task: 03 0B 00 0B 00 30 41 FA 19 D8 60 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Connection established
Connected to HID device
HCI Command: 0x11 04 02 0B 00 
HCI Event Task: 1B 03 0B 00 05 30 41 FA 19 D8 60 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
ACL Event: 0B 20 
HCI Event Task: 0F 04 00 01 11 04 41 FA 19 D8 60 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
HCI Status for: 11
HCI Event Task: 17 06 30 41 FA 19 D8 60 19 D8 60 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Received Key Request
HCI Command: 0x0C 04 06 30 41 FA 19 D8 60 
HCI Event Task: 0E 0A 01 0C 04 00 30 41 FA 19 D8 60 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Command Complete for: 0C04
[... 61 lines cut ...]
HCI Event Task: 0F 04 00 01 05 04 00 00 00 00 00 00 00 00 00 00 7F 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
HCI Status for: 05
HCI Event Task: 03 0B 00 0B 00 B5 29 24 26 2C DC 01 00 00 00 00 7F 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Connection established
Connected to HID device
HCI Command: 0x11 04 02 0B 00 
HCI Event Task: 1B 03 0B 00 05 B5 29 24 26 2C DC 01 00 00 00 00 7F 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
HCI Event Task: 0F 04 00 01 11 04 29 24 26 2C DC 01 00 00 00 00 7F 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
HCI Status for: 11
HCI Event Task: 17 06 B5 29 24 26 2C DC 26 2C DC 01 00 00 00 00 7F 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Received Key Request
HCI Command: 0x0C 04 06 B5 29 24 26 2C DC 
HCI Event Task: 0E 0A 01 0C 04 00 B5 29 24 26 2C DC 00 00 00 00 7F 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Command Complete for: 0C04
HCI Event Task: 16 06 B5 29 24 26 2C DC 24 26 2C DC 00 00 00 00 7F 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Bluetooth pin is set too: 0000
HCI Command: 0x0D 04 17 B5 29 24 26 2C DC 04 30 30 30 30 00 00 00 00 00 00 00 00 00 00 00 00 
[... cut ...] 00 00 
Pairing successful with HID device
Send HID Control Connection Request
HCI Event Task: 05 04 00 0B 00 13 B5 29 24 26 2C DC 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
HCI Event Task: 13 05 01 0B 00 01 00 29 24 26 2C DC 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
````

Lauszus, 2015-12-10:
@superdavex okay super. Simple paring would be a great contribution!

superdavex, 2015-12-13:
These devices do not support pin based pairing. I have add simple secure pairing and they connect and authenticate fine.  I'll get the code cleaned up and submit a patch.

Lauszus, 2015-12-13:
@superdavex thanks. Looking forward to your PR :)

Lauszus, 2016-01-03:
@superdavex any news?

Lauszus, 2016-01-21:
@superdavex can you publish your code you have so far somewhere? It's okay if you don't have time to create a PR. I will just clean up the code and merge it. I got another user who is having the same problem as you and I believe your code should fix it.

deexe001, 2016-07-02:
@superdavex This is exactly the problem I am facing and it would be so helpful if a solution was posted.

deexe001, 2016-07-02:
@Lauszus It looks like @superdavex is no longer active. Can you think of any other way to get around this problem?

andremalenfant, 2023-01-18:
Hi, wondering if anyone made progress on this. I am trying to pair a keypad and it is not detected. It works with the same dongle in a computer, the shield detects other devices, but not the keyboard. It seems to be using the simple pairing method and the shield library doesn't see it.