felis/USB_Host_Shield_2.0 #183: Dongle not responding

kirchoff2000, 2015-11-13:
Hi my name is Jonathan, i’m from Montreal Canada.  
I’ve recently purchase the usb host shield version 2.0.
I want to use the ps3bt fonction.  I’m usind a mega 2560 v2, and Arduino ide 1.6.6
 
I’m usind a cheap bt dongle from china ( supposed the be supported).  I have about 5 of them and not one seem to work. 
The dongles’ light dont even light up (but they do light up in my pc).
 
I’ve tested with my multimetre, and there’s power on the chip.
 
I am not a soldering master but the pins seem nicely solder.
 
I also tryed an external power on my mega....   Still nothing.

I've enable debugging as well.
 
I’m runnig out of idea, can anyone help me?

The output is " OSC did not start "
 
Regards,
 
Jonathan B.

xxxajk, 2015-11-13:
This usually indicates there is a problem with wiring, indicating it isn't an official shield. 
The most recent shield version is 2.0.1, many clones are not produced properly, and will only work on an uno, and not work on a mega, due, etc.
Is this an official shield?

kirchoff2000, 2015-11-13:
Problem solved!!!  
One of the pin was not soldered properly.
thanks for your anwser xxxajk.

xxxajk, 2015-11-13:
Happy to help, especially when the solution is simple.

closed 2015-11-13 by xxxajk