felis/USB_Host_Shield_2.0 #184: epAttribs = 0 overriding bmNakPower in ACM::EndpointXtract and FTDI::EndpointXtract

drjrkuhn, 2015-11-17:
I have been trying to track down a timeout problem with my ACM endpoint device. The device was blocking in `USB::InTransfer`. I traced it through to the `USB::SetAddress` where `bmNakPower` was consistently reading a value of 0, which in turn calculated `nak_limit = 0`, which caused `USB::dispatchPkt` to hang up until a timeout.

This value of `bmNakPower = 0` was getting through despite the setting of `bmNakPower` to something other than zero in the ACM constructor

**cdcacm.cpp**
````C++
ACM::ACM(USB *p, CDCAsyncOper *pasync) :
...
{
        ...
        for(uint8_t i = 0; i < ACM_MAX_ENDPOINTS; i++) {
                ...
                epInfo[i].bmNakPower = (i == epDataInIndex) ? USB_NAK_NOWAIT : USB_NAK_MAX_POWER;
        }
        ...
}
````

I managed to trace the `bmNakPower` reset to zero to a line in `ACM::EndpointXtract`. Because `epAttribs` and `bmNakPower` are a union in the `EpInfo` structure, setting `epInfo[index].epAttribs = 0;` any time after the constructor **also** erases the value of `bmNakPower`. This problem crops up in both `ACM::EndpointXtract` and `FTDI::EndpointXtract`. For example:

**cdcacm.cpp**
````C++
void ACM::EndpointXtract(uint8_t conf, uint8_t iface, uint8_t alt, uint8_t proto, const USB_ENDPOINT_DESCRIPTOR *pep) {
        ...
        // Fill in the endpoint info structure
        epInfo[index].epAddr = (pep->bEndpointAddress & 0x0F);
        epInfo[index].maxPktSize = (uint8_t)pep->wMaxPacketSize;
        // PLEASE REMOVE THE FOLLOWING LINE:
        epInfo[index].epAttribs = 0;
        ...
}
````

Removing the line `epInfo[index].epAttribs = 0;` fixed my hanging problem. The `bmNakPower` in `USB::SetAddress` retains its initial value, `nak_limit` is calculated properly, and `USB::InTransfer` now immediately returns with a count of zero if nothing was in the input after 1 `hrNAK` result.

Lauszus, 2015-11-17:
Thank you very much for tracking down the problem! Do you mind sending a pull request? :)

Lauszus, 2015-11-17:
@drjrkuhn please test: https://github.com/felis/USB_Host_Shield_2.0/pull/185 and report back.

PR #185 (2015-11-22): Don't set epAttribs to 0, as this will set bmNakPower to 0 as well. I…
…nstead set bmSndToggle and bmRcvToggle explicit to 0.

Fixes: #184
commit fb723fcb (2015-11-22): Don't set epAttribs to 0, as this will set bmNakPower to 0 as well. Instead set bmSndToggle and bmRcvToggle explicit to 0.

Fixes: #184
closed 2015-11-22 by Lauszus