felis/USB_Host_Shield_2.0 #186: PS3BT: setLed will activate last rumble action

mman454, 2015-11-24:
I have found that setLedOn, setLedOff, and setLedToggle all have the side effect of triggering whatever the last rumble action was.

I have included a basic sketch that calls the setRumbleON function upon the connection of a PS3 controller.
Pressing select, PS, or start will call each of the different setLed functions.

I am using a Azio BTD-V201 dongle with an original Sony DualShock 3 Sixaxis controller.

```
#include <PS3BT.h>
#include <usbhub.h>

int newconnect = 0;
USB Usb;
USBHub Hub1(&Usb);
BTD Btd(&Usb);
PS3BT PS3(&Btd);

void setup() {
  Serial.begin(115200);
  if (Usb.Init() == -1)
  {
    Serial.print(F("\r\nOSC did not start"));
    while (1);
  }
  Serial.print(F("\r\nPS3 Bluetooth Library Started"));
}

void loop() {

  Usb.Task();

  if (PS3.PS3Connected)
  {
    if (newconnect == 0)
    {
      newconnect++;
      Serial.println("Rumble is on!");
      PS3.setRumbleOn(100, 255, 100, 255); //VIBRATE!!!
    }

    if (PS3.getButtonClick(SELECT)) {
      PS3.setLedOn(LED2);
    }
    if (PS3.getButtonClick(START)) {
      PS3.setLedOff(LED1);
    }
    if (PS3.getButtonClick(PS)) {
      PS3.setLedToggle(LED3);
    }
  }
}
```

matt-bull, 2016-01-13:
@Lauszus Is there any additional info I could provide or anything I could try to help you out on this one?

Lauszus, 2016-01-13:
@matt-bull sorry for not getting back to you. This PR should fix it: https://github.com/felis/USB_Host_Shield_2.0/pull/198. Please test it and report back!

matt-bull, 2016-01-13:
@Lauszus It appears that PR #198 has removed all rumble functionality as well as the ability to set the LEDs with any of the functions.

matt-bull, 2016-01-19:
PR #198 resolves this issue. Thanks @Lauszus! :)

commit 8ba8fe96 (2016-01-13): Store rumble value in local buffer, so the rumble is not affected when setting the LEDs

Fixes #186
commit 627c0ebc (2016-01-19): Store rumble value in local buffer, so the rumble is not affected when setting the LEDs

Fixes #186
PR #198 (2016-01-20): Store rumble value in local buffer, so the rumble is not affected whe…
…n setting the LEDs

Fixes #186
closed 2016-01-19 by matt-bull