felis/USB_Host_Shield_2.0 #187: Unsupported Bluetooth dongle

PaulLabbe, 2015-12-05:
My BT is not recognized, and I don't know how to progress using 1.6.6 and library 1.2.1. My Bluetooth dongle is version 3. PS3BT with debugging reports:

PS3 Bluetooth Library Started
Unknown Device Connected - VID: 0A5C PID: 4500
BTD Init Failed, error code: D1

Result from  board_qc:

Circuits At Home 2011
USB Host Shield Quality Control Routine
Reading REVISION register... Die revision 03
SPI long test. Transfers 1MB of data. Each dot is 64K................ SPI long test passed
GPIO test. Connect GPIN0 to GPOUT7, GPIN1 to GPOUT6, and so on
Test failed. Value written: 00 Value read: FF 
Press any key to continue...
GPIO test passed.
PLL test. 100 chip resets will be performed
Resetting oscillator

Reset number 0 Time to stabilize - 384 cycles
...
Getting device descriptor
Descriptor Length:	12
Descriptor type:	01
USB version:		0200
Device class:		09
Device Subclass:	00
Device Protocol:	00
Max.packet size:	08
Vendor  ID:		0A5C
Product ID:		4500
Revision ID:		0100
Mfg.string index:	01
Prod.string index:	02
Serial number index:	00
Number of conf.:	01
 
looks like this:
http://www.kijiji.ca/v-computer-networking/ville-de-montreal/mini-adaptateur-usb-bluetooth-3-0-ou-4-0-sans-fil-win7-yyyy/593045482?enableSearchNavigationFlag=true
[image]

felis, 2015-12-05:
Have you tried to run it from external power supply?

PaulLabbe, 2015-12-06:
Tried with a 5v transfo and a 7.2v transfo. I first made sure the board lit up with the external power sup, then added the usb cable. Same results:

PS3 Bluetooth Library Started
Unknown Device Connected - VID: 0A5C PID: 4500
BTD Init Failed, error code: D1

Lauszus, 2015-12-06:
Your dongle have a hub inside. Simply uncomment the following line: https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/Bluetooth/PS3BT/PS3BT.ino#L17 and it should work ;)

Lauszus, 2015-12-06:
Please check out the following pull request: https://github.com/felis/USB_Host_Shield_2.0/pull/190 and please confirm that it prints a more useful error message when running the PS3BT example when the USBHub instance is not uncommented.

PaulLabbe, 2015-12-08:
Commit 190 works exactly as intended: The error message is clear and provides the exact procedure to fix the issue.

Thanks!
Paul

Lauszus, 2015-12-08:
Thanks for your feedback!

closed 2015-12-08 by Lauszus

EPiress, 2016-06-24:
Hey Lauszus,

i have the exact same problem, when i try to export the PS3BT to my Arduino UNO i get this message in the serial monitor:

PS3 Bluetooth Library Started
Unknown Device Connected - VID: 050D PID: 065A
BTD Init Failed, error code: D1

I'm rocking a belkin v4.0 bluetooth dongle, and I can not find out what the problem is, even with all your help to other people!

PaulLabbe, 2016-06-24:
Have you tried uncommenting this line?

USBHub Hub1(&Usb); // Some dongles have a hub inside

Note that i had issues when i upgraded to 1.6.9. I had to erase everything
and reinstall from scratch.

Paul

EPiress, 2016-06-24:
Hey Paul,

yeah i've tried commenting that line, and i've also uncommented this line in BTD.cpp: 

#define EXTRADEBUG // Uncomment to get even more debugging data

and i get this from the serial monitor:

PS3 Bluetooth Library Started
BTD ConfigureDevice
BTD Init
Addr: 01
Unknown Device Connected - VID: 050D PID: 065A
BTD Init Failed, error code: D1

Thanks for the help :)

Lauszus, 2016-06-26:
@EPiress please try to upload and run the following code: https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/USB_desc/USB_desc.ino and then post the output here.

EPiress, 2016-06-26:
ok, so this is what i get from the usb_desc.ino file:

Device descriptor: 
Descriptor Length:	12
Descriptor type:	01
USB version:		0200
Device class:		FF
Device Subclass:	01
Device Protocol:	01
Max.packet size:	40
Vendor  ID:		050D
Product ID:		065A
Revision ID:		0112
Mfg.string index:	01
Prod.string index:	02
Serial number index:	03
Number of conf.:	01

Configuration descriptor:
Total length:		00DA
Num.intf:		04
Conf.value:		01
Conf.string:		00
Attr.:			A0
Max.pwr:		32

Interface descriptor:
Intf.number:		00
Alt.:			00
Endpoints:		03
Intf. Class:		FF
Intf. Subclass:		01
Intf. Protocol:		01
Intf.string:		00

Endpoint descriptor:
Endpoint address:	81
Attr.:			03
Max.pkt size:		0010
Polling interval:	01

Endpoint descriptor:
Endpoint address:	82
Attr.:			02
Max.pkt size:		0040
Polling interval:	01

Endpoint descriptor:
Endpoint address:	02
Attr.:			02
Max.pkt size:		0040
Polling interval:	01

Interface descriptor:
Intf.number:		01
Alt.:			00
Endpoints:		02
Intf. Class:		FF
Intf. Subclass:		01
Intf. Protocol:		01
Intf.string:		00

Endpoint descriptor:
Endpoint address:	83
Attr.:			01
Max.pkt size:		0000
Polling interval:	01

Endpoint descriptor:
Endpoint address:	03
Attr.:			01
Max.pkt size:		0000
Polling interval:	01

Interface descriptor:
Intf.number:		01
Alt.:			01
Endpoints:		02
Intf. Class:		FF
Intf. Subclass:		01
Intf. Protocol:		01
Intf.string:		00

Endpoint descriptor:
Endpoint address:	83
Attr.:			01
Max.pkt size:		0009
Polling interval:	01

Endpoint descriptor:
Endpoint address:	03
Attr.:			01
Max.pkt size:		0009
Polling interval:	01

Interface descriptor:
Intf.number:		01
Alt.:			02
Endpoints:		02
Intf. Class:		FF
Intf. Subclass:		01
Intf. Protocol:		01
Intf.string:		00

Endpoint descriptor:
Endpoint address:	83
Attr.:			01
Max.pkt size:		0011
Polling interval:	01

Endpoint descriptor:
Endpoint address:	03
Attr.:			01
Max.pkt size:		0011
Polling interval:	01

Interface descriptor:
Intf.number:		01
Alt.:			03
Endpoints:		02
Intf. Class:		FF
Intf. Subclass:		01
Intf. Protocol:		01
Intf.string:		00

Endpoint descriptor:
Endpoint address:	83
Attr.:			01
Max.pkt size:		0019
Polling interval:	01

Endpoint descriptor:
Endpoint address:	03
Attr.:			01
Max.pkt size:		0019
Polling interval:	01

Interface descriptor:
Intf.number:		01
Alt.:			04
Endpoints:		02
Intf. Class:		FF
Intf. Subclass:		01
Intf. Protocol:		01
Intf.string:		00

Endpoint descriptor:
Endpoint address:	83
Attr.:			01
Max.pkt size:		0021
Polling interval:	01

Endpoint descriptor:
Endpoint address:	03
Attr.:			01
Max.pkt size:		0021
Polling interval:	01

Interface descriptor:
Intf.number:		01
Alt.:			05
Endpoints:		02
Intf. Class:		FF
Intf. Subclass:		01
Intf. Protocol:		01
Intf.string:		00

Endpoint descriptor:
Endpoint address:	83
Attr.:			01
Max.pkt size:		0031
Polling interval:	01

Endpoint descriptor:
Endpoint address:	03
Attr.:			01
Max.pkt size:		0031
Polling interval:	01

Interface descriptor:
Intf.number:		02
Alt.:			00
Endpoints:		02
Intf. Class:		FF
Intf. Subclass:		FF
Intf. Protocol:		FF
Intf.string:		00

Endpoint descriptor:
Endpoint address:	84
Attr.:			02
Max.pkt size:		0020
Polling interval:	01

Endpoint descriptor:
Endpoint address:	04
Attr.:			02
Max.pkt size:		0020
Polling interval:	01

Interface descriptor:
Intf.number:		03
Alt.:			00
Endpoints:		00
Intf. Class:		FE
Intf. Subclass:		01
Intf. Protocol:		01
Intf.string:		00
Unknown descriptor:
Length:		09
Type:		21
Contents:	058813400010010809

Addr:1(0.0.1)

EPiress, 2016-06-30:
I still haven't figured out how to solve my problem...
Any help would be very welcome!

Haruffaizi, 2018-04-26:
Hi, please help since i'm newbies.
here is my problem-

PS3 Bluetooth Library Started
BTD ConfigureDevice
BTD Init
Addr: 01
Unknown Device Connected - VID: 0A12 PID: 0001
BTD Init Failed, error code: D1

had try different type of dongle but still cannot solve the problem
best regards

ellensp, 2019-12-07:
Also got a Unknown Device Connected - VID: 0A12 PID: 0001 which errors BTD Init Failed, error code: D1
USB desc follows 
Start
01
--
Device descriptor: 
Descriptor Length:	12
Descriptor type:	01
USB version:		0110
Device class:		E0
Device Subclass:	01
Device Protocol:	01
Max.packet size:	40
Vendor  ID:		0A12
Product ID:		0001
Revision ID:		7558
Mfg.string index:	00
Prod.string index:	00
Serial number index:	00
Number of conf.:	01

Configuration descriptor:
Total length:		00B1
Num.intf:		02
Conf.value:		01
Conf.string:		00
Attr.:			E0
Max.pwr:		32

Interface descriptor:
Intf.number:		00
Alt.:			00
Endpoints:		03
Intf. Class:		E0
Intf. Subclass:		01
Intf. Protocol:		01
Intf.string:		00

Endpoint descriptor:
Endpoint address:	81
Attr.:			03
Max.pkt size:		0040
Polling interval:	01

Endpoint descriptor:
Endpoint address:	82
Attr.:			02
Max.pkt size:		0040
Polling interval:	00

Endpoint descriptor:
Endpoint address:	02
Attr.:			02
Max.pkt size:		0040
Polling interval:	00

Interface descriptor:
Intf.number:		01
Alt.:			00
Endpoints:		02
Intf. Class:		E0
Intf. Subclass:		01
Intf. Protocol:		01
Intf.string:		00

Endpoint descriptor:
Endpoint address:	83
Attr.:			01
Max.pkt size:		0000
Polling interval:	01

Endpoint descriptor:
Endpoint address:	03
Attr.:			01
Max.pkt size:		0000
Polling interval:	01

Interface descriptor:
Intf.number:		01
Alt.:			01
Endpoints:		02
Intf. Class:		E0
Intf. Subclass:		01
Intf. Protocol:		01
Intf.string:		00

Endpoint descriptor:
Endpoint address:	83
Attr.:			01
Max.pkt size:		0009
Polling interval:	01

Endpoint descriptor:
Endpoint address:	03
Attr.:			01
Max.pkt size:		0009
Polling interval:	01

Interface descriptor:
Intf.number:		01
Alt.:			02
Endpoints:		02
Intf. Class:		E0
Intf. Subclass:		01
Intf. Protocol:		01
Intf.string:		00

Endpoint descriptor:
Endpoint address:	83
Attr.:			01
Max.pkt size:		0011
Polling interval:	01

Endpoint descriptor:
Endpoint address:	03
Attr.:			01
Max.pkt size:		0011
Polling interval:	01

Interface descriptor:
Intf.number:		01
Alt.:			03
Endpoints:		02
Intf. Class:		E0
Intf. Subclass:		01
Intf. Protocol:		01
Intf.string:		00

Endpoint descriptor:
Endpoint address:	83
Attr.:			01
Max.pkt size:		0019
Polling interval:	01

Endpoint descriptor:
Endpoint address:	03
Attr.:			01
Max.pkt size:		0019
Polling interval:	01

Interface descriptor:
Intf.number:		01
Alt.:			04
Endpoints:		02
Intf. Class:		E0
Intf. Subclass:		01
Intf. Protocol:		01
Intf.string:		00

Endpoint descriptor:
Endpoint address:	83
Attr.:			01
Max.pkt size:		0021
Polling interval:	01

Endpoint descriptor:
Endpoint address:	03
Attr.:			01
Max.pkt size:		0021
Polling interval:	01

Interface descriptor:
Intf.number:		01
Alt.:			05
Endpoints:		02
Intf. Class:		E0
Intf. Subclass:		01
Intf. Protocol:		01
Intf.string:		00

Endpoint descriptor:
Endpoint address:	83
Attr.:			01
Max.pkt size:		0031
Polling interval:	01

Endpoint descriptor:
Endpoint address:	03
Attr.:			01
Max.pkt size:		0031
Polling interval:	01

Addr:1(0.0.1)

PR #190 (2015-12-08): Print error message if a USB Hub is detected in the BTD class
This is useful as some Bluetooth dongles have a hub inside

Fixes #187
commit 50ae4015 (2015-12-08): Print error message if a USB Hub is detected in the BTD class

This is useful as some Bluetooth dongles have a hub inside

Fixes #187