felis/USB_Host_Shield_2.0 #188: Arduino Mega ADK and Nexus 6 connection problem

thaussma, 2015-12-05:
Hi,
for some hours I'm trying to connect my Android phones with my Arduino Mega ADK, but regardless what I did, the results are very unstable.

* I'm using the Arduino sketch and the Android app from the [ArduinoBlinkLED](https://github.com/Lauszus/ArduinoBlinkLED) example
* platformio to build and flash
* [USB-Host-Shield-20](http://platformio.org/#!/lib/show/59/USB-Host-Shield-20) trough platformio
* I used an external power source for the Arduino
* Enabled `USE_UHS_MEGA_ADK` (not required is it?)

My Nexus 6 seems not recognise the Arduino: 
* The intent filter in the AndroidManifest.xml seems not to work, the app is not started.
* UsbManager#getAccessoryList() does not return results
* Adk#isReady() returns always false
* When I connect the phone to the Arduino I can see a Notification "USB Debugging - to disable USB-Debugging [...]" for 2-3 seconds then it disappears.

Twice the Android App was started when I connected the phone with the board. Both times I had reinstalled the app. A permission dialog was presented and the blinking example worked fine.

I enabled the debug mode for the USB host. This is what I'm seeing:
```
Arduino Blink LED Started
ADK Init
ADK protocol rev. 0002

Accessory mode switch attempt

ADK Init
ADK protocol rev. 0002

Accessory mode switch attempt
```

My Galaxy Nexus works better. Very often the usb connection works and I can send and receive.

###### Setup
* platformio
* Example (Arduino sketch and Android app) that I'm using ArduinoBlinkLED
* USB-Host-Shield-20 (most current from platformio) 
* Arduino Mega ADK
* Galaxy Nexus with Cyanogen 11 (Android 4.4.4)
* Nexus 6 with stock ROM, Android 5.1.1

Lauszus, 2015-12-06:
To be honest this sounds like a issue with your Android phone and not the code, as it works for your Galaxy Nexus.

closed 2015-12-06 by Lauszus

thaussma, 2015-12-06:
Actually I expected professional help in debugging this issue or some tips and hints.

xxxajk, 2015-12-07:
I say talk to the company that made the phone. The problem is on their end,
not with the library. All of my devices work here, thus you already got
your answer.