felis/USB_Host_Shield_2.0 #189: Missing wheel mouse reports in HID full mode (however HID boot mode reports OK)

joseangeljimenez, 2015-12-06:
We have been working with several different USB mouse devices, about 10 in total, testing the behaviour of the wheel and how the mouse reports the wheel (Z axis) movements.

What we found so far:
- Even tough the HID boot protocol should not include the wheel byte in the reports, we have found that about 40-50% of our tested devices DO report a fourth byte with the wheel byte counter, which is nice.
- Again, in HID protocol, thouse devices reporting the fourth/wheel byte, do it as expected, without missing any movements coming from the wheel. The reports will include slow and fast wheel movements, without losing events.
- When using the full/complete HID mode, all 100% tested devices report the fourth/wheel byte. However, now the devices will not report correctly the wheel movements. Some of the movements coming from the wheel are missed, not reported. The rest of X, Y, Z movements, however, are timely reported without trouble.
- We have tested the both modes above with the examples included in the USH Host Shield library, with the debug mode active, without adding/modifying a single line of code.

Questions:
- Could this be a subtle bug in the library?
- How can we dig deeper into the subject. Any help is welcome.
- Is there any additional (programatic) step that we have to carry to instruct the mouse to report the wheel in HID full mode? Sending an specific low-level command for the USB device so that it better reports the wheel movement?

Lauszus, 2015-12-06:
Have you looked at more than the first four bytes? Maybe it is sending out some more information.

Also have you replaced the following line: https://github.com/felis/USB_Host_Shield_2.0/blob/ad966a5fb1bda601daa394da7b8c99af3efa3278/hidboot.h#L459 with:
```C++
rcode = SetProtocol(i, HID_RPT_PROTOCOL);
```

Or are you calling SetProtocol: https://github.com/felis/USB_Host_Shield_2.0/blob/ad966a5fb1bda601daa394da7b8c99af3efa3278/hid.h#L176 within your Arduino code?

joseangeljimenez, 2016-01-30:
I finally could take some time to continue testing this. Here are my findings,

A) If I modify hidboot.h as you suggested, I get the desired results, perfectly receiving the fourth byte (wheel) in the HID report. Great!

B) However, if I call SetProtocol() after the initialization of the USB object, I don't receive the fourth byte. This is the code I am using,
```
static USB Usb;
static HIDBoot<HID_PROTOCOL_MOUSE> HidMouse(&Usb);
if (Usb.Init() == -1) {
  log("USB Host Init Error");
  return false;
}
HidMouse.SetProtocol(0, HID_RPT_PROTOCOL);
```

C) I also tried with HIDUniversal instead of HIDBoot. Again with no success.
```
static USB Usb;
static HIDUniversal HidMouse(&Usb);
if (Usb.Init() == -1) {
  log("USB Host Init Error");
  return false;
}
HidMouse.SetProtocol(0, HID_RPT_PROTOCOL);
```

Your comments are welcome. Is it possible to work out the approaches B or C? Am I doing something wrong or is it the approach A the only one valid?

Lauszus, 2016-01-31:
@joseangeljimenez no you will have to modify hidboot.h. Alternatively you could make the class take HID_RPT_PROTOCOL as an argument. A pull request is more than welcome :)

I will close this issue for now, as I guess you problem has been solved.

closed 2016-01-31 by Lauszus

joseangeljimenez, 2016-02-01:
Thanks. I will make the pull request in short as I think this is a nice feature to have in the library (having the possibility to Init the device alternatively in boot or full mode.