felis/USB_Host_Shield_2.0 #192: Xbox Recv issues

mcgeheeiv, 2015-12-15:
Good Afternoon Everyone!, 

Before seeking online help.. I did a bit of research and tried many of the problem solving issues that others have seen. I have no idea what I am doing wrong but something so simple just fails to work and I'm wondering at this point, if one of the components do not work.

I tried going off existing code (multiple web attempts) for xbox 360 recv, the examples here and even tried editing with no success. Pretty much my official xbox 360 controller blinks and blinks.. my official xbox controller recv blinks and blinks. The controllers do not turn solid green and I even tried pushing all buttons and the sync button with no success. 

Example of what I have: 
USB Host Shield (circuits@home)
Xbox 360 recv attached and blinks but does not go solid.
_I__I__I__I__
Peasy breakout board (Pin 9 and 10 are used for two esc motor controls that go to DC motors) 
Note: I have tried to use other pins and motors work but the recv does not communicate.
_I__I__I__I___ 
Arduino Uno R3 Plus by OSEPP and have tried changing to Arduino Ethernet as well.
_I__I__I__I___ 

This is what I am using and have tried both the Arduino Uno to PC and Arduino Ethernet with serial cable to PC..
Arduino IDE 1.6.5 and tried 1.6.6.
Arduino Uno R3 Plus OSEPP and official Arduino Ethernet (Note: I only use one of these but tested both)
Peasy Breakout Board 1.0 with 2x dc motors with esc controllers pins 9 and 10 (tank style).
Circuits@home USB host shield with xbox recv

No fancy lights or wiring.. just the 3 boards with pin 10 and 9 on the peasy breakout board (ground for ground, power for power and signal for the yellow/white signal cable.  

Also, I have tried a battery pack with 5x AA batteries and a 12v car type battery with a circuit breaker setup.. pretty much throws the power potential problem out the window. 

I'm looking forward to hearing back and hope someone can help.

Best Regards,

Edgar McGehee

felis, 2015-12-15:
Did you move pins 9, 10 on a shield to some other Arduino pins? Hw manual here -> https://www.circuitsathome.com/usb-host-shield-hardware-manual

mcgeheeiv, 2015-12-15:
Hi Felis, 

This looks like it could be my issue from reading your link. Also, the following in no way means to criticize any product or company but I just want to learn why and how because this helps me better learn/understand. 

I tried using other pins but after reading this manual, pins on the uno might need to cross over to match the USB host shield, if I understand it? How do I go about doing this when stacking? Also, if Arduino uno is a common board with pins 11-13.. why would circuits at home make a board that makes a cross over to 9 and 10? What can I do to solve this? The manual here does not explain it well for someone new to this. 

Note: I'm not completely new and even got into Kinect coding a bit but being self taught hurts at times. 

Thanks!

felis, 2015-12-16:
If you need pins 11-13 for something else then you can't do it on
standard-size Arduino and you need to migrate to a larger board. On boards
like Mega pins 11-13 are not shared with SPI, then you only need to
re-route 9 and 10, per manual, under 5.

mcgeheeiv, 2015-12-16:
I don't have preference with pins and can use the motors on any allowable pin. I really want to avoid crossing pins because this makes stacking difficult. Any way around this?

1) What pins should I use for motors that you recommend for Arduino Uno that will not conflict with this USB shield? 

2) Will I still need to re rout 9 and 10 from the usb host shield to to 5 on arduino uno? How do I stack two pins (9 and 10) into one (5) and do you have a diagram of what this looks like? 

3) why would the circuits at home use a board setup that does not stack with a common board? This seems like something that should be fixed in the future or is there a reason?

felis, 2015-12-16:
1) If you can allocate pins 9-13 to the shield everything should be good.
3) depends on what is considered a common board. Here I give examples of stacking USB Host Shield with several other boards -> https://www.circuitsathome.com/mcu/running-multiple-slave-devices-on-arduino-spi-bus

I don't understand 2)

mcgeheeiv, 2015-12-16:
1) I just wanted to confirm.. do not connect motors to 9-13 because the shield will use them with pins 5 and 6 ok for motors or any other pin than 9-13. Does this mean that I can sandwich the 3 boards (arduino uno, motor shield, host shield) and not require re-rout or do I still need to do that? If so.. what pins do I need to cross to what other pins. I don't know, if I'm not asking this correctly but the answer does not seem clear. 

2) was in relation to #1 and what you said about "On boards like Mega pins 11-13 are not shared with SPI, then you only need to re-route 9 and 10, per manual, under 5. 

3) I was just curious to why different companies change pin numbers to make things incompatible and is this due to design patents? Just seems crazy that there is not a set standard across boards or if there is a purpose. 

Thanks in advance for the patience and I'm very grateful for the support.

xxxajk, 2015-12-16:
>
> 1) I just wanted to confirm.. do not connect motors to 9-13 because the
shield will use them with pins 5 and 6 ok for motors or any other pin than
9-13. Does this mean that I can sandwich the 3 boards (arduino uno, motor
shield, host shield) and not require re-rout or do I still need to do that?
If so.. what pins do I need to cross to what other pins. I don't know, if
I'm not asking this correctly but the answer does not seem clear.

If motor does not use 9-13, and does not use the icsp connector you are ok.
You can stack in any order you like.

> 2) was in relation to #1 and what you said about "On boards like Mega
pins 11-13 are not shared with SPI, then you only need to re-route 9 and
10, per manual, under 5.

On ALL boards thi icsp connector is spi. On UNO these are duplicated on the
side. On the mega, they are duplicared on the rear connector.

> 3) I was just curious to why different companies change pin numbers to
make things incompatible and is this due to design patents? Just seems
crazy that there is not a set standard across boards or if there is a
purpose.

Usually it is due to lack of forsight and/or not knowing how boards are
different. As stated above, icsp is the only real universal connection for
spi. Other boards that do not use it are in effect UNO only.

mcgeheeiv, 2015-12-16:
Thanks! I'll try this in the morning when I get to the  lab.

xxxajk, 2015-12-16:
You are very welcome.

mcgeheeiv, 2015-12-16:
Wow.. I finally understand. Someone in the forum mentioned 5) but I get it
now. I just don't know what that physically looks like arduino uno to the
usb shield. Do you have a diagram?

xxxajk, 2015-12-16:
http://www.pighixxx.com/test/wp-content/uploads/2014/11/uno.png
Should be obvious after you look at the ICSP connector at the bottom of the picture.

xxxajk, 2015-12-16:
If not, compare to this one
http://www.pighixxx.com/test/wp-content/uploads/2014/11/mega.png

Notice where SPI is...

mcgeheeiv, 2015-12-16:
I looked at everything and I see where the ICSP (6 pins) structure is at the bottom. 
There is clearly something that I am missing..  

This is my Arduino Uno R3 from the two links.
http://osepp.com/wp-content/uploads/2012/06/unor3plus_new.jpg
http://osepp.com/wp-content/uploads/2012/10/OSEPP_Uno_R3_Plus_Arduino_Compatible-Rev3.0-Schematic.pdf 

Peasy breakout for motors passes directly through with the Arduino (so.. the conflict is only arduino and the usb host shield).

I understand that the USB Host Shield uses 9 and 10 Int & SS. 

However, even with all this information.. reading the manual and understanding the changes required in theory with the changes required in the library, I can't seem to find how I am supposed to jump cables in the real world. I thought about using the breakout board to do the rerouting but something still looks off. 

Is there a Arduino board that will work with this host shield straight out of factory without rewiring or any insanity? I keep searching online with no layman understanding to what rerouting actually looks like. 

Thanks as always!

Lauszus, 2015-12-16:
Have you read the following section of the readme: https://github.com/felis/USB_Host_Shield_2.0#interface-modifications and the guide written by Oleg: https://www.circuitsathome.com/usb-host-shield-hardware-manual?

It explains how to do the modification if you want to rewire SS and INT.

mcgeheeiv, 2015-12-17:
Yes, but the instructions were not enough for someone like me that is semi
new to electronics. In theory.. I understand everything but practical or
actual placement of cables and spots on a proto shield or breadboard is
another thing.

Although, I just bought a mega 2560 because the circuits@home site says
that it will work out of the box with the usb host shield to not require
the headache and I'll test it tomorrow morning. However, I could be wrong
but in theory the diagram for the 2560, looks like the ss and int slots are
19 and 20. I don't see how this directly connects without rerouting. I wish
there was a diagram that shows what pins need to go where to make this
work. Also, how the ICSP works on it or if it needs to be connected or more
details.

Lauszus, 2015-12-17:
All Arduinos will work out of the box with the USB Host shield from Circuits@Home. They will use pins 9 and 10 for INT and SS (slave select) by default. Furthermore it will use the SPI interface which is located at pin 13-11 on the Uno and 52-50 on the Mega. Please refer to: https://www.arduino.cc/en/Reference/SPI. The USB Host shield connect to these pins using the ICSP connector, as mentioned already.

Note that the SS pin can be any pin and only need to be the hardware SS if using the Arduino as SPI slave, which you will not use in this case.

Please try the example with only the shield attached and confirm that it is working. Then try with the motor shield only and confirm that you can control two motors for instance. Now attach both of them and do the hardware modifications if necessary.
You will have more success if you do small steps at a time instead of trying to do everything at once.

The reason why motor shields often use pin 9 and 10 on the Uno is because these two pins are connected to a 16-bit timer (Timer 1) on the Uno which give you more flexility when setting the frequency, higher resolution etc.

mcgeheeiv, 2015-12-17:
Wow.. it's alive!!! I have no idea why but they are now both working (Uno R3 Plus by OSEPP and Mega 2560 clone from Inland). Something might be off with my Uno because I tried the Mega and it worked right away and when I went back to try the Uno for being curious.. it worked as well. You guys are awesome for the support and hope I was not too much of a pest. If I was to guess, it could be the Uno's ICSP pins have a faulty contact because the other slots seemed to work. Either way...... just happy it works. Many thanks once again to each of you! I'll add thanks with the library and code credits as well for the support.  

Best Regards, 

-Eddie

closed 2015-12-17 by mcgeheeiv

Lauszus, 2015-12-17:
@mcgeheeiv good to hear you got it working :)