felis/USB_Host_Shield_2.0 #193: Specify on readme, galileo and due no native usb

AndresBott, 2015-12-21:
sadly that happened, to me. i bought a DUE thinking i could use this library with the native USB, after some time looking arround i figured out, that native support is not jet there...

i looked into the code, to see if I could addapt it, but i think i won't be able...

if indicated in the spects,  at least, no one other makes the same mistake I made.

xxxajk, 2015-12-21:
UHS 2.x/1.x never said it was for Due, ever.
However Due native USB is planned for UHS 3.0.
You are welcome to contribute, or even test what is there currently.
Please note that the main focus of effort is on the USB Host Shield.

https://github.com/felis/UHS30

Lauszus, 2015-12-21:
Commit 30ec2e942fabb8a21b1e10ed9aad5f7e24199703 should fix this issue.

commit 30ec2e94 (2015-12-21): Added note about that native USB Host is not supported

Fixes: #193
closed 2015-12-21 by Lauszus