felis/USB_Host_Shield_2.0 #194: Barcode scanner problem

kemalmuderrisoglu, 2015-12-24:
Hello. I've used USBHIDBootKbd sample codes on two barcode readers. My USB HOST SHIELD connected on Arduino UNO. For 1st barcode reader its worked and i've seen the barcode on the serial monitor but for the second one the barcode reader locked itself. Do you have any idea about this issue?

felis, 2015-12-24:
I think the second unit is not suitable - don't use it.

kemalmuderrisoglu, 2015-12-24:
Well, for my project it should support any kind of barcode reader. According to my researches there are 3 types of USB 2.0 which are Low Speed, Full Speed, High Speed. So i'm thinking about maybe this USB Shield don't support the higher levels. Do you have an idea about its speed?

Lauszus, 2017-10-12:
Closing this issue, due to the age of the issue. Please reopen if needed.

closed 2017-10-12 by Lauszus