felis/USB_Host_Shield_2.0 #201: New RBL SDK breaks USB Host Shield

sieren, 2016-01-16:
The new RedBearLabs SDK for the nRF51288 breaks compatibility with this library,
mainly because:
- APIs for SPI changed
- Enum collisions for LEDs (LEDEnum controllerEnum.h vs. some enum in the SDK)

I'll just leave this here as a note for others, I'm already working on deploying a fix (changing the SPI definitions for RBL and moving the affected Enums into a namespace).
Currently WIP at: https://github.com/sieren/USB_Host_Shield_2.0/tree/fix-new-rbl-sdk

Will file PR when done.

Lauszus, 2016-02-25:
@sieren any updates?

sieren, 2016-02-26:
@Lauszus still working on it. Havent forgotten about it ;)

sieren, 2016-02-26:
FYI: I got everything working EXCEPT the GPIO Test which fails:

GPIO test. Connect GPIN0 to GPOUT7, GPIN1 to GPOUT6, and so on
Test failed. Value written: 00 Value read: FF 
Press any key to continue...
GPIO test passed.

Hard to tell what's going wrong there, to be fair I cannot remember whether it worked in the first place. Do I need to physically connect the GPIOs with each other in order for board_qc to pass?

Lauszus, 2016-02-26:
@sieren yes, but don't worry about it. This is done by @felis during production to ensure that all shields are working correctly.

sieren, 2016-02-26:
Yeah, I figured after I thought about it a bit more thoroughly ;)

Anyway, PR is filed: https://github.com/felis/USB_Host_Shield_2.0/pull/215

Not too happy about the namespace enum - almost tempted to give RedBear a shout to come up with something more unique.

Lauszus, 2016-02-26:
@sieren hmm I don't think it will affect the end user, so I think we will be fine?

Lauszus, 2016-02-26:
@sieren I actually think we should put all the enums in controllerEnums.h in the namespace? That would make it more consistent.

sieren, 2016-02-26:
Nope, had to update the examples though to point to `Controller::LED1` instead of just `LED1`.

Lauszus, 2016-02-26:
@sieren yeah okay, then I would not be happy to change that, as it will break a lot of code that is currently used and floating around the internet.

sieren, 2016-02-26:
@Lauszus Yeah, may be a good idea in general, even if its purely to avoid interference with other libraries.

sieren, 2016-02-26:
I agree... I'll investigate what the point of that enum is on RBLs end and maybe file a PR there so we can drop the commit.

Lauszus, 2016-02-26:
@sieren okay thanks. Really appreciate it!

Lauszus, 2016-02-26:
So I guess we can just close the PR?

sieren, 2016-02-26:
The problem is actually caused by mBed, which is supported by an increasing number of Arduino boards these days:
https://github.com/mbedmicro/mbed/tree/master/libraries/mbed/targets/hal
all of these targets have a PinNames.h which includes an unscoped enum with a LED1-4 definition.

I suppose keeping LED in the USBH library would essentially break potential future support for quite a few boards out there.

Lauszus, 2016-02-26:
Hmm maybe we could add a define, so the namespace will only be used on those processors? We will have make a define for the enum as well.
The only modification would using the drivers, but I would much rather have that than add the namespace, as I will have to use a lot of time supporting noobs struggling with their code because the namespace was added :)
Please using mbed or similar will often also be able to figure it out themselves compared to most Arduino users.

xxxajk, 2016-02-26:
Or do like UHS 3 does-- prepend common names with 'UHS_'

Lauszus, 2016-02-26:
@xxxajk another reason why just leaving it as is, as UHS3.0 already fixes this.

xxxajk, 2016-02-26:
Yes and I'm waiting for you to pick out some drivers lol

sieren, 2016-02-26:
@Lauszus Yeah, the issue in this case is that the Arduino Library is wrapping the mBed one.

Anyway, I could add a define to the Controller driver for the namespace/enum, that would break the examples though if targeted on RBL, but might weigh against the numbers of issues youd receive from noobs needing help with the namespace ;)

Lauszus, 2016-03-08:
@sieren yeah I really don't want to change all the examples, as it will require so much work on our side. If you send a PR that only affect RBL and mbed I would be happy to merge it.

@xxxajk I know ;)

sieren, 2016-03-08:
Makes sense. I'll drop that commit and will push a RBL specific fix later this week.

sieren, 2016-03-23:
Since this is merged to master now, I'll go ahead and close it :+1:

closed 2016-03-23 by sieren