felis/USB_Host_Shield_2.0 #203: ACM save device descriptor

drjrkuhn, 2016-01-21:
I have been doing some tests with my ACM-derived device, and it seems like the device descriptor can only be checked during the Init() phase (set address 0, get USB_DEVICE_DESCRIPTOR ). I would like to save and later check the vendor id, product id, etc from the device descriptor, but the only way I can figure out to do this is to overwrite the ACM::Init code with my own slightly-modified copy/pasted version of Init().

Is there any way to universal way to save the device descriptor information during Init? Perhaps, by letting a device set a pointer to a global address or class member pointing to a USB_DEVICE_DESCRIPTOR that will be filled during the Init phase.

Lauszus, 2016-01-21:
One easy option would just to make udd a public variable: https://github.com/felis/USB_Host_Shield_2.0/blob/master/cdcacm.cpp#L51. You could then simply use ```isReady()``` in order to check if the device has been initialised or not.

drjrkuhn, 2016-01-21:
Yes. I can re-implement `ACM::Init()` in my own derived class to have `udd` point to a class member variable instead of a `buf` local to the init function. That's how I do it now. 

I was wondering about a more universal method, such as a new virtual method in `USBDeviceConfig` like 
```cpp
    class USBDeviceConfig {
    ...
        virtual bool USBDeviceConfig::UDDOK(USB_DEVICE_DESCRIPTOR& udd) { 
            return true; 
        }
    ...
    };
```
which would be called during `USB::Configuring()` before `VIDPIDOK()`, `DEVCLASSOK()`, and `DEVSUBCLASSOK()` -- sort of a super `xxxOK()` check during `Configuring()`.

Overriding the `UDDOK()` method would allow any subclass of `USBDeviceConfig` to save the entire device descriptor by copying the whole structure, or just parts of it by copying pieces.

drjrkuhn, 2016-01-21:
I have not yet done git pull requests before, but I'll write a trial implementation.

Lauszus, 2016-01-21:
That sounds like a good idea. Looking forward to your PR :)

drjrkuhn, 2016-01-22:
I added the code and sent a pull request, but I don't know if I did it correctly. (I'm more of a subversion person.)

Lauszus, 2016-01-22:
You sent the PR to your fork and not this repository: https://github.com/drjrkuhn/USB_Host_Shield_2.0/pull/1, but why are you using a reference to the USB_DEVICE_DESCRIPTOR? Don't worry about the indentation for now. I will fix that.

drjrkuhn, 2016-01-22:
OK. I need to learn how to git properly. :-) Thanks for taking care of my mis-PR and changed indentation!

Here is my justification for passing a reference to large structures instead of a pointer (in general):

Passing a large structure on the stack is a bad idea because it wastes cycles and memory. So, the choice becomes pointer or reference. They have the same footprint, but the compiler treats them differently. A reference is a better contract between the caller and the function -- a contract that says the value passed will never be NULL. It is very hard to get a compiler to pass a reference to a NULL to a function, but *very* easy to pass a NULL pointer to a function. 

A constant reference is even safer, since the function can't change the structure, just see it.

```cpp
void fooPtrUnsafe(const BAR* pBar) 
{
    int i = pBar->value1;  // pBar could be NULL. So this is very unsafe
}

void fooPtrSafe(const BAR* pBar)
{
    if (pBar) {
        int i = pBar->value1;  // We checked that pBar is NULL, so this is safe
    } else {
        // But, pBar could have been NULL, so the check and 
        // even the entire function call was essentially wasted effort.
        // No information got through. But we had to check anyway
    }
}

void fooRef(const BAR& bar) 
{
   int i = bar->value1; // Much safer.  bar is never NULL.
}
```

At least, that's my understanding.

Lauszus, 2016-01-24:
Why would you want a pointer to begin with? A pointer would not even work, as it will become invalid once the variable is out of scope. The reason why a reference work is explained in this answer: http://stackoverflow.com/a/57734. Basically the life-time of the variable is associated with the reference instead.

Anyway a reference is fine. Just send a PR and I will be happy to merge it :)

Lauszus, 2016-01-24:
Btw if you are new to Git I would recommend the following GUI: https://www.sourcetreeapp.com/ :)

drjrkuhn, 2016-01-24:
More good points in the stackoverflow post about references and lifetime. Thanks!

**About integrating `DEVDESCRIPTOROK`**

I thought about this proposed `DEVDESCRIPTOROK` method for a while, and I asked myself if I *really* needed that extra information in the device descriptor. The iManufacturer, etc are really just indices for further "get" operations rather than actual values. They don't carry useful information, and it would take extra steps to get those device strings anyway.

Once I delved into the code a little more, I realized that the relevant target device information can be retrieved during the `VIDPIDOK`, `DEVCLASSOK`, and `DEVSUBCLASSOK` checks! I think my proposed changes are superfluous and I really can't justify the addition. Sorry to have wasted time on it!

**tldr; Don't integrate it after all.**

closed 2016-01-24 by drjrkuhn

Lauszus, 2016-01-25:
@drjrkuhn okay no problem. Just open another issue if you want anything else implemented in the future.