felis/USB_Host_Shield_2.0 #205: HID_Command delay issue in PS3BT.cpp

doogerk, 2016-01-26:
In the PS3BT.cpp file, the HID_Command requires a 150ms delay between commands. Instead of using a timer based comparison in the overall program loop, it uses a hard delay. A delay of this size can cause issues when used in conjunction with other high speed protocols (i.e CANbus)

Lauszus, 2016-01-26:
@doogerk a work around would be for you to check in your code that you are not turning the LEDs on and off to quickly for example. Can you share your code?

doogerk, 2016-01-26:
I tried turning the leds on and off in a state machine type setup but the individual up to 150ms delay is still too long. I am using an Arduino Due with the usb host shield and I am using the two built in CANbus channels to form an inline data modifier for a CAN driven device. It lets all the necessary traffic pass through and I manipulate the data frames I'm interested in or inject new data. The attached program has provisions to be controlled by both a PS3 controller or android phone using SPP. Everything works as expected except when I try to set the PS3 controller leds. I assume its related to the up to 150ms delay in the HID_Command. The problem lies in that the CANBus sends messages fairly fast, probably closer to the 20-50ms range. A hang-up of more than that can cause issues, especially when working as an inline traffic controller instead of a separate CANBus node. I can't think of an easy fix.

[Quandroid.txt](https://github.com/felis/USB_Host_Shield_2.0/files/105284/Quandroid.txt)

xxxajk, 2016-01-26:
Can you do the LED work in an ISR? That may work for you.

doogerk, 2016-01-26:
I just added a millisecond check in between commands to make sure I send each command at least 150ms apart so the delay is never triggered and that seems to work

closed 2016-01-26 by doogerk

xxxajk, 2016-01-26:
Good to hear can you share the code it may help others

doogerk, 2016-01-26:
[Quandroid1_0.txt](https://github.com/felis/USB_Host_Shield_2.0/files/105385/Quandroid1_0.txt)

Lauszus, 2016-01-26:
@doogerk good to hear you got it working :)

doogerk, 2016-01-27:
Thanks for the help!