felis/USB_Host_Shield_2.0 #206: Question: Sending Value to FTDI

stumpja, 2016-01-27:
I have been trying to use this library to send values to FTDI device (FT232RL) which is connected to a mircrocontroller. Basically, I want to send a number followed by a carriage return (\r\n). Sending 1\r\n would be a good starting point.

Eventually, I will hook up 8 of these devices to an UNO, and send numbers to individual devices, but for now, just sending to 1 will work. Reading from it would be nice, but again, a starting point will be just sending to it.

I apologize for my lack of knowledge. I would sincerely appreciate anybodies input. The example I am trying right now is the FTDI loopback, but I havn't really figured it out yet.

Thanks for any help in advance,

Josh

Lauszus, 2016-01-28:
@stumpja so you have tried the following example: https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/ftdi/USBFTDILoopback/USBFTDILoopback.ino. What problems do you have? Please be specific.

stumpja, 2016-01-28:
@Lauszus,

First off, thanks for the response. I have been to many sites where help seems to be few and far between, so I genuinely appreciate a response.

OK, yes, this is the example I have started with. My hardware is a Circuits@Home Rev2 USB Host Shield connected to an Arduino UNO. The device I want to communicate with is an ATMEGA128 interfaced with a FT232RL usb to serial device (I'll refer to this as "LOAD" going forward) running at 9600 baud. Presently, I can use a serial terminal on a PC to communicate with the LOAD and change values. As stated above, it uses a number following by "enter". It parses these numbers and then does something. So my goal is to tell the UNO to send a value to the LOAD. Eventually, I want to have several LOADs connected through a USB HUB to the host shield, but 1 step at a time.

So when I run the stock USBFTDILoopback example linked above, I get the following output in the serial terminal

> Start
> 0000: 09 02 20 00 01 01 00 A0 2D 09 04 00 00 02 FF FF 
> 0010: FF 02 07 05 81 02 40 00 00 07 05 02 02 40 00 00 
> .......................................................... 

I continuously get decimals. It seems like it is doing something. At this point I assumed my USB Host shield is working, I just needed to change the code to get it to do what I want. So I changed line 67

`char strbuf[] = "DEADBEEF";`

to this:

`char strbuf[] = "1\r\n";`

I got the same output as above. it actually wasn't till this morning that I noticed another setting to change, what I assume is, the Host Shield baud rate. Next, I changed line 22

`rcode = pftdi->SetBaudRate(115200);`

to this:

`rcode = pftdi->SetBaudRate(9600);`

Presto! The load responded properly! So now my next step is to plug in a hub to the USB host shield and try to control 2 individual loads separately. I have written control software for the loads and I talk to them individual using a serial number programmed into the FTDI. Can I do something similar with this code. I see the following code in cdcftdi.h

>       virtual uint8_t GetAddress() {
>               return bAddress;
>       };

I will start here, but I am not sure this will do what I want it too. In my control software, there is an FTDI dll that bascially lets me determine what virtual COM port a LOAD was attached to based on the serial number mentioned above, a serial number that the user can program. i want to be able to do this automatically.

Anyways, I will play around with this today and post my results. But if you have any clues, it would be much appreciated.

Thanks again!

Josh

Lauszus, 2016-02-01:
You properly want to read the iSerialNumber from the device descriptor and then read the corresponding string descriptor using the ```getStrDescr``` function: https://github.com/felis/USB_Host_Shield_2.0/blob/master/Usb.cpp#L796-L798.

You can find a list of the language ID's here: http://www.usb.org/developers/docs/USB_LANGIDs.pdf. I guess you can just use 0x0409 which is English (United States). You can also read a list of supported languages from index, but 99.9% of all device will support 0x0409. Please see: http://www.beyondlogic.org/usbnutshell/usb5.shtml#StringDescriptors.

I guess you can just set the number of bytes to 255 and see if it works.

stumpja, 2016-02-09:
@Lauszus,

Thanks for the info! I will truly need it. However, I've been stumped trying to figure out another issue right now, that I need to solve before moving on to what you've mentioned above. Maybe you can provide some insight.

I've modified the usbLoopback code to allow me to enter values from the serial monitor to change what is being sent. It works, sort of. My fully modified code is here:[uno2load.txt](https://github.com/felis/USB_Host_Shield_2.0/files/123721/uno2load.txt)

Basically, I wait for a value to be entered then send it to the load. My results are strange, however. 

```
Start

|--Loop Count = 0----------------------------|
Waiting for data
1

|--Loop Count = 1----------------------------|
Waiting for data
2

|--Loop Count = 2----------------------------|
Waiting for data
3

|--Loop Count = 3----------------------------|
Waiting for data
4
0000: 09 02 20 00 01 01 00 A0 2D 09 04 00 00 02 FF FF 
0010: FF 02 07 05 81 02 40 00 00 07 05 02 02 40 00 00 

|--Loop Count = 4----------------------------|
Waiting for data
5
.
Send Data To LOAD

|--Loop Count = 5----------------------------|
Waiting for data
8197
.
Send Data To LOAD
1.0.0.0

|--Loop Count = 6----------------------------|
Waiting for data
```

In my test (results above), I enter 1, 2, 3, 4, 5, & 8197 respectively. The LOAD doesn't receive a value till 5 is sent. 8197 is something that requests a value from the LOAD, which it does receive (the string 1.0.0.0). So what I am trying to figure out is why it takes this long for the Arduino to send a value with the host shield?

When I connect a hub, it takes longer that 5 values entered, and the values are delayed once they finally start getting sent. IE if i send a 1, then a 2, the 1 doesn't get sent till the 2 is entered. Anyways, this may be a different issue, so 1 at a time.

Any insights would be very helpful.

Thanks in advance,

Josh

stumpja, 2016-02-16:
@Lauszus,

In an attempt to move on, I started down the path you mentioned in your comment. I have been able to read the iSerialNumber and the corresponding string value. I also know the UsbDeviceAddress. I assume this is what I will need to select the 'LOAD' I want to talk too. I'm not sure how to use it to do this yet. Can you point me in the right direction?

Thanks,

Josh

Lauszus, 2016-02-17:
@stumpja you should just use the iSerialNumber, as the device address change depending in the order to plug in the devices, but the iSerialNumber should be unique to each device.

I got one comment regarding this piece of code:
```C++
for( i = 2; i < length; i+=2 ) {
   Serial.print((char) buf[i]);
 }
```
Why do you iterate using ```i+=2```? Won't you skip every second character that way?

Lauszus, 2016-02-17:
Ahh the string is UTF-16 encoded ;)

stumpja, 2016-02-17:
@Lauszus I am missing something. So the iSerialNumber seems to be the serial number index of the string descriptor. For all of my FTDI devices, this index value is 03. 

The corresponding string value is what I set the FTDI as. In my case, I programmed the serial numbers to be 'LOAD_000' through 'LOAD_007'. They will all be unique because there will never be any more than 8 "LOAD's' used together and I will program each serial number. 000 represents CH1 while 001 will represent CH2 and so on up to 007 which is CH8. 

Below is a snippet of my serial port debug section. I have commented out all of the device descriptors because it makes it messy very quickly.

>String Descriptors:
>Product:		USB 2.0 Hub
>
>String Descriptors:
>Product:		USB 2.0 Hub
>
>String Descriptors:
>Manufacturer:	FTDI
>Product:		FT232R USB UART
>Serial:		LOAD_000
>
>String Descriptors:
>Manufacturer:	FTDI
>Product:		FT232R USB UART
>Serial:		LOAD_007
>
>String Descriptors:
>Product:		USB 2.0 Hub
>
>String Descriptors:
>Manufacturer:	FTDI
>Product:		FT232R USB UART
>Serial:		LOAD_004
>
>String Descriptors:
>Manufacturer:	FTDI
>Product:		FT232R USB UART
>Serial:		LOAD_001
>
>Addr:41(1.0.1)
>
>Addr:4A(1.1.2)
>
>Addr:A(0.1.2)
>
>Addr:B(0.1.3)
>
>Addr:53(1.2.3)
>
>Addr:19(0.3.1)
>
>Addr:1C(0.3.4)

As you can see, I have 4 'LOAD's' plugged into my USB Hub. Apparently my 10 port hub is made up of multiple hubs. The string descriptors are at the top, while the addresses are at the bottom. So bascially LOAD_000 has an address of A, LOAD_007 has an address of B, LOAD_004 has an address of 19, and LOAD_001 has an address of 1C, The numbers beside the address are the HUB value (1 saying its a hub), the parent value and the device value from what I can gather.

Below is a snipped of LOAD_000's device descriptors for reference.

>41
>--
>
>Device descriptor: 
>Descriptor Length:	12
>Descriptor type:	01
>USB version:		0200
>Device class:		09
>Device Subclass:	00
>Device Protocol:	00
>Max.packet size:	40
>Vendor  ID:		1A40
>Product ID:		0101
>Revision ID:		0111
>Mfg.string index:	00
>Prod.string index:	01
>Serial number index:	00
>Number of conf.:	01
>
>Configuration descriptor:
>Total length:		0019
>Num.intf:		01
>Conf.value:		01
>Conf.string:		00
>Attr.:			E0
>Max.pwr:		32
>
>Interface descriptor:
>Intf.number:		00
>Alt.:			00
>Endpoints:		01
>Intf. Class:		09
>Intf. Subclass:		00
>Intf. Protocol:		00
>Intf.string:		00
>
>Endpoint descriptor:
>Endpoint address:	81
>Attr.:			03
>Max.pkt size:		0001
>Polling interval:	FF

So, this is where I am stuck. I have all the information, the unique items that seems like the one I should use to select which load I want to talk to is the address or the iSerialNumber string descriptor which is what you referenced. However, I don't know what code should be used to do this. 

Now matter what I try, the load with the smallest address is where the data gets sent. Where exactly do you change what load is selected.

What I have tried is the following:

>UsbDeviceAddress adr;
>adr.devAddress = [insert address here (per above, A, B, 19, or 1C)]
>
>rcode = Ftdi.SndData(strlen(databuf), (uint8_t*)databuf);

I tried changing the address right before I send the data, but this isn't working. I haven't figured out a place in the code where I can change the iSerialNumber string descriptor. Can you provide a little more detail?

Thanks again,

Josh

stumpja, 2016-02-17:
@Lauszus ok, I believe this is what I need to use. Can you verify? [https://github.com/felis/USB_Host_Shield_2.0/blob/master/Usb.cpp#L80-L118](url) (Can't get the link to work like you did), but basically I found a Usb.SetAddress command. I just can't figure what values to put for the 3rd and 4th arguments. 

I've tried Usb.SetAddress(0x1C,0), but it seems the other arguments are not optional. Am I on the right path?

Thanks,

Josh

stumpja, 2016-02-19:
@Lauszus Well I still haven't been able to figure it out. I have tried using Ftdi.Init(uint8_t parent, uint8_t port, bool lowspeed) but this returns an error that it has already been initialized, it doesn't set the arduino to talk to this address. Next I tried Usb.DefaultAddress, AttemptConfig and Configuring, but these all return an error 0x0D. I found this error, but it didn't make sense to me. I haven't been able to successfully try Usb.SetAddress because I don't know what to put for the 3rd and 4th arguments, but it looks like this isn't even want I want anyhow.

Anyways, an insight would be greatly appreciated. I'm so close to getting what I want. I have questions on some workarounds that I had to implement, but I'm getting more and more comfortable with this library everyday.
[uno2load.txt](https://github.com/felis/USB_Host_Shield_2.0/files/138300/uno2load.txt)
[uno2load_debug.txt](https://github.com/felis/USB_Host_Shield_2.0/files/138301/uno2load_debug.txt)

Thanks again in advance! Current code is and debug window are linked.

Josh

Lauszus, 2016-02-22:
@stumpja you should not try to init the driver by yourself. The driver will handle that for you. The address will change depending on which order you plug in the FTDI chip, so it has no real use in your application?

In general you should not try to mess with all the low level stuff, as it is already handles by the driver. Except getting the iSerialNumber string descriptor which you need of course.

stumpja, 2016-02-22:
@Lauszus, I still don't see anything that uses the iSerialNumber string descriptor, however. Can you point me to the function that I should be using to change between USB devices? 

I'm sorry that I'm not getting it. I have read it, I should have all the information I need, I just don't understand how to change which device I send to. It defaults to the first device plugged into the hub. And you are right, if I change what order devices are plugged in, their addresses change.

Lauszus, 2016-02-22:
@stumpja you need to create a FTDI instance for each device:
```C++
FTDI          Ftdi1(&Usb, &FtdiAsync);
FTDI          Ftdi2(&Usb, &FtdiAsync);
FTDI          Ftdi3(&Usb, &FtdiAsync);
```
Then you can simply send and receive values from each device independently:
```C++
rcode1 = Ftdi1.SndData(strlen(databuf1), (uint8_t*)databuf1);
rcode2 = Ftdi2.SndData(strlen(databuf2), (uint8_t*)databuf2);
rcode3 = Ftdi3.SndData(strlen(databuf3), (uint8_t*)databuf3);
```

You can get the address from each device using the ```GetAddress()``` function and then use ```Usb.getStrDescr``` to get the iSerialNumber from each device.

You might want to add a the following function to cdcftdi.h:
```C++
bool isReady() {
    return bPollEnable;
};
```
So you can check if a device is connected or not. A pull request with the change is welcome.

stumpja, 2016-02-23:
@Lauszus Hurray! It works! Thank you so much. I just did a quick proof that it will work and I was able to talk to two loads. Now I just need to do some parsing to make the switching between devices automatic and clean up my code (a lot) and I should be good to go. As it turns out, I was making it much more complicated than it needed to be. It makes sense that I would need to create an FTDI instance for each device. I get it (at least better) now.

I like the idea of the isReady() function. I think I can get around it, but I will try it, and possibly make a poll request if I find it will suit me well. Seems like it should get a poll request either way however.

Again, I can't thank you enough. This is an awesome library with phenomenal support. I'm going to close this issue now. I plan to upload my final code for reference if anybody has a similar question in the future, but all the information needed is in @Lauszus comments above.

Sincerely,

Josh

closed 2016-02-23 by stumpja

stumpja, 2016-02-23:
Cleaned up code for reference. It could definitely still be cleaned up, but I am satisfied with it at this point.

```arduino
#include <cdcftdi.h>
#include <usbhub.h>
#include "pgmstrings.h"

// Satisfy the IDE, which needs to see the include statment in the ino too.
#ifdef dobogusinclude
#include <spi4teensy3.h>
#include <SPI.h>
#endif

char databuf[8];
byte x = 0;
byte count = 0;
byte datarx = 0;
byte usbReady = 0;
boolean found = false;
char sn[16];
char ch[] = "00000000";
byte rcode1 = 0;
byte rcode2 = 0;
byte rcode3 = 0;
byte rcode4 = 0;
byte rcode5 = 0;
byte rcode6 = 0;
byte rcode7 = 0;
byte rcode8 = 0;
byte chCase = 0;

byte getdevdescr( byte addr, byte &num_conf );
byte PrintString(uint8_t addr);

class FTDIAsync : public FTDIAsyncOper {
public:
    byte OnInit(FTDI *pftdi);
};

byte FTDIAsync::OnInit(FTDI *pftdi) {
    byte rcode = 0;

    rcode = pftdi->SetBaudRate(9600);

    if (rcode)
    {
        ErrorMessage<byte>(PSTR("SetBaudRate"), rcode);
        return rcode;
    }
    rcode = pftdi->SetFlowControl(FTDI_SIO_DISABLE_FLOW_CTRL);

    if (rcode)
        ErrorMessage<byte>(PSTR("SetFlowControl"), rcode);

    return rcode;
}

USB           Usb;
USBHub        Hub1(&Usb);
USBHub        Hub2(&Usb);
USBHub        Hub3(&Usb);
USBHub        Hub4(&Usb);
USBHub        Hub5(&Usb);
USBHub        Hub6(&Usb);
USBHub        Hub7(&Usb);
USBHub        Hub8(&Usb);

FTDIAsync     FtdiAsync;
FTDI          Ftdi1(&Usb, &FtdiAsync);
FTDI          Ftdi2(&Usb, &FtdiAsync);
[... 336 lines cut ...]
  for (n = 1; n <= num_places; n++) {
    mask = (mask << 1) | 0x0001;
  }
  v = v & mask; // truncate v to specified number of places

  num_nibbles = num_places / 4;
  if ((num_places % 4) != 0) {
    ++num_nibbles;
  }
  do {
    digit = ((v >> (num_nibbles - 1) * 4)) & 0x0f;
    Serial.print(digit, HEX);
  }
  while (--num_nibbles);
}

/* Print a string from Program Memory directly to save RAM */
void printProgStr(const char* str)
{
  char c;
  if (!str) return;
  while ((c = pgm_read_byte(str++)))
    Serial.print(c);
}
```

xxxajk, 2016-02-24:
> /* prints hex numbers with leading zeroes */// copyright, Peter H Anderson, Baltimore, MD, Nov, '07// source: http://www.phanderson.com/arduino/arduino_display.htmlvoid print_hex(int v, int num_places)

Unless you actually use 8 hubs, you only need to have one instance.

stumpja, 2016-02-24:
@xxxajk I thought that at first, but for some reason I need at least 3 of these hub instances to reach all the ports. I have 1 hub, but it is a [10 port hub](http://www.newegg.com/Product/Product.aspx?Item=N82E16817182257&nm_mc=KNC-GoogleAdwords-PC&cm_mmc=KNC-GoogleAdwords-PC-_-pla-_-Hubs+-+Network+%2F+USB+%2F+Firewire-_-N82E16817182257&gclid=Cj0KEQiAxrW2BRCFidKbqKyq1YEBEiQAnMDWxnIzfDkHUB3_YE5SIUoDp-v1nUtUj_21oN6mH4Y6PlQaAjlG8P8HAQ&gclsrc=aw.ds). I assumed that this 10 port hub uses 3 controller IC's making it appear as three separate hubs?

Anyways, I put 8 in my code just in case. I figured it wasn't hurting anything. What negative effects could this cause?

Thanks for taking the time to view this issue! I love this github, very active and helpful!

xxxajk, 2016-02-24:
3 hubs sounds correct. I would eliminate the additional ones to save on RAM.

stumpja, 2016-02-24:
@xxxajk thanks for the info. I have updated my code, and the code in the comments above to reflect your suggestion.

Thanks again!

Josh

xxxajk, 2016-02-24:
Not a problem :-) Next version (3.0) you won't have to worry about it. hubs
will be added as needed as long as you ask to install the code. Although
right now it isn't quite there yet and you currently need to do as in
version 2, I want to eliminate the confusion. Also there's not a lot of
drivers yet though....

Thanks for showing me something like this, it allows me to see how others
think, so that i may improve the next version.

Lauszus, 2016-02-25:
@stumpja if you have time it would be nice if you could update the hub example: https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/hub_demo/hub_demo.ino, so it prints the string descriptors and then send a pull request, as I think it might be useful for others :)

stumpja, 2016-02-25:
@Lauszus absolutely! I will do this as soon as I have some time. Hopefully within a week. I think it would also be nice to have it included in the USB Description example: [https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/USB_desc/USB_desc.ino](https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/USB_desc/USB_desc.ino). Let me know what you think.

Lauszus, 2016-02-26:
@stumpja yes it would be nice to have it included there as well.