felis/USB_Host_Shield_2.0 #209: Correct way of initializing the library

joseangeljimenez, 2016-02-01:
This is a fragment of the setup() code of the example "USBHIDBootMouse" included with the library package,
```
void setup()
{
    // Some lines removed
    if (Usb.Init() == -1)
        Serial.println("OSC did not start.");
    delay(200);
    next_time = millis() + 5000;
    HidMouse.SetReportParser(0,(HIDReportParser*)&Prs);
}
```
My question is about the delay(200) line. Is it really necessary? Where does the 200ms delay comes from? The USB specification? The Maxim IC datasheet?

Also, shouldn't SetReportParser() be called before Usb.Init()?

Lauszus, 2016-02-01:
@felis since you wrote that example maybe you can comment on this?

@joseangeljimenez it does not matter in what order you call ```SetReportParser``` as long as it called before Usb.Task() is called.

felis, 2016-02-02:
I can't. IIRC this example predates Arduino 1.0rc. The best way to find out whether something is "really necessary" today is to remove it and see if the code still works. It works on my machine without a delay :-) but I'd leave it as-is - the delay that is happening once during runtime is not a big deal.

Lauszus, 2016-02-02:
@felis okay thanks.

@joseangeljimenez I will close this issue for now. You're welcome to open up the issue again if you have any questions.

closed 2016-02-02 by Lauszus

joseangeljimenez, 2016-02-03:
@felis @Lauszus The bottom line of my original question is,
- Let's say that the external USB device (mouse) is connected.
- Then, the MCU starts and Usb.Init() is executed. Is the enumeration happening at this place? Or will it happen after we start calling Usb.Task() in the main event loop?
- If the enumeration takes place at Usb.Init(), we should not delay for 200ms, which is a long time in terms of the USB specification. Instead, we should start doing Usb.Task() in order not to miss any deadline in the maximum polling interval suggested by the external USB device.
Your feedback on this is welcome. Thank you!

Lauszus, 2016-02-03:
@joseangeljimenez only the MAX3421E chip is initialised in ```Usb.Init()```. Please see: https://github.com/felis/USB_Host_Shield_2.0/blob/master/usbhost.h.

The device connected is not initialised until ```Usb.Task()``` is called.