felis/USB_Host_Shield_2.0 #211: Accessing the USB stream?

spurra, 2016-02-22:
I've bought the UBS host shield for arduino and am attempting to access the USB byte stream (similar like Serial.read() or Serial.write()). I've come across your library which I think is by the same people who design the shield. However, I cannot seem to find a library function or similar which allows me to do this. Could anyone point me in the right direction?

Lauszus, 2016-02-22:
@spurra all the low-level stuff is in here: https://github.com/felis/USB_Host_Shield_2.0/blob/master/UsbCore.h and https://github.com/felis/USB_Host_Shield_2.0/blob/master/Usb.cpp. Maybe you can try to describe what you want to achieve, then I might be able to help you better.

spurra, 2016-02-23:
Thanks for pointing that out. Basically what I'm attempting is having a device connected to the USB host shield, read out the data it sends, modify it, and send it to my computer.

Lauszus, 2016-02-25:
@spurra if you do not have any experience with the USB protocol it might be a bit complicated, as you need to make the microcontroller show up as a USB device, so the computer see it as a printer and then write a USB host driver in order to communicate with the printer.

closed 2016-02-25 by Lauszus

spurra, 2016-03-06:
I am sorry to reopen this again, but I really need an answer. Is the only thing the files UsbCore.h and Usb.cpp do, is implementing the USB protocol? If that is the case, it should be sufficient to read up on the USB communication protocol to understand the code, correct?

Lauszus, 2016-03-07:
@spurra yes it's only in those files.

Vamshi253, 2019-03-19:
Hi Spurra,

Have you been successful in establishing the communication between Printer and Arduino?

spurra, 2019-03-19:
> Hi Spurra,
> 
> Have you been successful in establishing the communication between Printer and Arduino?

Unfortunately not.