felis/USB_Host_Shield_2.0 #212: when connecting the xbox360 reciver it beeps

orsegev, 2016-02-22:
i really dont know what cuses this problom but at the begining the liabery and reciver worked, after a few trys it just started to cut off and then just stop working after that evry time i try to power it the reciver make noises and beeps, i dont know alot in the whole usb area so i need help

Lauszus, 2016-02-22:
@orsegev are you using an official receiver and the official shield? Also please enable serial debugging: https://github.com/felis/USB_Host_Shield_2.0#enable-debugging and post the output.

orsegev, 2016-02-23:
im using sparks fun usb host shild and the original reciver and i enabled debugging and got the same out put:
Xbox Wireless Receiver Library Started
and thats it, im assuming that by serial debugging im suppose to see something diffrent in the serial right?

Lauszus, 2016-02-25:
@orsegev sorry but we do not support the Sparkfun shield. Please contact Sparkfun support for help. Please consider buying the official shield in the future to support the actual developers of the hardware and software. Links to official distributors are available here: https://github.com/felis/USB_Host_Shield_2.0#summary.

closed 2016-02-25 by Lauszus