felis/USB_Host_Shield_2.0 #216: PS3USB and XBOXONE not working

bbx10, 2016-03-04:
IDE 1.6.7, UHS2 library 1.2.1

The discussion started on arduino.cc.

http://forum.arduino.cc/index.php?topic=382891.0

My PS3 controller connected via USB worked with an older version (1.0.0) of the library but does not work with 1.2.1. I found a workaround which involves commenting out 1 line in Usb.cpp. I am not sure this is the right way to fix the problem since this affects all USB devices. Debug output showed a stream of data toggle errors so I commented out the line that recovers from toggle errors. I can create a PR but this might break some other device.

```
Usb.cpp
@@ -234,7 +234,7 @@
                 rcode = dispatchPkt(tokIN, pep->epAddr, nak_limit); //IN packet to EP-'endpoint'. Function takes care of NAKS.
                 if(rcode == hrTOGERR) {
                         // yes, we flip it wrong here so that next time it is actually correct!
+                        // no, do not flip it because it generates more toggle errors
+                        //pep->bmRcvToggle = (regRd(rHRSL) & bmRCVTOGRD) ? 0 : 1;
-                        pep->bmRcvToggle = (regRd(rHRSL) & bmRCVTOGRD) ? 0 : 1;
                         regWr(rHCTL, (pep->bmRcvToggle) ? bmRCVTOG1 : bmRCVTOG0); //set toggle value
                         continue;
                 }

```

Another user has a similar problem with an XBOX ONE controller but the 1 line workaround does not help. I do not have an XBOX ONE controller so I can not do much more. I will post a link to this issue on the arduino.cc thread.

Lauszus, 2016-03-05:
@bbx10 I believe this is a duplicate of: https://github.com/felis/USB_Host_Shield_2.0/pull/174.

bbx10, 2016-03-07:
This fix works for PS3USB. I will see if this works for the user with the XBOX ONE.

xxxajk, 2016-03-07:
Toggle errors are rare these days, and only appear on old broken hardware. Its probabbly safe to comment it out. UHS30 doesn't correct for it anymore either.

bbx10, 2016-03-07:
The XBOXONE user reports #174 does not help. USB keyboard/mouse works so the shield hardware is working.

Lauszus, 2016-03-08:
@bbx10 is he using an official controller? Also he could try to power it through a USB Hub: https://github.com/felis/USB_Host_Shield_2.0#faq.

bbx10, 2016-03-09:
The user is using a Keyes board and tried plugging in external power via the Uno barrel connector. I can suggest trying a powered USB hub.

Timesquad, 2016-03-18:
Hi :) 
I'm the guy who can't get his Xbox One controller to work, it is an official controller and it works fine when plugged into a windows machine. It also runs fine from an un-powered USB hub that runs the arduino itself.

the 2 problems that I can think of are the following:

1) I've got the latest firmware installed which may cause incompatibilities

2) I've got the updated controller with the built in headphone jack which might be delivering signals slightly differently

The symptoms are as follows: once I've uploaded the example sketch to the arduino and I plug the controller in I get an initial buzz to let me know it has power but the serial monitor doesn't send anything other then "XBOX USB Library Started" on start up regardless of buttons pressed. The LED in the Xbox buttons stays off during all this.

that's about as best as I can describe the problem I'm having with the Xbox One controller but here's the link to the arduino.cc forum post: https://forum.arduino.cc/index.php?topic=382891.0

Thanks for your help!

Lauszus, 2016-03-21:
Unfortunately I don't own a Xbox One controller. If someone donates one I will look into the issue.

@Timesquad did you remember to enable serial debugging: https://github.com/felis/USB_Host_Shield_2.0#enable-debugging, so we can see what is going on?

Timesquad, 2016-03-26:
Debug gives the following information:
 I don't actually know what to do with this information.

XBOX USB Library Startedror code: D1

Xbox One Controller Disconnected

XBOX USB Library Started
Unknown Device Connected - VID: 045E PID: 02DD
Xbox One Init Failed, error code: D1

Xbox One Controller Disconnected

Lauszus, 2016-03-26:
@Timesquad looks like its another hardware revision of the controller. Try setting this value: https://github.com/felis/USB_Host_Shield_2.0/blob/master/XBOXONE.h#L39 to 0x02DD.

Also please run the following example: https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/USB_desc/USB_desc.ino and post the output.

Timesquad, 2016-03-26:
Slight progress, when I run the XBOXONE example sketch. I get the following output:

XBOX USB Library Started
Xbox One Controller Connected

XboxCommand, Return: 00
XboxCommand, Return: 00

USB_desc gives me :

Start
dpoint descriptor:
Endpoint address:	83
Attr.:			02
Max.pkt size:		0040
Polling interval:	00

Addr:1(0.0.1)
l:	00

Endpoint descriptor:
Endpoint address:	83
Attr.:			02
Max.pkt size:		0040
Polling interval:	00

Addr:1(0.0.1)
Start

01
--

Device descriptor: 
Descriptor Length:	12
Descriptor type:	01
USB version:		0200
Device class:		FF
Device Subclass:	47
Device Protocol:	D0
Max.packet size:	40
Vendor  ID:		045E
Product ID:		02DD
Revision ID:		0203
Mfg.string index:	01
Prod.string index:	02
Serial number index:	03
Number of conf.:	01

Configuration descriptor:
Total length:		0060
Num.intf:		03
Conf.value:		01
Conf.string:		00
Attr.:			A0
Max.pwr:		FA

Interface descriptor:
Intf.number:		00
Alt.:			00
Endpoints:		02
Intf. Class:		FF
Intf. Subclass:		47
Intf. Protocol:		D0
Intf.string:		00

Endpoint descriptor:
Endpoint address:	01
Attr.:			03
Max.pkt size:		0040
Polling interval:	04

Endpoint descriptor:
Endpoint address:	81
Attr.:			03
Max.pkt size:		0040
Polling interval:	04

Interface descriptor:
Intf.number:		01
Alt.:			00
Endpoints:		00
Intf. Class:		FF
Intf. Subclass:		47
Intf. Protocol:		D0
Intf.string:		00

Interface descriptor:
Intf.number:		01
Alt.:			01
Endpoints:		02
Intf. Class:		FF
Intf. Subclass:		47
Intf. Protocol:		D0
Intf.string:		00

Endpoint descriptor:
Endpoint address:	02
Attr.:			01
Max.pkt size:		00E4
Polling interval:	01

Endpoint descriptor:
Endpoint address:	82
Attr.:			01
Max.pkt size:		00E4
Polling interval:	01

Interface descriptor:
Intf.number:		02
Alt.:			00
Endpoints:		00
Intf. Class:		FF
Intf. Subclass:		47
Intf. Protocol:		D0
Intf.string:		00

Interface descriptor:
Intf.number:		02
Alt.:			01
Endpoints:		02
Intf. Class:		FF
Intf. Subclass:		47
Intf. Protocol:		D0
Intf.string:		00

Endpoint descriptor:
Endpoint address:	03
Attr.:			02
Max.pkt size:		0040
Polling interval:	00

Endpoint descriptor:
Endpoint address:	83
Attr.:			02
Max.pkt size:		0040
Polling interval:	00

Addr:1(0.0.1)

Lauszus, 2016-03-26:
@Timesquad try to uncomment this line: https://github.com/felis/USB_Host_Shield_2.0/blob/master/XBOXONE.cpp#L25.

Timesquad, 2016-03-26:
I get the following

02 20 51 1C 00 00 5E 04 
02 20 55 1C 7E ED 8C 4E 21 F1 00 00 5E 04 
0 53 1C 7E ED 8C 4E 21 F1 00 00 5E 04 
02 20 54 1C 7E ED 8C 4E 21 F1 00 00 5E 04 
02 20 55 1C 7E ED 8C 4E 21 F1 00 00 5E 04 

XBOX USB Library Started
Xbox One Controller Connected

XboxCommand, Return: 00
XboxCommand, Return: 0002 20 5A 1C 7E ED 8C 4E 21 F1 00 00 5E 04 
02 20 5B 1C 7E ED 8C 4E 21 F1 00 00 5E 04 
02 20 5C 1C 7E ED 8C 4E 21 F1 00 00 5E 04 
02 20 5D 1C 7E ED 8C 4E 21 F1 00 00 5E 04 
02 20 5E 1C 7E ED 8C 4E 21 F1 00 00 5E 04 
02 20 5F 1C 7E ED 8C 4E 21 F1 00 00 5E 04 
02 20 60 1C 7E ED 8C 4E 21 F1 00 00 5E 04 
02 20 61 1C 7E ED 8C 4E 21 F1 00 00 5E 04 
02 20 62 1C 7E ED 8C 4E 21 F1 00 00 5E 04 
02 20 63 1C 7E ED 8C 4E 21 F1 00 00 5E 04 
02 20 64 1C 7E ED 8C 4E 21 F1 00 00 5E 04 
            ^
This number just keeps climing

Lauszus, 2016-03-26:
@Timesquad okay. Do the bytes change when you press a button etc?

Timesquad, 2016-03-26:
Sadly no :/

Timesquad, 2016-04-20:
Anything else I could try?

Lauszus, 2016-04-20:
@Timesquad I don't really know, since don't own a Xbox One controller, so it is hard for me to debug your problem.

Timesquad, 2016-04-21:
How about I lend you mine?

Lauszus, 2016-04-25:
@Timesquad I don't have much time to look into it at the moment, so I would rather not borrow one.

Lenni, 2017-06-15:
I found a fix to get newer controllers workling! You have to send 4 bytes to initialize the Controller if it has the id 0x02DD. The original driver only sends two, so ther Controller cant connect. You can fix this by changing these lines: https://github.com/felis/USB_Host_Shield_2.0/blob/master/XBOXONE.cpp#L180 to 
    writeBuf[0] = 0x05;
    writeBuf[1] = 0x20;
    writeBuf[2] = 0x00;
    writeBuf[3] = 0x01;
    writeBuf[4] = 0x00;
    rcode = XboxCommand(writeBuf, 5);

@Lauszus Is it possible to implement this fix in such a way, that the driver chooses the right bytes to start the Controller by itself?

Lauszus, 2017-06-15:
@Lenni thanks a lot for your finding. I will prepare a PR right now.

Lauszus, 2017-06-15:
@Lenni do you have an older Xbox One controller by any change? It would be nice to know if you can send out all five bytes to the old controllers as well?

xxxajk, 2017-06-15:
check vid:pid perhaps, and the revision as hints for what to send?

Lauszus, 2017-06-15:
@xxxajk I don't think that is necessary. My guess is that we can just send the longer message by default.

Lenni, 2017-06-15:
@Lauszus I dont have an older Controller, so I cant check. But if you can wait 1 or 2 days, I can write a patch, so that the driver automatically determines the Version of the Controller. With this we would also fix the Error of the Driver not knowing the newer PICS.

Lauszus, 2017-06-15:
@Lenni please see https://github.com/felis/USB_Host_Shield_2.0/pull/298 and report back.

Lenni, 2017-06-15:
@Lauszus Looks good, but it would be great if you could implement the full list of PICs als found here: 
https://github.com/torvalds/linux/blob/master/drivers/input/joystick/xpad.c#L137

Lauszus, 2017-06-15:
@Lenni will do that now. Btw it's called PIDs ;)

Lauszus, 2017-06-15:
@Lenni see: https://github.com/felis/USB_Host_Shield_2.0/pull/298/commits/e7d0695616c9f8f1d845737e04140f0495cbba18.

Lenni, 2017-06-15:
@Lauszus  Oops, sorry. I just recently started tinkering with this :D  Looks great BTW! I can test the fix as soon as it is released!

Lauszus, 2017-06-15:
@Lenni just pull the XBOXONE_fix branch or download it at the following link: https://github.com/felis/USB_Host_Shield_2.0/archive/XBOXONE_fix.zip.

Lenni, 2017-06-15:
@Lauszus Works like a charm. I could only test with one Controller though....

Lauszus, 2017-06-15:
@Lenni awesome! I'll wait a couple of days for @guruthree to hopefully test it as well and then merge it.

Timesquad, 2017-06-15:
I'll test mind when I get home.

Rather exited about this!

Thanks for your work weather it works or not :)

Lauszus, 2017-06-15:
@Timesquad thanks! :)

Timesquad, 2017-06-15:
IT LIVES!

SO happy!

nice work :D THANKS!

Lauszus, 2017-06-15:
@Timesquad great. Can you try to run the ```USB_desc``` example and print the output when the Xbox One controller is plugged in? Just want to see the VID and PID of your controller.

Timesquad, 2017-06-15:
this is what it says:

Start
ndpoint descriptor:
Endpoint address:	83
Attr.:			02
Max.pkt size:		0040
Polling interval:	00

Addr:1(0.0.1)
l:	00

Endpoint descriptor:
Endpoint address:	83
Attr.:			02
Max.pkt size:		0040
Polling interval:	00

Addr:1(0.0.1)
Start

01
--

Device descriptor: 
Descriptor Length:	12
Descriptor type:	01
USB version:		0200
Device class:		FF
Device Subclass:	47
Device Protocol:	D0
Max.packet size:	40
Vendor  ID:		045E
Product ID:		02DD
Revision ID:		0203
Mfg.string index:	01
Prod.string index:	02
Serial number index:	03
Number of conf.:	01

Configuration descriptor:
Total length:		0060
Num.intf:		03
Conf.value:		01
Conf.string:		00
Attr.:			A0
Max.pwr:		FA

Interface descriptor:
Intf.number:		00
Alt.:			00
Endpoints:		02
Intf. Class:		FF
Intf. Subclass:		47
Intf. Protocol:		D0
Intf.string:		00

Endpoint descriptor:
Endpoint address:	01
Attr.:			03
Max.pkt size:		0040
Polling interval:	04

Endpoint descriptor:
Endpoint address:	81
Attr.:			03
Max.pkt size:		0040
Polling interval:	04

Interface descriptor:
Intf.number:		01
Alt.:			00
Endpoints:		00
Intf. Class:		FF
Intf. Subclass:		47
Intf. Protocol:		D0
Intf.string:		00

Interface descriptor:
Intf.number:		01
Alt.:			01
Endpoints:		02
Intf. Class:		FF
Intf. Subclass:		47
Intf. Protocol:		D0
Intf.string:		00

Endpoint descriptor:
Endpoint address:	02
Attr.:			01
Max.pkt size:		00E4
Polling interval:	01

Endpoint descriptor:
Endpoint address:	82
Attr.:			01
Max.pkt size:		00E4
Polling interval:	01

Interface descriptor:
Intf.number:		02
Alt.:			00
Endpoints:		00
Intf. Class:		FF
Intf. Subclass:		47
Intf. Protocol:		D0
Intf.string:		00

Interface descriptor:
Intf.number:		02
Alt.:			01
Endpoints:		02
Intf. Class:		FF
Intf. Subclass:		47
Intf. Protocol:		D0
Intf.string:		00

Endpoint descriptor:
Endpoint address:	03
Attr.:			02
Max.pkt size:		0040
Polling interval:	00

Endpoint descriptor:
Endpoint address:	83
Attr.:			02
Max.pkt size:		0040
Polling interval:	00

Addr:1(0.0.1)

Lauszus, 2017-06-15:
@Timesquad hmm that is the new Xbox One controller. It would be nice if someone with a controller with a PID of 0x02D1 could test it...

guruthree, 2017-06-17:
It works OK with my controller, which is a 0x02D1. Notably my controller doesn't work without the patch. I did a firmware update maybe a year ago on it, which is probably the issue.

commit f286114a (2017-06-15): Newer Xbox One controllers requires a longer initialisation command

Fixes #216
PR #298 (2017-06-25): Newer Xbox One controllers requires a longer initialisation command
Fixes #216
closed 2017-06-25 by Lauszus