felis/USB_Host_Shield_2.0 #217: Example for full HID report for keyboard

HakanL, 2016-03-05:
I'm trying to read the special multimedia keys from a keyboard. I'm using the latest git code so I can pass in true to the hidboot constructor, and I have added a dummy parser which I assume should be called for any report coming in, but it's only called for the standard boot-type reports. Any example or suggestion how I can capture all keys from the keyboard? I realize I have to parse the data myself, I'm ok with that.

Thanks!

Lauszus, 2016-03-07:
@HakanL it will properly be easier to inherit the ```HIDUniversal``` class and then inherit the virtual function ```ParseHIDData```.

Check out the PS4USB driver: https://github.com/felis/USB_Host_Shield_2.0/blob/master/PS4USB.h and the SRWS1 driver which is still to be merged into master: https://github.com/felis/USB_Host_Shield_2.0/blob/SRWS1/SRWS1.h.

HakanL, 2016-03-08:
Thank you, that helped. However I still couldn't get more data than the basic keys. I decided to run a USB sniffer session with "Device Monitoring Studio" and i now understand this a little more. The keyboard is actually a composite device which has two endpoints, one for the regular keyboard keys (numeric, arrow up/down, etc) and a different endpoint for the special keys. I was able to capture all this data in the sniffer program, so one further step forward. And the 2nd endpoint also shows up as HID device (albeit a different type, consumer control).

When I run USB_Desc I see this endpoint: https://gist.github.com/HakanL/7845dc3eba869948a3f8

It's endpoint address 82 (the normal keyboard keys are on 81). I've wired up a HidUniversal which captures the normal keys, but my question is now, how can I config/init the HidUniversal to select this other endpoint? I figured it would happen in `EndpointXtract`, but I can't figure out how to select this endpoint/interface.
Any help is greatly appreciated. I hope to write an example for how to capture the full keyboard (normal + multimedia keys) and contribute that to this project.

xxxajk, 2016-03-08:
This is where UHS2 falls short. You will have to implement it so it handles
both. UHS3 will be capable of doing this sort of thing a lot easier.

Lauszus, 2016-03-08:
@HakanL you should be able to by adding one more endpoint here: https://github.com/felis/USB_Host_Shield_2.0/blob/0fce0106ed4005a6f9611e7990b3b471168cc8db/usbhid.h#L149-L150. Then increase this by one: https://github.com/felis/USB_Host_Shield_2.0/blob/0fce0106ed4005a6f9611e7990b3b471168cc8db/usbhid.h#L153 and then modify these lines: https://github.com/felis/USB_Host_Shield_2.0/blob/master/hiduniversal.cpp#L318-L321, so it checks if it is the first or the second input endpoint.

If you wan't to write a driver from scratch you can have a look at the PS3USB driver: https://github.com/felis/USB_Host_Shield_2.0/blob/master/PS3USB.h.

HakanL, 2016-03-10:
Thanks for your help! Turns out it wasn't a new endpoint that I needed to listen to, it was a new interface (which then had another interrupt-in endpoint), identified as a mouse, that I had to listen to. I added a new class called HIDComposite that calls SelectInterface where you can specify if you want to listen to an interface or not. I've tested this with the multimedia keyboard I was experimenting with and now I see all keys. I had tremendous help from a hardware USB Protocol Analyzer and a friend who understood what USB packets to expect. My commit also includes a sample that reports the keys coming in from the two interfaces. I haven't done a PR yet, I wanted to test it out a little more first.

Lauszus, 2016-03-13:
Sounds like the following example would have worked then: https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/HID/USBHIDBootKbdAndMouse/USBHIDBootKbdAndMouse.ino?

HakanL, 2016-03-14:
I tried that, adding a debug message in the hid parser, but it never received the special messages. I didn't analyze it deeper, but my guess is that a keyboard and a mouse has a different configuration than a multimedia keyboard (which also show up as a keyboard and a mouse). Maybe the different is that a multimedia keyboard is one configuration, two interfaces? But again, I didn't compare exactly, I just know the USBHIDBootKbdAndMouse didn't work and I was able to to create a class to make it work.

Lauszus, 2016-03-17:
@HakanL did you try to use "Report Protocol mode" which can now be set in the constructor: https://github.com/felis/USB_Host_Shield_2.0/pull/210.

HakanL, 2016-03-17:
I tried that, didn't work. I learned (by using a USB hardware analyzer) that these multimedia keys aren't sent as Report Protocol even in Windows, they follow the boot protocol, they come over as a 2nd interface on the mouse configuration.

xxxajk, 2016-03-21:
Oh nasty! 
Just another :8ball: device in the wild, where you have to flip it and see what you get.
Most normalish keyboards will spit out these extra keys as additional keycodes instead, which would have been the right way for them to do this.

Lauszus, 2016-03-21:
@HakanL okay. Will close this issue now, as it has been resolved.

closed 2016-03-21 by Lauszus

HakanL, 2016-03-21:
Actually, I think this is how many multimedia keyboard do it, I have verified two different keyboards do it the same way (granted they are both from Logitech...). And Windows recognize it fine.

Lauszus, 2016-03-21:
@HakanL okay. If you have time a PR would be awesome :)

HakanL, 2016-03-21:
Of course, done. Not sure it's the cleanest way to add a new class (HIDcomposite), please comment/change as necessary.