felis/USB_Host_Shield_2.0 #218: Why there are wValueLo and wValueHi?

Terminus-IMRC, 2016-03-15:
Why there are ```wValueLo``` and ```wValueHi``` even though there is ```wValue``` in ```UsbCore.h```? This does not make problems since they are union. However, because of this, I have to write bit shifts in the arguments of ```USB::ctrlReq``` in ```Usb.cpp```. Is there any reason why they are so? If there is no reason, please fix this.

xxxajk, 2016-03-21:
As it says in the comment...
it all depends on bRequest.

Some requests are a single value (16bit... is what it should be, int is clearly incorrect here!)
Others want two parameters, 8bits each.
BOTH are in the same position.

xxxajk, 2016-03-21:
This is how it really should be. 
[https://github.com/felis/UHS30/blob/master/libraries/UHS_host/UHS_UsbCore.h#L258](url)
UHS3.0 fixes this problem.
UHS2.0 will be deprecated as soon as I can possibly do it.

Terminus-IMRC, 2016-03-21:
Thank you for your reply. Now I understand what the comment means.
However, there still be ```wValLo``` and ```wValHi``` as arguments to ctrlReq: https://github.com/felis/UHS30/blob/master/libraries/UHS_host/UHS_usbhost.h#L234 . Wouldn't it be only ```wVal```?

xxxajk, 2016-03-21:
Good question! 
I have to check but depending on the CPU passing 2 8bit args v.s. 1 16bit arg may actually make for a smaller stack frame. I'll have to check that out. 
It may end up as an improvement in UHS3, or not. It really depends on the usage across drivers. If most of them use 8bit values, it may actually be better to use them separately.

Terminus-IMRC, 2016-03-21:
Thank you for considering my suggestion.
Regardless of the size of a stack frame, I suggest you to make ```wValue``` be 1 16bit argument because:

* Users who want to set 1 16-bit ```wValue``` don't need to do bit shift operations like this: https://github.com/felis/UHS30/blob/master/libraries/UHS_CDC_ACM/UHS_CDC_ACM_INLINE.h#L268
* Users who want to set 2 8-bit ```wValue``` would use a function or a macro like this: ```#define UHS30_CONV_WVAL8(lo, hi) ((lo) | ((hi) << 8))```
* libusb's ```libusb_control_transfer``` has 1 16-bit ```wValue``` argument: http://libusb.sourceforge.net/api-1.0/group__syncio.html#gadb11f7a761bd12fc77a07f4568d56f38

xxxajk, 2016-03-21:
I can do even better... a nice macro that converts the 8 bytes of args into a uint32_t :-)

xxxajk, 2016-03-21:
I mean 2 32's.... could try for a uint64_t too, 8-)

Terminus-IMRC, 2016-03-21:
I see. So what would the arguments of ```ctrlReq``` look like?

xxxajk, 2016-03-21:
the same, more-or-less.... just that it would have some compile-side optimization

Terminus-IMRC, 2016-03-21:
I see. Thank you! I'm looking forward to seeing the new arguments!

xxxajk, 2016-03-21:
Here's two examples that I'll give you from UHS3.0 code
`
pUsb->ctrlReq(bAddress, bmREQ_CDCOUT, UHS_CDC_SEND_BREAK, (duration & 0xff), (duration >> 8), bControlIface, 0, 0, NULL);
`
would be written as
`

pUsb->ctrlReq(bAddress, mkSETUP_PKT8(bmREQ_CDCOUT, UHS_CDC_SEND_BREAK, (duration & 0xff), (duration >> 8), bControlIface, 0), 0, NULL);
or (better for this case)
pUsb->ctrlReq(bAddress, mkSETUP_PKT16(bmREQ_CDCOUT, UHS_CDC_SEND_BREAK, duration, bControlIface, 0), 0, NULL);

`

and

`
pUsb->ctrlReq(bAddress, UHS_BULK_bmREQ_OUT, UHS_BULK_REQ_BOMSR, 0, 0, bIface, 0, 0, NULL)
`
would be written as
`
pUsb->ctrlReq(bAddress, mkSETUP_PKT8(UHS_BULK_bmREQ_OUT, UHS_BULK_REQ_BOMSR, 0, 0, bIface, 0), 0, NULL)
pUsb->ctrlReq(bAddress, mkSETUP_PKT16(UHS_BULK_bmREQ_OUT, UHS_BULK_REQ_BOMSR, 0, bIface, 0), 0, NULL)

Neither of these has any advantage
`
In both cases the compiler wil optimize as much as possible with the constants. This should give a faster runtime and save a few bytes here and there at the same time, while remaining fully readable.

Terminus-IMRC, 2016-03-21:
Looks great!

xxxajk, 2016-03-21:
Thank you very much for pointing out a very obvious little-endian optimization!
UHS3.0 now uses it to shave a bunch of cycles off the code, and have as much as possible per-calculated by the compiler.
Total win here, because not only do I get speed, but it actually did shave a few bytes off.

Terminus-IMRC, 2016-03-21:
Thank you!

This issue is solved in this commit: https://github.com/felis/UHS30/commit/112689dc41acf84d21f4c5fc98d31ab8471b4fb1 . So I'm closing it.

closed 2016-03-21 by Terminus-IMRC